letscontrolit/ESPEasy #4433: [BUILD] Add 2-step minimal uploader binary with LittleFS (instead of SPIFFS)

clumsy-stefan, 2022-12-25:
I recently changed all my nodes to LittleFS as it results in smaller binaries (especially handy for 1M builds). Now the problem is, that the 2-step uploader binary included with ESPEasy can't read the config anymore from the filesystem.

So I suggest to include a minimal 2-step uploader binary with only LittleFS compiled in it (could result in even smaller binary and therefore more space for the ESPEasy builds).

IMHO as always...

TD-er, 2022-12-25:
I think we should have a 2-step OTA build without any file system and then store the WiFi credentials in the currently unused EEPROM "partition".
This will make the binary a lot smaller as we don't need to implement any file system libraries, nor implement the security.dat struct.

clumsy-stefan, 2022-12-25:
that would even be the better idea, so besides the smaller size it wouldn't matter what file system is used... it juest would need to make sure, that the existing FS isn't getting corrupted during reboots and flashing...

Probably for a first step it would be quite easy to have a version with LittleFS (as it only requires a simple search and replace and one include) and later have it done as you suggest...

TD-er, 2022-12-25:
> Probably for a first step it would be quite easy to have a version with LittleFS (as it only requires a simple search and replace and one include) and later have it done as you suggest...

Nope, as that will never ever fit.
For LittleFS we need to use a quite recent SDK -> minimal bin size is too larger -> Won't fit.

The only things we can do here:
- Use the EEPROM section as I mentioned
- Make a new 1M build type with no file system at all (as is done by the Tasmota team)

This last option will leave a lot more room for the sketch + OTA space and the minimal 2-step OTA can be a lot smaller.

Still need to figure out if we can have a build which does allow for .bin.gz files, but I am not sure if that will give us a lot of extra room.

clumsy-stefan, 2022-12-25:
> For LittleFS we need to use a quite recent SDK -> minimal bin size is too larger -> Won't fit.

Ah, didn't realise this. Probably that's why my own builds are always larger than the one provided (I thought it's because of the ArduinoIDE)..

At least the WiFi credentials need to be stored somewhere (if there's no FS), everything else just need to survive the flashing (eg. FS)..

So I guess as for now, it's not possible to use LittleFS on a 1M node as OTA will not work anymore... Probably this needs to be at least noted somewhere not that others run into the same issue and brick their 1M units (when not easy accessible)...

TD-er, 2022-12-25:
> So I guess as for now, it's not possible to use LittleFS on a 1M node as OTA will not work anymore... Probably this needs to be at least noted somewhere not that others run into the same issue and brick their 1M units (when not easy accessible)...

As far as I know, we have not made any 1M build with LittleFS... right?

clumsy-stefan, 2022-12-25:
> As far as I know, we have not made any 1M build with LittleFS... right?

basically true.. I made my own... I switched all my custom builds to LittleFS (as it seems quicker and needs much less space...) still, probably I wouldn't have changed the 1M builds if I knew it before ;)

TD-er, 2022-12-25:
Well, at least good to know, it can still work on 1M builds with LittleFS.
Problem with SPIFFS is that it is no longer maintained, so we need to make some switch at some point.

clumsy-stefan, 2022-12-25:
Yep, that was the trigger for me, as the compiler gave lots of warnings...

As I scripted a complete backup for all my nodes it was relatively easy to push the configs once the nodes were flashed with the new build..

Going to LittleFS save me roughly 5K on my builds....

clumsy-stefan, 2022-12-30:
I tried to put together a bare HTTP or OTA uploader and stripped everthing out (especially all FS stuff, configuration, AP etc.) statically defining SSID & Key in the source. Still when I compile in ArduinoIDE the binary is >300k (actually larger than the one provided in ESPEasy). Unfortunately as this would have solved my issue...

I can't see why the binaries get so much larger in ArduinoIDE...

TD-er, 2022-12-30:
It probably isn't because of the Arduino IDE, but more likely because of the immer growing SDK size.

clumsy-stefan, 2022-12-30:
Could be, unfortunately I can only go back to 2.1. but it does seem to make a difference...

clumsy-stefan, 2023-01-13:
I succeeded in building a minimal OTA binary stripping everything out except ArduinoOTA. The WiFi credentials are fixed in the code. By using the 2.2.1 IDE it resulted in a binary that's slightly bigger than your minimal build. However like this I can stay with littleFS for my 1M build which saves ~8k. My minimal flash binary is about 5k bigger than the one provied in the distribution, so I still save some space over all and am not dependant on SPIFFS anymore.

BTW: going back to 2.1.1 IDE makes the binary bigger again (as does all later versions), so for me I found the optimal combination.

It still would be great if you could have a look into it if it's possible to either use LittleFS or no FS at all for the 2step flash!

TD-er, 2023-01-13:
> or no FS at all for the 2step flash!

I'm thinking about a no-FS ESPEasy build for 1M units.
This way we can start writing settings from the end of the flash, leaving more space for OTA updates.