letscontrolit/ESPEasy #4440: IRext ESP8288 1M files + log output

tomatos-st, 2022-12-29:
Is it possible to add ESP8288 1M build with IR plugin and log window?

[image]

At the moment, in the only version 1M (minimal_IRext_ESP8266_1M) there is no log function, which makes it a bit difficult to analyze the IR codes when I cannot physically connect the RS port

TD-er, 2022-12-29:
Is the web log feature not included at all, or are you missing the JavaScript to fetch the log?
In other words, do you get a JSON when opening this URL:  `http://<ip>/logjson`  ?

tomatos-st, 2022-12-29:
The web log feature not included at all.
[image]

[image]

tonhuisman, 2022-12-29:
For the `minimal` builds, the Web Log interface is disabled, to make room for other features.
I have made a special build of `minimal_IRext_ESP8266_1M` that has the Web Log enabled:
[ESP_Easy_mega_20221229_minimal_IRext_ESP8266_1M.zip](https://github.com/letscontrolit/ESPEasy/files/10322159/ESP_Easy_mega_20221229_minimal_IRext_ESP8266_1M.zip)

I have created this build by commenting out `#undef WEBSERVER_LOG` around line 513 in `define_plugin_sets.h`, and then building the environment named above. The developer guide on how to compile ESPEasy using VSCode and PlatformIO is [here](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html), so you might be able to do that yourself?

tomatos-st, 2023-01-05:
I'm weak in programming :) But I managed to compile and it works. In the next build, it can append to the package, it doesn't take much space I guess.
Thanks for the fix.

tonhuisman, 2023-02-24:
@tomatos-st Can this issue be closed? I don't expect to change the `minimal_IRext` build, as for minimal 1M builds all logging is disabled by default, but you can create your own Custom build if desired. That's also documented in the link provided earlier.

tomatos-st, 2023-02-24:
We can.
However, in my opinion, compiling IRext without logging is unreasonable.

closed 2023-02-24 by tomatos-st