letscontrolit/ESPEasy #4444: PCF8574 plugin missing from CLIMATE collection?

agroszer, 2023-01-02:
The doc states https://espeasy.readthedocs.io/en/latest/Plugin/P019.html should be present in the CLIMATE collection.
Using Build: ESP_Easy_mega_20221224_climate_ESP8266_4M1M Dec 24 2022 I don't find it in the devices list.
[image]
Is it me or is it a bug?

TD-er, 2023-01-02:
The external GPIO support has been moved to be internal commands.
See: https://espeasy.readthedocs.io/en/latest/Reference/Command.html?highlight=pcf#external-pcfgpio

agroszer, 2023-01-02:
nice, thanks

agroszer, 2023-01-06:
Something is wrong, i2cscan detects
```
0x20 | PCF8574 MCP23017 LCD PCF8575
```
but issuing the command with Tools/Command
```
PCFGPIO,1,1
```
returns
```
Command unknown: PCFGPIO,1,1
```
is it me again?

TD-er, 2023-01-06:
The I2C scan just returns a number of strings based on the scan result.
This is no indication of whether a plugin is included in the build.

agroszer, 2023-01-06:
:-( then I guess it's not included, using the CLIMATE build.

TD-er, 2023-01-06:
Moah, does sound like a plugin which can be associated with a build named "climate" ;)

agroszer, 2023-01-06:
digging in `define_plugin_sets.h` shows 
```
#ifdef PLUGIN_SET_STABLE
...
#define USES_P019   // PCF8574
```
looks like PCF8574 support is only in that build, prolly the NORMAL one?

agroszer, 2023-01-08:
So what? 
- Fix the doc according to reality
- Try to squeeze in p019 into the climate collection
I could come up with a pull request either way, I just can't tell whether p019 will fit the climate collection

tonhuisman, 2023-01-08:
I'll add it in the Climate build, after checking that for size.

tonhuisman, 2023-01-08:
> using the CLIMATE build to have the AHT20 plugin.

The AHT20 is also in the Collection A builds, and that _does_ include the PCF8574 plugin, as each Collection build is based on Normal.

Documentation does state that PCF8574 should be included in Climate, but obviously it isn't (yet).

closed 2023-01-10 by TD-er

agroszer, 2023-01-14:
Thank you, could someone please hit the release button?

tonhuisman, 2023-01-14:
Have you tried the Collection A build? There both the AHT20 and PCF8574 _are_ already available. You can upgrade/replace the firmware, even using OTA, without losing any config, as long as you use the same flash layout, like 4M1M or 4M316k (ESP32).

agroszer, 2023-01-18:
Yes, thank you, works

PR #4443 (2023-01-10): [Build] Reduce bin size by string reductions and other optimizations
Shaving off a few bytes from build .bin size

Features:
- Minimize strings in `PLUGIN_WEBFORM_LOAD` and `PLUGIN_WEBFORM_SAVE`
- Remove 1-time used variables
- Some source reformatting
- Other minor optimizations
- Other features:
  - [P103] Refactor javascript UI functions
  - [P039] Add Low temperature threshold option
  - [Build] Add P019 PCF8574 to `Climate` build, as documented
  - [PLUGIN_INIT] All plugins will now return `success = true` if initialization is successful
  - [Devices] Check Task settings on save except `M [... cut]