letscontrolit/ESPEasy #4446: [Feature] Add PostToHTTP command for HTTP POST

TD-er, 2023-01-03:
As mentioned on the [forum](https://www.letscontrolit.com/forum/viewtopic.php?p=62188#p62188)

A new command, like SendToHTTP, but then to make a HTTP POST call.

Not sure yet about what parameters it may need as the arguments of what to POST may be a bit tricky.
Perhaps it can be a single string? (N.B. the URL is of course also a separate string)

BasvanH, 2023-01-06:
I need this very much, when you want to update an item in OpenHAB thru the REST API it requires the HTTP method to be POST.

I want to use a ESPEasy rule to update an OpenHAB item, currently the SendToHTTP command uses the HTTP GET method.

https://www.openhab.org/docs/configuration/restdocs.html

tonhuisman, 2023-01-15:
@BasvanH I have created a PR implementing the `PostToHTTP` command, can you test from the Github [Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/3923695893) (once that's finished) to see if this fits your usage scenario?

BasvanH, 2023-01-16:
@tonhuisman, great, yes I will conduct tests and report back here.

BasvanH, 2023-01-18:
@tonhuisman, Ive successfully used the PostToFTTP in a rule to make POST requests to the OpenHAB REST API. 

```
PostToHTTP <IP>,8080,/rest/items/Item_Name,'Content-Type:text/plain',ON
```

```
2688013: ACT : PostToHTTP <IP>,8080,/rest/items/Item_Name,'Content-Type:text/plain',ON
```

PR #4465 (2023-01-18): [Commands] Add PostToHTTP command
Resolves #4446 

Add support for command `PostToHTTP <host>,<port>,<path>,[<headers>][,<postbody>]`

Features:
- Send data to an external host via the HTTP `POST` verb.
- Support for Authorization:
  - Basic by supplying `<user>:<password>@<host>`.
  - Others like Bearer by supplying the required data via headers (`"Authorization: Bearer <token>"`).
- Optional multiple headers, separated using the `%LF%` variable.
- Documentation updated

NB: Not included in the `Minimal` builds.

TODO:
- [ ] Testing (tested locally already)
closed 2023-01-18 by TD-er