letscontrolit/ESPEasy #4454: recent FW, how can I set the unit number equal to the last MAC byte?

rudiratlos, 2023-01-09:
I want to set the unit number to the last mac byte by a rule.

I could set the hostname by:

NAME,ESP
UNIT,%lastMacByte%

TD-er, 2023-01-09:
Where do you need this?
If it is for connecting to a MQTT broker, then you can already configure the client ID like this.

tonhuisman, 2023-01-10:
Setting the unit number from last byte of MAC address can be achieved by this command:
`unit,{strtol:16:{substring:15:17:'%mac%'}}`

tonhuisman, 2023-01-15:
@rudiratlos The command I've provided above should solve this request. Can you confirm, and close this issue?

chromoxdor, 2023-01-17:
I might add, that this command doesn’t work with at least the latest build!
This seems to work:
`unit {strtol:16:{substring:15:17:'%mac%'}}`

tonhuisman, 2023-01-17:
`unit` is an internal command, so can only be applied from rules or serial console, not from Tools page as that's effectively using http.
Your test _does_ prove that this security mechanism is working as intended 👍

chromoxdor, 2023-01-17:
> `unit` is an internal command, so can only be applied from rules or serial console, not from Tools page as that's effectively using http. Your test _does_ prove that this security mechanism is working as intended 👍

i tested it with logentry

chromoxdor, 2023-01-17:
Here with `unit` in rules:

`796561: ACT : unit {strtol:16:D9}`

chromoxdor, 2023-01-17:
> `unit` is an internal command, so can only be applied from rules or serial console, not from Tools page as that's effectively using http.
> Your test _does_ prove that this security mechanism is working as intended 👍

if this is security related than i might write this in capital  letters:
`unit` DOES SET THE UNIT NUMBER FROM THE TOOLS PAGE IN EVERY BUILD I TESTED!!! (even if it sets the number to 0 because of strtol not working)

😁

tonhuisman, 2023-01-17:
Ah, I'll have a look at `strtol`, as it's not working as I expected.

I've turned it into a 2-liner:
```
On test1 Do
  Let,16,0x{substring:15:17:'%mac%'}
  LogEntry,"Unit: [int#16]"
Endon
```
Resulting in this output:
```
285038592: HTTP: event,test1
285038680: EVENT: test1
285038704: ACT : Let,16,0x88
285038717: ACT : LogEntry,'Unit: 136'
285038722: Unit: 136
```

TD-er, 2023-03-29:
Just curious, why not simply use the `mod` operator on the MAC address value?
```
let,1,%mac_int%
let,1,[int#1]%256
```