letscontrolit/ESPEasy #4455: Devices: Invalid character in name. Do not use ',-+/*=^%!#[]{}()' or space.

rudiratlos, 2023-01-10:
tested on newest 8266 and 32

Devices - Generic - System Info 

using .e.g a / in Values - Name

#1 status/RSSI
#2 status/VCC

I can use / in the Name fields on 8266 collection-A, but not in 32 collection-A

tonhuisman, 2023-01-10:
That check is excluded from the ESP8266 builds because of limits in the .bin size.
Possibly, things could go horribly wrong when using them, you will not know until if/when it fails 👻 ...

TD-er, 2023-01-10:
I think we should reinstate this check for LIMIT_BUILD_SIZE builds.
Hopefully it doesn't add to much space to it

rudiratlos, 2023-01-10:
Thnaks, now I'm using %tskname% in the publishing. Therefore I dont's neet to set it in device section

tonhuisman, 2023-01-10:
So you would get in trouble if we would decide to remove these invalid characters, sometime in the future?

TD-er, 2023-01-10:
Ah, that's rather clever.
I don't know why the / would be a not desired character. Any idea why this was included in this list of forbidden chars?
Only thing I can come up with is when parsing formulas, but as long as you don't name your tasks `0xa234/345` or something like that, I don't see why it can't have these.

tonhuisman, 2023-01-10:
> Any idea why this was included in this list of forbidden chars?

> ...encoding issues...

and it can be mistaken for a division operation during calculations.

tonhuisman, 2023-01-10:
But despite the warning you can still save these names containing invalid characters 😉

PR #4443 (2023-01-10): [Build] Reduce bin size by string reductions and other optimizations
Shaving off a few bytes from build .bin size

Features:
- Minimize strings in `PLUGIN_WEBFORM_LOAD` and `PLUGIN_WEBFORM_SAVE`
- Remove 1-time used variables
- Some source reformatting
- Other minor optimizations
- Other features:
  - [P103] Refactor javascript UI functions
  - [P039] Add Low temperature threshold option
  - [Build] Add P019 PCF8574 to `Climate` build, as documented
  - [PLUGIN_INIT] All plugins will now return `success = true` if initialization is successful
  - [Devices] Check Task settings on save except `M [... cut]
closed 2023-01-10 by TD-er