letscontrolit/ESPEasy #4459: [C014 & P086] Can't set Generic Homie Receiver parameter

fedemox, 2023-01-11:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [x] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [x] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x] Tested previous/other build (mention which one already tested)
- [x] Tested on other node to make sure hardware isn't defective.
- [x] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

I'd like to control a brushless fan driven by a ESC, providing PWM signal at 50Hz on pin 13.
Successfully tested command `PWM,13,value,0,50` with integer values from 11 to 30 (3D printed fan is super scary at high speed!!).
Successfully tested with controller `C005 - Home Assistant (openHAB) MQTT` & `P033 - Generic - Dummy Device`.
Successfully tested CMD and GPIO set of controller C014 (Homie).
**Failed** test with `C014 - Controller - Homie MQTT convention` & `P086 - Generic - Homie receiver`.

### Expected behavior

Controller algorithm should not discard properly formatted MQTT command. 

### Actual behavior

MQTT messages on topic `homie/ESPeasySerever/fan/speed/set` are always flagged as INVALID MSG and ignored.

### Steps to reproduce

1. Use MQTT Explorer or OpenHAB to post on topic `homie/ESPeasySerever/fan/speed/set` the value `11`
2. Observe the logged record `C014 : MQTT received: /set: N: fan V: speed INVALID MSG`
3. Gnash your teeth.

### System configuration

Hardware:  NodeMCU Amica V2 ESP8266 ESP-12F ([amazon.it](https://www.amazon.it/gp/product/B06Y1LZLLY/))

ESP Easy version: released binaries with names
- `ESP_Easy_mega_20221105_collection_A_ESP8266_4M1M.bin`
- `ESP_Easy_mega_20221224_collection_A_ESP8266_4M1M.bin`

ESP Easy settings/screenshots: 
 - Controller IP: `192.168.1.113` (tested also `test.mosquitto.org` and `broker.hivemq.com`)
 - Controller Port: `1883`
 - Credentials: username & password set
 - Controller Client ID: `%sysname%` (---> `ESPeasyServer`)
 - Controller Subscribe: `homie/%sysname%/#` (prefilled value was rejected by every broker)
 - Controller Publish: `homie/%sysname%/%tskname%/%valname%`
 Device:
- Name: `fan`
- Function#1 event name: `speed`
- Function#1 parameter type: `integer`
- Function#1 decimals: `0`
everything else untouched

### Rules or log data

Rules:
```
On fan#speed do
LogEntry,"%eventvalue1|ND%, %eventvalue2|ND%"
PWM,13,%eventvalue1%,0,50
homieValueSet,1,1,%eventvalue1%
endon
```
Log record:
```
341382: WIFI : DHCP IP: 192.168.1.147 (ESPeasyServer) GW: 192.168.1.1 SN: 255.255.255.0 duration: 25 ms
341392: WIFI : Connected! AP: mox-casa (00:14:BF:E1:C2:34) Ch: 6 Duration: 1056 ms
341396: EVENT: WiFi#Connected
341453: MQTT : Intentional reconnect
341472: MQTT : Connected to broker with client ID: ESPeasyServer
341478: Subscribed to: homie/ESPeasyServer/#
341536: C014 : autodiscover information of 3 Devices and 2 Nodes sent with no errors! (25 messages)
341541: EVENT: WiFi#ChangedAccesspoint
341553: EVENT: WiFi#ChangedWiFichannel
341561: EVENT: MQTT#Connected
346572: WD : Uptime 6 ConnectFailures 0 FreeMem 18216 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
361843: C014 : MQTT received: /set: N: fan V: speed INVALID MSG
362843: C014 : MQTT received: /set: N: fan V: speed INVALID MSG
363343: C014 : MQTT received: /set: N: fan V: speed INVALID MSG
364093: C014 : MQTT received: /set: N: fan V: speed INVALID MSG
364843: C014 : MQTT received: /set: N: fan V: speed INVALID MSG
366343: C014 : MQTT received: /set: N: fan V: speed INVALID MSG
367343: C014 : MQTT received: /set: N: fan V: speed INVALID MSG
375017: P086 : Value 4: 0.00
375027: EVENT: fan#speed=0
375035: ACT : PWM,13,0,0,50
375041: PWM : GPIO: 13 duty: 0 f: 50 Hz
375045: ACT : homieValueSet,1,1,0
375050: P086 : deviceNr:1 valueNr:1 valueType:0 integer/float set to 0.00
375055: C014 : homie acknowledge: fan taskIndex:1 valueNr:1 valueName:speed valueType:0 topic:homie/ESPeasyServer/fan/speed valueInt:0
375060: EVENT: fan#=0.00
375069: EVENT: fan#=0.00
375076: EVENT: fan#=0.00
376572: WD : Uptime 6 ConnectFailures 0 FreeMem 17712 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
```
[sysinfo.pdf](https://github.com/letscontrolit/ESPEasy/files/10392164/sysinfo.pdf)

[image]
[image]

tonhuisman, 2023-01-11:
Can you upgrade to the 20221224 release, as just before that release a bug in the area of finding a task by name was fixed, that _may_ be the cause of this issue.

TD-er, 2023-01-11:
`ESP_Easy_mega_20221224_collection_A_ESP8266_4M1M` is listed in the description as tested.

fedemox, 2023-01-11:
Failed also with `C014 - Controller - Homie MQTT convention` & `P033 - Generic - Dummy Device`.
Same behaviour:
```
374010: C014 : MQTT received: /set: N: dummyfan V: dummyspeed INVALID MSG
```

tonhuisman, 2023-09-09:
@fedemox Have you been able to configure your Homie receiver task so it will receive your values?

fedemox, 2023-09-19:
No. Same behaviour on firmware `ESP_Easy_mega_20230822_collection_A_ESP8266_4M1M.bin`

tonhuisman, 2023-09-19:
@fedemox I've done some work on MQTT controllers and the Homie receiver (with the intent to add Homie receiver support to all MQTT controllers), can you test the [last successful GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/6209579890) for that PR?

fedemox, 2023-09-25:
Half success! Now the controller acknowledges the received command, but its value isn't passed to the device.
Every minute the rule (see above) is triggered, but parameters are 0/empty.
```
212628 : Info   : WD   : Uptime 4 ConnectFailures 0 FreeMem 20120 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
217639 : Info   : C014 : MQTT received: /set: N: fan V: speed cmd:  OK
241072 : Info   : EVENT: Clock#Time=Mon,18:18
241278 : Info   : P086 : Value 1:
241279 : Info   : P086 : Value 2:
241280 : Info   : P086 : Value 3:
241281 : Info   : P086 : Value 4:
241286 : Info   : EVENT: fan#speed
241290 : Info   : ACT  : LogEntry,"ND, ND"
241292 : Info   : ND, ND
241294 : Info   : ACT  : PWM,13,0,0,50
241296 : Info   : PWM  : GPIO: 13 duty: 0 f: 50 Hz
241299 : Info   : ACT  : homieValueSet,1,1,0
241302 : Info   : P086 : deviceNr: 1 valueNr: 1 valueType: 0 integer/float set to 0.00
241305 : Info   : C014 : homie acknowledge: fan taskIndex:1 valueNr:1 valueName:speed valueType:0 topic:homie/Collection_A/fan/speed valueInt:0 valueStr:0 success!
241313 : Info   : EVENT: fan#
241316 : Info   : EVENT: fan#
241318 : Info   : EVENT: fan#
242628 : Info   : WD   : Uptime 4 ConnectFailures 0 FreeMem 20120 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
```

PR #4770 (2024-12-30): [MQTT] Controller improvements
Resolves #2341 
Resolves #4864 
Resolves #2560 
Resolves #4459 

Affected controllers:
- [C002] Domoticz MQTT
- [C005] Home Assistant (openHAB) MQTT
- [C006] PiDome MQTT
- [C014] Homie MQTT convention (V3/V4)
- [C007] Emoncms

Features:
- MQTT Controllers: generic support for `/cmd` and `/set` topics where applicable
- String/resource usage reductions (most of that in Homie controller)
- Added generic `CPlugin_helper_mqtt` module with MQTT support functions
- [C004] Small code improvements, add documentation and examples
- [C007] Add configurable url, stored in ` [... cut]
closed 2024-12-30 by TD-er