letscontrolit/ESPEasy #4466: [Question] gzipped byte array for css

chromoxdor, 2023-01-16:
It is possible to have gzipped css and it is also possible to serve gzipped html as a bytearray .(https://www.mischianti.org/2020/10/26/web-server-with-esp8266-and-esp32-byte-array-gzipped-pages-and-spiffs-2/)
Wouldn´t it be possible to have gzipped css in a bytearray for espeasy? It would free almost 4k for other cool stuff. 🙂

TD-er, 2023-01-16:
Not sure if there is any browser out there who may not be able to read GZipped CSS and/or JS.
But I don't see any issues here and if you can free up 4k+, please be my guest :)

chromoxdor, 2023-01-16:
> please be my guest :)

Thank you for the invitation :) I´ll try to be a good guest an wait patiently while I admire you at your work. 😁
And i wouldn't say no if you offer me a cup of tee while watching.

TD-er, 2023-01-16:
> And i wouldn't say no if you offer me a cup of tee while watching.

Can't, I'm busy... as you can see :)

PR #4477 (2023-01-26): [Build] Reduce build size by minifying some inlined javascript code
- Reduce inlined javascript code by minifying, scraping off ca. 500 bytes for most builds.
  Tested to still be working as before 😉.
- Reduce size of embedded css by gzipping that and applying the matching http header, scraping off ~3.6 kB for builds with the css embedded (supported by all browsers).

Resolves #4466
closed 2023-01-26 by TD-er