letscontrolit/ESPEasy #4470: https://github.com/letscontrolit/ESPEasy/actions/runs/3925210875 Crashes

fly74, 2023-01-20:
I've tried https://github.com/letscontrolit/ESPEasy/actions/runs/3925210875

but it results in a reboot loop - ESP_Easy_mega_20221222_collection_E_ESP8266_4M1M work's until this version.

```
10153 : Info   : WIFI : Set WiFi to OFF
10266 : Info   : INIT : Free RAM:26560
10280 : Info   : P144 : Init:   ESP GPIO-pin RX:4 TX:-1
10475 : Info   : INFO : Plugins: 75 [Normal][Collection][Collection_E][No Debug Log] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
10476 : Info   : EVENT: System#Wake
10693 : Info   : WIFI : Set WiFi to STA
10798 : Info   : WIFI : Connecting xxxxxxxxxxxxxxx Ch:1 (-56dBm) WPA2/PSK attempt #0
                                                                                                            --------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (9):
epc1=0x4029e000 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff1576 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3fff2ac0 end: 3fff2de0 offset: 0190
3fff2c50:  3fff2cd0 00000001 3fff2c80 4028fd35
3fff2c60:  00002df0 000005a1 000005a1 4010133b
3fff2c70:  00000000 00000000 3fff2cd0 0000004b
3fff2c80:  3fff1a88 00000001 00000020 4010162a
3fff2c90:  402d03be 4029ec70 016319ac 4029ec70
3fff2ca0:  00ffffff 4029ec70 016319ac 4029ec70
3fff2cb0:  206319ac 5355445a 3fff0d00 4024c2d3
3fff2cc0:  00000009 00000001 3fff0d00 4024ce72
3fff2cd0:  57505300 5355445a 80ff0054 40244574
3fff2ce0:  3fff1a88 3fff0fa4 3fff0d00 4024cfd5
3fff2cf0:  3fff3a54 000d000f 00a196c3 3fff59e4
3fff2d00:  0020002f 00000000 00002743 ffffffc8
3fff2d10:  00000001 30cb9b98 0401b920 00000001
3fff2d20:  40000000 3fff5a00 003e003f 80000000
3fff2d30:  40200000 402074dc 80000020 3fff5a00
3fff2d40:  0000005f 80000001 3fff0d00 4024c134
3fff2d50:  3fff4400 3fff1560 3fff0d00 4024d148
3fff2d60:  3fff1a88 3fff1560 3ffe85a0 40249d59
3fff2d70:  3fff1a88 3fff1560 3ffe85a0 40250073
3fff2d80:  3fff3b00 000b000f 80efeffe 3fff6800
3fff2d90:  0000009f 80efeffe feefeffe feefeffe
3fff2da0:  feefeffe feefeffe feefeffe 3fff2e2c
3fff2db0:  3fffdad0 00000000 3fff2de8 402106e8
3fff2dc0:  feefeffe feefeffe feefeffe 40291753
3fff2dd0:  feefeffe feefeffe 3ffe86e0 401004f9
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
```

TD-er, 2023-01-20:
Just to be complete, it is a build of this PR: Add support for Ikea Vindriktning (#4361) #4366

fly74, 2023-01-20:
> Just to be complete, it is a build of this PR: Add support for Ikea Vindriktning (#4361) #4366

Yes, it is.

TD-er, 2023-01-20:
What did you already do to make it crash?
Just flashed it cleanly on an empty ESP node? Or did it already have a task with this plugin from a previous test build?

fly74, 2023-01-20:
I flashed it over ESP_Easy_mega_20221222_collection_E_ESP8266_4M1M, otherwise you have to open it and flash it serial.

TD-er, 2023-01-20:
Was there already a task configured for this Ikea device?

TD-er, 2023-01-20:
Oh and before I forget... do you use a static IP?
If so, then we already fixed this....

TD-er, 2023-01-20:
I just merged the fix for the static IP issue and rebased the Ikea Vindriktning PR.
GH Actions build will be here: https://github.com/letscontrolit/ESPEasy/actions/runs/3967010026

fly74, 2023-01-20:
> Oh and before I forget... do you use a static IP?
> 
> If so, then we already fixed this....

Yes, I do.

TD-er, 2023-01-20:
Then you could test the latest GH Actions build I linked.

fly74, 2023-01-20:
> Then you could test the latest GH Actions build I linked.

Yes, works as expected. You can close it. Thanks again!

closed 2023-01-20 by TD-er