letscontrolit/ESPEasy #4472: [Feature request] Add support for push* notifications

tonhuisman, 2023-01-21:
As has already been requested several times in the forum, support of some type of push-notification would be nice to have.
Most of these services require the connection to be via HTTPS, and that is challenging, to say the least, and expect the `POST` verb to send messages.

A quick roundup surfaced at least 1 candidate, that (still) supports HTTP connections:
- [PushSafer](https://www.pushsafer.com/) that also has an [Arduino library](https://www.pushsafer.com/en/arduino) with example code

Since we recently have added support for the HTTP `POST` verb, this seems feasible to implement.

chromoxdor, 2023-01-21:
Uhhh that's a good idea. And since the messages can be read with http, nodes could communicate via the internet...
🤩 
https://www.pushsafer.com/en/pushapi#api-read

(i hope reading messages with plain http is a feature and not a bug 😬)

chromoxdor, 2023-01-21:
Just testet it with `SendToHTTP pushsafer.com,80,/api-m?k=xxxxxxxxxxxxxxx&d=61921`
Result:
`38134996: HTTP : SendToHTTP pushsafer.com GET... HTTP code: 200 Received reply: {'status':1,'messages':{'29276719':{'id':'29276719','titl`

Needs a bit of parsing...

chromoxdor, 2023-01-21:
But it is a paid service with just a small amount of free api-calls.
What i don´t get is, that when i send a message via curl the amount of api-calls doesn’t change. And receiving a message via GET doesn’t seem to count too.... I mean its good, because so the whole communication could be free  (at least for sending some numbers from one esp to another)
Edit: Jep, just tested it. When you send a notification to a device (like a phone) the amount of available messages decrease. It probably works for me because i registered a kind of "not existent" device which just stores the messages. And since this is probably an unwanted behavior you can actually ignore my previous messages. I makes no sense to put effort in something that probably will be disappear after somebody read this. :)

chromoxdor, 2023-01-22:
I did a bit of research and i think [ntfy](https://ntfy.sh/) is the better alternative. I offers almost the same functionality including reading messages (e.g. json formatted ) and you are able to easily selfhost it and add then password protection.
What do you think?