letscontrolit/ESPEasy #4473: [P077] un-hardcode gpio number to support Sonoff Pow R3

krzmip, 2023-01-21:
I tried to use ESP Easy with new Sonoff Pow R3 Origin device. Espeasy as such works fine, but it seems is not possible to read voltage/current/power. I am using [ESP_Easy_mega_20221224_energy_ESP32_4M316k Dec 24 2022] build, the CSE7766 device can be created, but shows constantly zeros. 
I have found no clues in the existing support, maybe I have overlooked something (I guess sonoffs are pretty popular devices), so started to dig.
I found that the CSE7766 is connected to different pins than in previous Pow R2 (according to https://templates.blakadder.com/sonoff_POWR316.html rx pin is currently gpio16), while the plugin P077 is not configurable, there is no gpio selection in the interface and the source code comment in _P077_CSE7766.ino says "No pins selectable, all hard coded".
Looked like relatively simple change I can make for own use or submit a pull request even without detailed knowledge of the project. However, I have not found exact place where this is hardcoded. I appreciate you are busy (there is a lot of open issues), so I would appreciate some hints, I can then try to code, test and hopefully make a pull request (if it works).

TD-er, 2023-01-21:
Yep, this still has to be adapted.
So far I only added support for the display on the elite THR316D and POWR316D (and the 20 Amp versions)
I still need to adapt the CSE7766 plugin for this.

tonhuisman, 2023-02-11:
@krzmip Can you test this [Github Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/4151064881), once it's finished?
You'll need the ESP32 `Energy` build, that also has the plugin supporting the Display of the Elite models.

krzmip, 2023-02-11:
Boots normally, shows some voltage but no current, power, pulses. will wait and try updated url when finished

tonhuisman, 2023-02-11:
I don't expect it to have been actually tested on ESP32 before, maybe it's related to that. I don't own a Sonoff R3, so can't verify myself... But TD-er owns (at least) one, so he'll probably be able to test 😄

krzmip, 2023-02-11:
OK, better now. All four variables are non-zero when current flows, voltage is reasonable. Power and current are off but this can be calibrated I guess.
[image]

tonhuisman, 2023-02-11:
Calibration should start with the Voltage, using a digital multimeter, by calculating and adjusting the `U Ref` value. 242V seems a bit high.
Then calibration of power and current can be done by connecting a known, resistive, load (optionally comparing to another calibrated power meter), and adjusting the `I Ref` and `P Ref` values.

krzmip, 2023-02-11:
Yes, that is why I plugged a 250W bulb. The meter shows 239V, so pretty close. Anyway, seems the solution works, thank you.

tonhuisman, 2023-02-11:
Please re-open, for others to easily find it, as this issue will be automatically closed by Github once the PR is merged.

krzmip, 2023-02-11:
ok

krzmip, 2023-02-11:
One remark after the calibration: seems that range of I_ref is too small, I would need an integer of >40k while only 32k is supported. Not a major inconvenience as this can be adjusted by rules before publishing, but maybe can be easily fixed (maybe if negative values make no sense it would be sufficient to change the type to unsigned int?)

TD-er, 2023-02-11:
What did you exactly use as settings for the CSE7766 
I don't seem to get any readings from it.

[image]

krzmip, 2023-02-11:
The same up to calibration
[image]

TD-er, 2023-02-11:
Ah, disabling "Use Serial" on the Tools->Advanced page and then setting it to HW Serial0 does indeed work.
Well by measured values are way off, but that I can probably fix :)  (Voltage is reported as 14.13V

krzmip, 2023-02-11:
Yes, it ts strange. After boot shows ca 15v, but after I turn on the relay (gpio,13,1 command) it is correct
[image]
[image]

TD-er, 2023-02-11:
Ah that makes sense....
I was now multiplying the value by a factor of about 15, to get a plausible value.

[image]

With the relais enabled, it does indeed give a fairly accurate value with the defaults.

Very good to know :)
Thanks for simultaneous debugging, this way Ton can fix it and we can play with new hardware :)

PR #4501 (2023-02-13): [P077] Feature: CSE7766 add serial GPIO pin configuration
Resolves #4473
Resolves #4502 
Resolves #3991 
and some Forum requests (f.e.: [1](https://www.letscontrolit.com/forum/viewtopic.php?t=8417), [2](https://www.letscontrolit.com/forum/viewtopic.php?t=9525))

Features:
- Add serial GPIO pin configuration

TODO:
- [x] Test on Sonoff POW R2 (ESP8266), supports both HWSerial0 and SoftwareSerial.
- [x] Add commands to set the calibration values.
- [x] Add documentation.
- [x] Testing Sonoff Origin (POWR3xx) ([Confirmed](https://github.com/letscontrolit/ESPEasy/issues/4473#issuecomment-142676487 [... cut]
closed 2023-02-13 by TD-er