letscontrolit/ESPEasy #4474: Can't connect to wifi hotspot?

Shohreh, 2023-01-21:
Hello,

Using the web browser, I just reflashed my ESP32S2-powered [Lilygo T8](https://www.aliexpress.com/item/1005004074698699.html) with the "ESP_Easy_mega_20221224_normal_ESP32s2_4M316k.factory" firmware.

After rebooting, I can access its web server though my smartphone's wifi connection, but the board seems unable to connect to my wifi hotspot so I can proceed to add a sensor using my home computer.

Any idea what it could be?

Thank you.

Here's the log when I hit the RST button on the board:
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x7c
load:0x4004c000,len:0x7fc
load:0x40050000,len:0x2598
SHA-256 comparison failed:
Calculated: dc1b4e1d4dca8d6ed2ebad4f61cb51b1e3a045f457459e79a147712bfef18c07
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x4004c15c
983 : Info   : WIFI : Set WiFi to STA
7845 : Info   : WIFI : Set WiFi to AP+STA
8156 : Info   : WIFI : Set WiFi to OFF
8361 �Using PSRAM
8374 : Info   : INIT : Free RAM:157788
8375 : Info   : INIT : Cold Boot - Restart Reason: CPU0: Vbat power on reset
8376 : Info   : FS   : Mounting...
8383 : Info   : FS   : Mount successful, used 141815 bytes of 290156
8389 : Info   : CRC  : Settings CRC           ...OK
8394 : Info   : CRC  : SecuritySettings CRC   ...OK 
8398 : Info   : INIT : SPI not enabled
8399 : Info   : WiFi : Best AP candidate: My_great_wifi_router 0E:17:FA:1A:57:50 Ch:11 (-53dBm) WPA/PSK bgn
8500 : Info   : WIFI : Set WiFi to STA
8507 : Info   : WiFi : Event STA Started
8608 : Info   : WiFi : Start network scan all channels
15227 : Info   : WiFi : Scan finished, found: 13
15229 : Info   : WiFi : Best AP candidate: My_great_wifi_router 0E:17:FA:1A:57:50 Ch:11 (-51dBm) WPA/PSK bgn
15230 : Info   : WiFi : Added known candidate, try to connect
15231 : Info   : WiFi : WifiDisconnect()
15333 : Info   : WIFI : Disconnected! Reason: '(1) Unspecified'
15334 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
15434 : Info   : WIFI : Set WiFi to OFF
15549 : Info   : INIT : Free RAM:151776
15601 : Info   : INFO : Plugins: 47 [Normal] (ESP32 SDK v4.4.3)
15702 : Info   : WIFI : Set WiFi to STA
15810 : Info   : WiFi : Best AP candidate: My_great_wifi_router 0E:17:FA:1A:57:50 Ch:11 (-51dBm) WPA/PSK bgn
15811 : Info   : WIFI : Connecting My_great_wifi_router 0E:17:FA:1A:57:50 Ch:11 (-51dBm) WPA/PSK bgn attempt #0
15819 : Info   : Webserver: start
17193 : Info   : WD   : Uptime 0 ConnectFailures 0 FreeMem 125308 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
18240 : Info   : WiFi : Event STA Stopped
18243 : Info   : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2420 ms
18244 : Info   : WIFI : Arduino wifi status: WL_NO_SHIELD 255 ESPeasy internal wifi status: DISCONNECTED
18346 : Info   : WIFI : Connecting My_great_wifi_router 0E:17:FA:1A:57:50 Ch:11 (-51dBm) WPA/PSK bgn attempt #1
18353 : Info   : WiFi : Event STA Started
38715 : Info   : WIFI : Set WiFi to AP+STA
38717 : Info   : WiFi : Event AP Started
40575 : Info   : WiFi : Event AP Stopped
40576 : Info   : WiFi : Event AP Started
40579 : Info   : WIFI : AP Mode ssid will be ESP-Easy with address 192.168.4.1
47170 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 123604 WiFiStatus WL_NO_SSID_AVAIL 1 ESPeasy internal wifi status: DISCONNECTED
58349 : Info   : WiFi : Best AP candidate: My_great_wifi_router 0E:17:FA:1A:57:50 Ch:11 (-51dBm) WPA/PSK bgn
58351 : Info   : WIFI : Connecting My_great_wifi_router 0E:17:FA:1A:57:50 Ch:11 (-51dBm) WPA/PSK bgn attempt #2
58356 : Info   : WIFI : Arduino wifi status: WL_NO_SSID_AVAIL 1 ESPeasy internal wifi status: DISCONNECTED
77170 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 123472 WiFiStatus WL_NO_SSID_AVAIL 1 ESPeasy internal wifi status: DISCONNECTED
98357 : Info   : WIFI : Connecting My_great_wifi_router 0E:17:FA:1A:57:50 Ch:11 (-51dBm) WPA/PSK bgn attempt #3
98363 : Info   : WIFI : Arduino wifi status: WL_NO_SSID_AVAIL 1 ESPeasy internal wifi status: DISCONNECTED
107165 : Info   : WD   : Uptime 2 ConnectFailures 0 FreeMem 123604 WiFiStatus WL_NO_SSID_AVAIL 1 ESPeasy internal wifi status: DISCONNECTED
137164 : Info   : WD   : Uptime 2 ConnectFailures 0 FreeMem 123604 WiFiStatus WL_NO_SSID_AVAIL 1 ESPeasy internal wifi status: DISCONNECTED
138367 : Info   : WiFi : Start network scan all channels
145417 : Info   : WiFi : Scan finished, found: 20
145419 : Info   : WiFi : Best AP candidate: My_great_wifi_router 0E:17:FA:1A:57:50 Ch:11 (-49dBm) WPA/PSK bgn
145421 : Info   : WiFi : Added known candidate, try to connect
145422 : Info   : WIFI : Arduino wifi status: WL_NO_SSID_AVAIL 1 ESPeasy internal wifi status: DISCONNECTED
167162 : Info   : WD   : Uptime 3 ConnectFailures 0 FreeMem 123076 WiFiStatus WL_NO_SSID_AVAIL 1 ESPeasy internal wifi status: DISCONNECTED
```

tonhuisman, 2023-01-21:
Do you have a mesh-type network router? As especially the Asus mesh-routers seem to cause issues like this, that can be avoided by changing a setting, something to with 'smart' connect. Have to find the link to the exact setting (not using Asus myself, and don't have the link available...)

TD-er, 2023-01-21:
Have you tried power cycling the ESP board after flashing?
Is it possible your wifi password only contains 0...9, A...F?
Somewhere I read that there were some issues with some specific chip revision not being able to connect when it only contained hex characters. But this was mainly a specific revision of ESP32, no mention about the -S2.
But perhaps worth to try?

Also you need at least 8 characters in the wifi password (found out the hard way once myself)

Shohreh, 2023-01-21:
It's not a mesh router, just a plain, straight connection to an ADSL wifi + router modem.

Yes, the password is eight characters long and only contains alphanumeric characters (lowercase alpha and some numbers.)

TD-er, 2023-05-11:
Yesterday I merged support for ESP32-S3/C3 and also updated the WiFi behavior of ESP32 classic and ESP32-S2.

I have noticed some ESP32-S2 and ESP32-C3 boards have really really poor designed antennas.
Typically those purple "Wemos mini" ones with PCB antenna.
However, when using a lower TX power, it may work fine-ish.
So by default the settings for all ESP32-xx boards is now to have the TX power changed automatically based on the strength of the AP.

Please let me know if this is actually an improvement or not for your board(s).

See [here](https://github.com/letscontrolit/ESPEasy/actions/runs/4941358646) for a test build with this latest code.

closed 2023-09-06 by TD-er