letscontrolit/ESPEasy #4479: ESP32 Access point does not work

popaserge, 2023-01-24:
I tried the web flash tool on a Lilygo ESP32 4Mo and on a Wemos Mini32 V1.0.0. flashing ESP_Easy_mega_20221224_normal_ESP32_4M316k.factory
The Esp32 are well flashed and I can see the log of the operations.
But the AP does not work. As soon as it is set, it is stopped with errors.
Thus it is not possible to set an access point candidate with SSID and password to allow the ESP to connect to a network.
I tried to use WifiAPMode serial command, sometimes the AP was visible but desappeared rapidly without I have time to connect.
I could get arround this by sending some configuration data by serial commands (WifiSSID, WifiKey,Reboot)
After that, all was OK and the ESP web site could be accessed on the local network.
Configuring the access point by the WifiAPmode command, the AP is visible but the connection to it very painfull, and changing the menu choice is very long or impossible.
Hereafter the log.

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12588
ho 0 tail 12 room 4
load:0x40080400,len:2916
entry 0x400805c4
�U122 : Info   : WIFI : Set WiFi to STA
6447 : Info   : WIFI : Set WiFi to AP+STA
6756 : Info   : WIFI : Set WiFi to OFF
8869 : Info   : INIT : Free RAM:255340
8870 : Info   : INIT : Soft Reboot #1 Last Action before Reboot: Intended Reboot: ResetFactory Last systime: 21 - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
8871 : Info   : FS   : Mounting...
8895 : Info   : FS   : Mount successful, used 141815 bytes of 290156
8913 : Info   : CRC  : Settings CRC           ...OK
8919 : Info   : CRC  : SecuritySettings CRC   ...OK 
8925 : Info   : INIT : SPI not enabled
9027 : Info   : WIFI : Set WiFi to STA
9039 : Info   : WiFi : Event STA Started
9140 : Info   : WiFi : Start network scan all channels
15154 : Info   : WiFi : Scan finished, found: 4
15156 : Info   : WiFi : Start network scan all channels
21168 : Info   : WiFi : Scan finished, found: 4
21271 : Info   : WIFI : Set WiFi to AP+STA
21274 : Info   : WiFi : Event AP Started
21835 : Info   : WiFi : Event AP Stopped
21835 : Info   : WiFi : Event AP Started
21836 : Info   : WIFI : AP Mode ssid will be ESP-Easy with address 192.168.4.1
21838 : Info   : WiFi : Start network scan all channels
28276 : Info   : WiFi : Scan finished, found: 4
28278 : Info   : WiFi : WifiDisconnect()
28382 : Info   : WIFI : Disconnected! Reason: '(1) Unspecified'
28383 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
28484 : Info   : WIFI : Set WiFi to OFF
29451 : Info   : INIT : Free RAM:248864
29519 : Info   : INFO : Plugins: 47 [Normal] (ESP32 SDK v4.4.3)
29621 : Info   : WIFI : Set WiFi to STA
29632 : Info   : WiFi : Event STA Started
29733 : Info   : WiFi : Start network scan all channels
35747 : Info   : WiFi : Scan finished, found: 4
35849 : Info   : WIFI : Set WiFi to AP+STA
35852 : Info   : WiFi : Event AP Started
36412 : Info   : WiFi : Event AP Stopped
36413 : Info   : WiFi : Event AP Started
36414 : Info   : WIFI : AP Mode ssid will be ESP-Easy with address 192.168.4.1
36416 : Error  : WIFI : No valid wifi settings
36417 : Info   : WiFi : Start network scan all channels
42856 : Info   : WiFi : Scan finished, found: 4
42859 : Info   : WiFi : Start network scan all channels
49498 : Info   : WiFi : Scan finished, found: 5
49500 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
49505 : Info   : Webserver: start
50852 : Info   : WD   : Uptime 0 ConnectFailures 0 FreeMem 211892 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
80849 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 211876 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
86610 : Info   : AP Mode: Client connected: DE:E7:78:5A:17:C5 Connected devices: 1
87573 : Error  : UNKNOWN WIFI/ETH EVENT: 14
98380 : Info   : AP Mode: Client disconnected: DE:E7:78:5A:17:C5 Connected devices: 0
110852 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 208216 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
129502 : Info   : WiFi : Start network scan all channels
135941 : Info   : WiFi : Scan finished, found: 4
135943 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
140846 : Info   : WD   : Uptime 2 ConnectFailures 0 FreeMem 209540 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
170847 : Info   : WD   : Uptime 2 ConnectFailures 0 FreeMem 209540 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
200847 : Info   : WD   : Uptime 3 ConnectFailures 0 FreeMem 209540 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
215950 : Info   : WiFi : Start network scan all channels
222590 : Info   : WiFi : Scan finished, found: 4
222592 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
230853 : Info   : WD   : Uptime 3 ConnectFailures 0 FreeMem 209536 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED

TD-er, 2023-01-25:
I have seen this too, right after flashing.
The remedy was to either power cycle, or press the reset button on the boards that have it.
There is something strange with the state of the WiFi radio right after flashing the ESP32.
I have the impression it only happens on relatively new builds, since we updates the ESP32 SDK to use IDF 4.4.

popaserge, 2023-01-26:
Could it be because of the type of Wifi that the ESP32 starts.
According to the info given by my router, all the ESP8266 work in Wifi g and the Esp32 with the Wifi n (the 2 ESP-Easy module on the joined screenshot)
[image]

TD-er, 2023-01-26:
I honestly have no idea why the WiFi on ESP32 right after flashing seems to be acting strange.
What I have noticed is that after a reset or power cycle, it does seem to work fine.

I have seen some issues a while ago stating people were not able to let the ESP32 connect to an AP on every even nr of attempts.

On ESP8266 I have set the WiFi to try connecting using 802.11g first and if that fails, retry with 802.11n.
The reason for this is that there are some access points (mainly those supporting WiFi mesh, regardless whether it is enabled or not) which may act strange on IoT devices (like the ESP's) when these connect using 802.11n.

However, I have not been able to reliably set the 802.11g mode on ESP32, so this is not enforced the same on those as it is on ESP8266.

It is possible this is somehow related, but then it doesn't explain why after a reboot the ESP32 does behave well.

TL;DR
I have no clue!

hrikken, 2024-01-06:
I have the same problem but this time with the ESP32-S3 16/8 M. I tried a few flash-scenarios (erase / flash with 16M etc) but no success. The I flashed it with the build version Build: ESP_Easy_mega_20230920_max_ESP32s3_16M8M_LittleFS_OPI_PSRAM_CDC Sep 20 2023. Now the AP works again. Maybe too fast to jump to conclusions but IMO there must me something wrong with the new builds.

TD-er, 2024-01-06:
@hrikken 
If it is that you can't connect to the ESP's access point using your phone, then that's being worked on: https://github.com/letscontrolit/ESPEasy/issues/4931

N.B. you can set the WiFi credentials manually via the serial port console:
```
wifissid,MyWiFiSSID
wifikey,MyWiFiSecretPassword
save
```
Then reboot just to be sure.

hrikken, 2024-01-06:
Ok thanks for the quick response. !!!