letscontrolit/ESPEasy #4484: [P083] NeoPixel (Basic) Color/Brightness issue

chromoxdor, 2023-01-26:
FYI:
After adding the HSV slider to easyfetch i tested it today.
I realized, that de color is always a little bit off (especially in the yellow/orange section).
So i compared it with Tasmota since i thought my sliders are somehow not correct.
What i found out:
My Slider look and work the same as the ones in Tasmota
I tested the same neoPixel with Tasmota as well as ESPeasy with the same values hsv(56,100,100).
In Tasmota it creates yellow as also indicated by the slider correctly,  where it is more greenish in ESPeasy.
Also even the LED set to maximum brightness and confirmed with `NeoPixelBright,255` it is way more dimmer than in Tasmota (maybe a safety measure?)

I guess a different formula was used by Tasmota?... in ESPeasy the already narrow spectrum of yellow/orange is even more narrow.

chromoxdor, 2023-01-26:
The brightness issue appears actually only when mixing the rgb values. The base colors are as bright as in Tasmota.

tonhuisman, 2023-01-26:
P038, NeoPixel (Basic), or any of the NeoPixel plugins, has to convert the HSV colors to their RGB counterparts as required by the NeoPixel library, and uses the HSV2RGB conversion available in ESPEasy. For Tasmota different code is used (I checked), but I don't have any background in that (yet?) to determine what the best/closest way to do that. Might need some investigation.

tonhuisman, 2023-01-26:
@chromoxdor Can you compare colors with P042 (NeoPixel Candle)? As that has, until PR #4471 is merged, a different algorithm for converting HSV to RGB, so it might end up with different colors.

chromoxdor, 2023-01-26:
> @chromoxdor Can you compare colors with P042 (NeoPixel Candle)? As that has, until PR https://github.com/letscontrolit/ESPEasy/pull/4471 is merged, a different algorithm for converting HSV to RGB, so it might end up with different colors.

I can do that tomorrow.
I just learned a bit about the different color spaces..
Here a very short overview: https://stackoverflow.com/questions/20853527/hsi-and-hsv-color-space
We use HSI disguised as HSV and Tasmota use actually HSV...

chromoxdor, 2023-01-26:
> Can you compare colors with P042 (NeoPixel Candle)? 

Just realized, that tomorrow i have no time for that.
It is much closer to Tasmota. Almost the same. 
Subjectively i find Tasmotas algorithm the best. 
But this is just the opinion of one guy with one pixel. 🙂

chromoxdor, 2023-01-26:
And there is a difference between neopixel(basic) und busfx+candle
The latter both are as bright as with tasmota. Only neopixel basic seems to be affected, wich is odd since basic and busfx share the same algorithm, right?

tonhuisman, 2023-01-26:
BusFX may have a different algorithm, probably the same or very close to Tasmota, haven't checked the details yet. This is going to take a bit of time to get right, so don't hurry/pressure yourself too much 😃

fly74, 2023-01-31:
> Also even the LED set to maximum brightness and confirmed with `NeoPixelBright,255`

One thing is, if the max brightness in the device is set lower than 255 the `NeoPixelBright,255` will not overrule that. The max color range is only with full brightness reachable.

chromoxdor, 2023-02-01:
> One thing is, if the max brightness in the device is set lower than 255 the NeoPixelBright,255 will not overrule that. The max color range is only with full brightness reachable.

Thats what i actually meant with that:

> Also even the LED set to maximum brightness

tonhuisman, 2023-11-04:
@chromoxdor We have replaced the `Adafruit NeoPixel` library by the `NeoPixelBus by Makuna` library (by implementing a wrapper-lib), so save some space, and because this NeoPixelBus lib has smarter hardware support on newer ESP32 chips 😎.
This library may have different color mappings then you have seen before, so maybe you can test the latest `mega` code to see if it now better matches your expectations?

fly74, 2023-11-04:
> on newer ESP32 chips

What means newer?

tonhuisman, 2023-11-04:
The later released types, supported by ESPEasy, ESP32-S2, -C3 and -S3

tonhuisman, 2023-11-20:
@chromoxdor Any feedback on these changes?

chromoxdor, 2023-11-23:
Hey @tonhuisman ,

sorry, but i was "out of order" for a couple of days....😳
I didn´t thoroughly tested it but it seems to work well 🙂
Right now i have to finish two projects that piled up and then i will do a comparison test.
But anyhow, thanks for the effort!
One thing broke btw. : 
`NeoPixelHSV` doesn´t work as expected anymore. But that was already before your latest change.
In the December 22 release it still works and in September 23 it already doesn´t
But i couldn´t find the related PR for that. Could you have a look at that?

TD-er, 2023-11-23:
Might be some code de-duplication I did which causes this mess-up?