letscontrolit/ESPEasy #4485: High Sysload on wifi lost ESP8266/ESP32 with releases 20220809 / 20221105 / 20221224 / Git from 23.01.22

tomcatxx, 2023-01-27:
I'm using an ESP32 to controll the flow of a waterpump over pwm in combination with a flowmeter (pulsecounter).
The process is done by a PID controller rule.
Everything works fine if the controller is connected to the wifi. 
If wifi get lost/turned off the sysload rises to 100 and the process stops working properly. 

If tested what will hapen on clean flashed devices ESP8266 Nodemcu and ESP32-WROOM-32D with above mentioned builds.
Here some logs:

`643867 : Info   : WIFI : Connecting ACCESSP 26:5A:5A:11:01:8F Ch:6 (-60dBm) WPA2/PSK bgn attempt #9
643875 : Info   : WIFI : Arduino wifi status: WL_CONNECT_FAILED 4 ESPeasy internal wifi status: DISCONNECTED
645467 : Info   : WD   : Uptime 11 ConnectFailures 0 FreeMem 210812 WiFiStatus WL_CONNECT_FAILED 4 ESPeasy internal wifi status: DISCONNECTED
646362 : Debug  : LoopStats: shortestLoop: 72 longestLoop: 5838038 avgLoopDuration: 15809.73 loopCounterMax: 416666 loopCounterLast: 1896
646363 : Debug  : Scheduler stats: (called/tasks/max_length/idle%) 1896/1580/7/13.71
675467 : Info   : WD   : Uptime 11 ConnectFailures 0 FreeMem 210812 WiFiStatus WL_CONNECT_FAILED 4 ESPeasy internal wifi status: DISCONNECTED
676362 : Debug  : LoopStats: shortestLoop: 72 longestLoop: 5838038 avgLoopDuration: 15795.73 loopCounterMax: 416666 loopCounterLast: 1898
676363 : Debug  : Scheduler stats: (called/tasks/max_length/idle%) 1898/1580/7/13.76
683871 : Info   : WIFI : Connecting ACCESSP 26:5A:4C:1C:99:0D Ch:6 (-71dBm) WPA2/PSK bgn attempt #10
683879 : Info   : WIFI : Arduino wifi status: WL_CONNECT_FAILED 4 ESPeasy internal wifi status: DISCONNECTED
683942 : Debug  : WIFI : Entering processConnect()
683944 : Info   : WIFI : Connected! AP: ACCESSP (26:5A:4C:1C:99:0D) Ch: 6 Duration: 64 ms
683965 : Debug  : WIFI : Entering processGotIP()
683966 : Info   : WIFI : DHCP IP: 192.168.1.20 (ESP-Easy) GW: 192.168.1.1 SN: 255.255.255.0   duration: 27 ms
683980 : Debug  : WiFi : WiFi services initialized
705454 : Info   : WD   : Uptime 12 ConnectFailures 0 FreeMem 210024 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
706343 : Debug  : LoopStats: shortestLoop: 51 longestLoop: 5838038 avgLoopDuration: 140.49 loopCounterMax: 588235 loopCounterLast: 211457
706345 : Debug  : Scheduler stats: (called/tasks/max_length/idle%) 211457/1880/7/60.63
735454 : Info   : WD   : Uptime 12 ConnectFailures 0 FreeMem 210024 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
736343 : Debug  : LoopStats: shortestLoop: 51 longestLoop: 5838038 avgLoopDuration: 104.59 loopCounterMax: 588235 loopCounterLast: 283073
736345 : Debug  : Scheduler stats: (called/tasks/max_length/idle%) 283073/1982/7/77.09
>`

So even on clean flashed normal builds this high load problem exist.

Any ideas how to get rid of this problems?

TD-er, 2023-01-27:
Can you enable "Collect Timing Statistics" on the tools->Advanced page?
Then after a few minutes, please load the "Timing Stats" page (link on the Tools page)
N.B. Each time this stats page is reloaded, the stats will be cleared.

If you could share the found stats,  we can try to narrow it down What may cause your high load.

Also check whether you've enabled "Eco mode" and if you have, please try also with this mode disabled.

tomcatxx, 2023-01-27:
TY for the fast reply :-)

Clean flashed 20221224 normal from https://td-er.nl/ESPEasy/
on ESP32-Wroom-32D.
Only changes:
WIFI connected 
Timing Stats enabled.
Client IP block level: Allow all
Stats after 5 mins:

Connected to wifi and idling:
Serial logs:
`1786499 : Info   : WD   : Uptime 29 ConnectFailures 0 FreeMem 209260 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
1787388 : Debug  : LoopStats: shortestLoop: 38 longestLoop: 7293151 avgLoopDuration: 65.58 loopCounterMax: 789473 loopCounterLast: 449022
1787390 : Debug  : Scheduler stats: (called/tasks/max_length/idle%) 449022/2131/6/92.94`
[image]

Not connected to wifi and idling:
`345317 : Info   : WD   : Uptime 6 ConnectFailures 0 FreeMem 206196 WiFiStatus WL_NO_SSID_AVAIL 1 ESPeasy internal wifi status: DISCONNECTED
346191 : Debug  : LoopStats: shortestLoop: 28 longestLoop: 6265090 avgLoopDuration: 13532.56 loopCounterMax: 1071428 loopCounterLast: 2216
346193 : Debug  : Scheduler stats: (called/tasks/max_length/idle%) 2216/1826/6/15.59`
[image]

-----

The Pump Controller:
Connected to wifi and idling:
[image]

The Pump Controllern not connected to wifi and idling:
[image]

tonhuisman, 2023-01-27:
Is there anything special in your rules? You should not use any `delay` command there, to start with, as at least _some_ time is spent there, but the `WiFi.isConnected() (fail)` is the obvious candidate for further investigation.

tomcatxx, 2023-01-27:
No noting special and no delays. 
I also simplified the rules as much as possible for testing and disabled all not directly nessesary tasks/sensors.
It reduce the sysload a bit but as soon as wifi goes offline sysload goes up to 100 and its not longer controlling the pump propperly.

TD-er, 2023-01-27:
So it looks like there is some reconnect attempt causing huge delays.

I think, given the affected builds, this might be related to the more dynamic timeout I made for the controllers.
I did implement it around the end of the summer I think.

I will have a look.

tomcatxx, 2023-01-27:
Hm I tested ESPEasy_ESP32_mega-20210802
`
393287 : Info   : WD   : Uptime 7 ConnectFailures 0 FreeMem 222128 WiFiStatus WL_CONNECT_FAILED 4 ESPeasy internal wifi status: DISCONNECTED
394190 : Debug  : LoopStats: shortestLoop: 1505 longestLoop: 2919059 avgLoopDuration: 13626.29 loopCounterMax: 19933 loopCounterLast: 2201
394191 : Debug  : Scheduler stats: (called/tasks/max_length/idle%) 2201/1650/7/20.17`
Also around 80% Sysload.

[image]

TD-er, 2023-01-27:
While you're at it, can you try one of the latest GH Action builds?
For example this one: https://github.com/letscontrolit/ESPEasy/actions/runs/4019311623
That's the last one of the current 'mega' branch.

tomcatxx, 2023-01-28:
Still 85% sysload after wifi is off.
`495317 : Info   : WD   : Uptime 8 ConnectFailures 0 FreeMem 211808 WiFiStatus WL_CONNECT_FAILED 4 ESPeasy internal wifi status: DISCONNECTED
496209 : Debug  : LoopStats: shortestLoop: 40 longestLoop: 6273039 avgLoopDuration: 15066.17 loopCounterMax: 750000 loopCounterLast: 1989
496210 : Debug  : Scheduler stats: (called/tasks/max_length/idle%) 1989/1632/7/15.46`
[image]

TD-er, 2023-01-28:
OK, so I need to look at the reconnect related code.

One thing I also see from your timing stats is that we can optimize P023 a bit more as it is loading the CustomTaskSettings every second while this does take quite a lot of time and can easily be cached.

tomcatxx, 2023-01-31:
Ok. If you need furter testings or informations just let me know :)

TD-er, 2023-01-31:
When I got something to test, I will let you know :)

TD-er, 2023-03-29:
Could you test with one of the most recent GitHub Actions builds?
For example this one: https://github.com/letscontrolit/ESPEasy/actions/runs/4540120477

TD-er, 2023-03-29:
There is always a high "load" when scanning, because performing a WiFi scan is blocking.

I have tried to make it async, but wasn't pleased with the results yet.

I can give it another try, by simply scanning per channel as we can do that now for ESP32 in the latest SDK code.
But it will still take upto 300 msec per channel, for 12 (13?) channels.

tomcatxx, 2023-03-30:
Yes but why is the Sysload that high?

`412736 : Debug More : UDP  : 01:3B:F1:A1:E9:3F,192.168.10.17,0
518727 : Info   : WD   : Uptime 9 ConnectFailures 0 FreeMem 216608 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
519615 : Debug  : LoopStats: shortestLoop: 57 longestLoop: 628675 avgLoopDuration: 105.83 loopCounterMax: 526315 loopCounterLast: 278689
519617 : Debug  : Scheduler stats: (called/tasks/max_length/idle%) 278689/2042/7/80.44
521243 : Error  : WiFi : Connecting not possible, unprocessed WiFi events:  disconn
521246 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: Conn. IP Init
521248 : Info   : WiFi : Event STA Stopped
521251 : Debug  : WIFI : Entering processDisconnect()
521251 : Info   : WIFI : Disconnected! Reason: '(4) Assoc expire' Connected for 8 m 29 s
521363 : Info   : WiFi : Event STA Started
521370 : Info   : WiFi : Start network scan all channels
527184 : Info   : WiFi : Scan finished, found: 5
527187 : Debug  : WiFi : Scan result: Test-1 26:5A:4C:2C:99:0D Ch:6 (-71dBm) WPA2/PSK bgn
527188 : Debug  : WiFi : Scan result: Test-2 26:5A:4C:1C:99:0D Ch:6 (-71dBm) WPA2/PSK bgn
527188 : Debug  : WiFi : Scan result: Test-3 26:1B:FF:20:01:8C Ch:6 (-71dBm) WPA2/PSK bgn
527189 : Debug  : WiFi : Scan result: #Hidden# 26:1B:FF:20:04:8C Ch:6 (-69dBm) WPA2/PSK bgn
527190 : Debug  : WiFi : Scan result: #Hidden# 26:1B:FF:20:05:8C Ch:6 (-71dBm) WPA2/PSK bgn
527190 : Info   : WiFi : Best AP candidate: Test-2 26:5A:4C:1C:99:0D Ch:6 (-71dBm) WPA2/PSK bgn
527191 : Info   : WiFi : Added known candidate, try to connect
527192 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
527192 : Debug  : WIFI : Disconnected: WiFi.status() = DISCONNECTED RSSI: 0 status: WL_DISCONNECTED 6
548745 : Info   : WD   : Uptime 9 ConnectFailures 0 FreeMem 218260 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
549622 : Debug  : LoopStats: shortestLoop: 57 longestLoop: 5955068 avgLoopDuration: 1828.98 loopCounterMax: 526315 loopCounterLast: 16384
549624 : Debug  : Scheduler stats: (called/tasks/max_length/idle%) 16382/1333/7/14.30
578731 : Info   : WD   : Uptime 10 ConnectFailures 0 FreeMem 218272 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
579622 : Debug  : LoopStats: shortestLoop: 57 longestLoop: 5955068 avgLoopDuration: 14843.08 loopCounterMax: 526315 loopCounterLast: 2020
579624 : Debug  : Scheduler stats: (called/tasks/max_length/idle%) 2020/1673/7/13.87
607199 : Info   : WIFI : Connecting Test-2 26:5A:4C:1C:99:0D Ch:6 (-71dBm) WPA2/PSK bgn attempt #0
607208 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
608769 : Info   : WD   : Uptime 10 ConnectFailures 0 FreeMem 218264 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
609651 : Debug  : LoopStats: shortestLoop: 57 longestLoop: 5955068 avgLoopDuration: 15346.61 loopCounterMax: 526315 loopCounterLast: 1955
609653 : Debug  : Scheduler stats: (called/tasks/max_length/idle%) 1951/1633/7/12.70
613046 : Debug  : WIFI : Disconnected: WiFi.status() = DISCONNECTED RSSI: 0 status: WL_NO_SSID_AVAIL 1
638731 : Info   : WD   : Uptime 11 ConnectFailures 0 FreeMem 218320 WiFiStatus WL_NO_SSID_AVAIL 1 ESPeasy internal wifi status: DISCONNECTED
639628 : Debug  : LoopStats: shortestLoop: 57 longestLoop: 5955068 avgLoopDuration: 14394.87 loopCounterMax: 526315 loopCounterLast: 2082
639630 : Debug  : Scheduler stats: (called/tasks/max_length/idle%) 2075/1628/7/16.51
641246 : Info   : WiFi : Event Lost IP
641258 : Debug  : WIFI : Disconnected: WiFi.status() = DISCONNECTED RSSI: 0 status: WL_IDLE_STATUS 0
647205 : Info   : WIFI : Connecting Test-2 26:5A:4C:1C:99:0D Ch:6 (-71dBm) WPA2/PSK bgn attempt #1
647213 : Info   : WIFI : Arduino wifi status: WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
668727 : Info   : WD   : Uptime 11 ConnectFailures 0 FreeMem 218356 WiFiStatus WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
669622 : Debug  : LoopStats: shortestLoop: 57 longestLoop: 5955068 avgLoopDuration: 14146.93 loopCounterMax: 526315 loopCounterLast: 2119
669624 : Debug  : Scheduler stats: (called/tasks/max_length/idle%) 2118/1669/7/17.92
687222 : Info   : WiFi : Start network scan all channels
693036 : Info   : WiFi : Scan finished, found: 10
693038 : Debug  : WiFi : Scan result: Test-1 26:1B:FF:20:01:8F Ch:6 (-61dBm) WPA2/PSK bgn
693039 : Debug  : WiFi : Scan result: Test-2 26:1B:FF:20:01:8A Ch:6 (-62dBm) WPA2/PSK bgn
693040 : Debug  : WiFi : Scan result: Test-3 26:1B:FF:20:01:8B Ch:6 (-63dBm) WPA2/PSK bgn
693041 : Debug  : WiFi : Scan result: Test-3 26:1B:FF:20:01:8C Ch:6 (-68dBm) WPA2/PSK bgn
693041 : Debug  : WiFi : Scan result: Test-2 26:5A:4C:1C:99:0D Ch:6 (-68dBm) WPA2/PSK bgn
693042 : Debug  : WiFi : Scan result: Test-1 26:5A:4C:2C:99:0D Ch:6 (-69dBm) WPA2/PSK bgn
693043 : Debug  : WiFi : Scan result: #Hidden# 26:1B:FF:20:02:8C Ch:6 (-59dBm) WPA2/PSK bgn
693043 : Debug  : WiFi : Scan result: #Hidden# 26:1B:FF:20:03:8C Ch:6 (-62dBm) WPA2/PSK bgn
693044 : Debug  : WiFi : Scan result: #Hidden# 26:1B:FF:20:04:8C Ch:6 (-67dBm) WPA2/PSK bgn
693044 : Debug  : WiFi : Scan result: #Hidden# 26:1B:FF:20:05:8C Ch:6 (-68dBm) WPA2/PSK bgn
693045 : Info   : WiFi : Best AP candidate: Test-2 26:1B:FF:20:01:8A Ch:6 (-62dBm) WPA2/PSK bgn
693046 : Info   : WiFi : Added known candidate, try to connect
[... cut ...]
[ 62292][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-76) UNKNOWN ERROR CODE (004C)
WiFi up, MQTT down: start MQTT
Executiontime : 649482 uS. CPU Load: 6%
WiFi up, MQTT up: finish MQTT configuration
Executiontime : 139 uS. CPU Load: 0%
Executiontime : 131 uS. CPU Load: 0%
Executiontime : 130 uS. CPU Load: 0%
Executiontime : 130 uS. CPU Load: 0%
Executiontime : 130 uS. CPU Load: 0%
Executiontime : 138 uS. CPU Load: 0%
Executiontime : 138 uS. CPU Load: 0%
```
Non blocking to my opinion.
Maybe it help you.