letscontrolit/ESPEasy #4489: Devices Position - HC-SR04 (2020) i2c

AlexUmnikov, 2023-01-30:
Greetings.
Now on sale there is a Position module - HC-SR04 (2020) that works not only GPIO but also UART and IIC (port 0x57)
In the assembly for ESP8266, I found it only for working with GPIO.
Is there a build to work with HC-SR04 with I2C?

TD-er, 2023-01-30:
Nope, not yet.
But do you have a link to the module?

AlexUmnikov, 2023-01-30:
Here is a discussion of how it works.
https://forum.arduino.ua/viewtopic.php?id=2428
Still it goes like V2
What I understood - everything is the same as in V1 GPIO, only the command is sent and the response is read via I2C

TD-er, 2023-01-30:
And where to buy them? You mentioned they were on sale.
I looked for it at the usual sites, but nowhere they mention anything regarding a "2020" version.

AlexUmnikov, 2023-01-30:
(I don't like ads), bought here
https://aliexpress.ru/item/32477198302.html?spm=a2g2w.orderdetail.0.0.66304aa6c34yNT&sku_id=12000024554661973
If you look at the reviews, there are two options, old and new. The difference is in the availability of places for soldering resistors and the inscription 2020 on the reverse side.

AlexUmnikov, 2023-01-31:
Yes, it is, but how to insert it into espeasy? As I understand it, in Rules there is no way to work with i2c, only with GPIO.
And the datasheet says:
distance = ((byte_h << 16) + (byte_m << 8) +byte_l)/1000
And it's in mm