letscontrolit/ESPEasy #4491: [P044] Parse P1 messages to make the values available in ESPEasy

tonhuisman, 2023-01-31:
To make all values, available in the P1 datagram, available for regular use (like on displays) the datagram should be parsed to split out these values.

With pending PR #4283 the code for handling P1 messages will be merged into P020, so once that's completed, this can be implemented.

JoostDkr, 2024-02-13:
Now that energy rates are going to become dynamic and the "Salderingsregeling" will soon be phased out, it would be nice if the individual data elements became available. So that you can control directly from the espeasy (without an intermediary from, for example, Domoticz).

TD-er, 2024-02-13:
@JoostDkr See [today's news](https://www.nu.nl/politiek/6301267/salderingsregeling-voor-zonnepanelen-wordt-komende-jaren-niet-afgebouwd.html)

About the dynamic tarifs, SolarEdge now has support for negative pricing in their latest firmware and I think others will follow soon.

However this doesn't mean it can be useful to have access to other elements in the parsed messages.