letscontrolit/ESPEasy #4498: [ADC] Add command to set calibration values

TD-er, 2023-02-09:
This can be used to set specific calibration values at boot from a separate rules file.
By splitting up the rules files, one can have a single file specific to a unit and another one which can be used among several units.

Another idea is to allow some plugins/tasks to store typical calibration data to a separate file?

chemmex, 2023-02-09:
In my project (which is air quality monitoring), I write only raw data into timeseries DB, and when the data is retrieved for visualization or other purposes, it is adjusted using calibration coefficients from an inventory DB. So everything is done externally. 
I would be happy to have a possibility to store the calibration within the device in a remote way. In my scenario, I assume that "raw" device is provisioned either by downloading a calibration file, or, most preferrably, via mqtt command. Then it may be stored somewhere in the config area, as there is already such space in the plugin configuration

TD-er, 2023-02-09:
> Edit: oh, I just realized that you are talking just about the ADC. Anyway, why not extend this approach to all sensors which have calibration functionality?

I did think of this, but most of the time, when I start make things "more generic", the idea kinda "explodes" in the amount of work and then gets postponed as I "need to think about it"....

But since you already guessed my mind, I guess I need to make it more generic to start with....

chemmex, 2023-02-16:
Actually, by calibration I meant putting the a and b into the formula field of any device as per simple linear regression:  y = a*x + b, where slope and intercept are calculated externally (using least squares or any other available  methods). I see that what you are going to implement is quite different as you are calculating thise coefficient inside the task

TD-er, 2023-02-17:
Aha, something like a smart interpolation function to be used in rules?
And then not just linear, but also using some curves?

I was planning on making the multi-point interpolation and binning feature of the ADC plugin into some kind of helper/module so we can use it on nearly any plugin.
Is that more like what you were thinking of?

PR #4507 (2023-02-17): [ADC] Add config commands to set calibration
```
config,task,<taskname>,SetCalib,TwoPoint,<ADC1>,<Out1>
config,task,<taskname>,SetCalib,TwoPoint,<ADC1>,<Out1>,<ADC2>,<Out2>
```

This allows for a generic 2nd (or 3rd...) rules file so all units can use the same set of rules and still have unit specific configuration.

For example:
```
On LoadCalibration Do
  if %cpu_id% = 0x1280E0
    config,task,bat,SetCalib,TwoPoint,392,13.9846,571,19.9924
    config,task,batBackup,SetCalib,TwoPoint,1785,10.000
    unit,2
  elseif %cpu_id% = 0x0C17E4
    config,task,bat,SetCalib,TwoPoint,392,13.9846,571,19.99 [... cut]
closed 2023-02-17 by TD-er