letscontrolit/ESPEasy #4499: IR Debug log

RBroks, 2023-02-10:
Hello, is there newer version than mega-20190731, with IR TSOP4838 and log debbug?

tonhuisman, 2023-02-10:
Is there any specific reason you need the `DEBUG` log?
As any required log to effectively/completely use the IR plugins is on `INFO` level, so still included in all builds having IR.

RBroks, 2023-02-10:
Ou that would be nice, but no idea how it can be done...

tonhuisman, 2023-02-10:
You have not answered my question on _why_ you need the debug log, as for normal use of ESPEasy that's not needed at all, only when troubleshooting a specific error situation; and when you are in such situation, you can always come here and ask for help.

tonhuisman, 2023-09-09:
This can be closed as there is no more response.

closed 2023-09-09 by TD-er