letscontrolit/ESPEasy #4502: Sonoff POW Origin

sigalou, 2023-02-11:
Hello to all, do you know if the CSE7766 plugin takes into account the following new device:

Description :
https://sonoff.tech/product/diy-smart-switches/pow-origin/

info :
https://templates.blakadder.com/sonoff_POWR316.html

Plugin CSE7766
https://espeasy.readthedocs.io/en/latest/Plugin/P077.html

The consumption calculation always remains at zero.
I can't get it to work for the consumption measurement part. The other functionalities (buttons, or relays) work well

Thank

tonhuisman, 2023-02-11:
I already responded to your Forum request that I started work on this plugin yesterday, and created a PR an hour ago...

Can you test this [Github Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/4151309179), once it's finished ? (did a bit of housekeeping after development, in the source 😉)

sigalou, 2023-02-11:
THANKS.
Power consumption works well with this new version
I configured like this:
HWSerial0 / GPIO16 / None

TD-er, 2023-02-11:
Yep and for those who come across this issue, you need to have relais1 active to measure voltage as it is apparently connected to the V_out.

sigalou, 2023-02-11:
Yes, all Sonoff POWs work the same.

tonhuisman, 2023-02-11:
Can you re-open this issue, for others searching for this? Once the linked pull request is merged and the issue is _really_ solved, all linked issues will automatically be closed by github 👍

sigalou, 2023-02-11:
ok

PR #4501 (2023-02-13): [P077] Feature: CSE7766 add serial GPIO pin configuration
Resolves #4473
Resolves #4502 
Resolves #3991 
and some Forum requests (f.e.: [1](https://www.letscontrolit.com/forum/viewtopic.php?t=8417), [2](https://www.letscontrolit.com/forum/viewtopic.php?t=9525))

Features:
- Add serial GPIO pin configuration

TODO:
- [x] Test on Sonoff POW R2 (ESP8266), supports both HWSerial0 and SoftwareSerial.
- [x] Add commands to set the calibration values.
- [x] Add documentation.
- [x] Testing Sonoff Origin (POWR3xx) ([Confirmed](https://github.com/letscontrolit/ESPEasy/issues/4473#issuecomment-142676487 [... cut]
closed 2023-02-13 by TD-er