letscontrolit/ESPEasy #4503: Support for Soil moistrure sensor MS-01

Cosik, 2023-02-12:
Hello,

I want to ask if you could add support from Soil moisture sensor support - Sonoff MS-01 [https://sonoff.tech/product/accessories/ms01/](url). It is supported by tasmota, but due to better battery solution in ESPEasy I want to switch to it. It is working on 1-wire bus.

Br,
Cosik.

TD-er, 2023-03-27:
Can you test with this testbuild, when it is ready?
https://github.com/letscontrolit/ESPEasy/actions/runs/4536566710

Cosik, 2023-03-28:
Hi @TD-er ,

Thank you. I will have time at the weekend and I will try to test it.

Cosik, 2023-04-02:
Hi @TD-er ,

I confirm that it is working :) Thanks.

TD-er, 2023-04-02:
Is it also a plausible value?
I thought the curve of the values is rather hard to use and calibrate if needed.

Anyway, thanks for testing.

PR #4574 (2023-04-02): Sonoff Soil moistrure sensor MS-01 #4503
Fixes: #4503

ToDo:

- [x] Test on Sonoff THR316D (GPIO-25 for data, GPIO-27 Output High on Hardware tab)
- [ ] Document
- [ ] Add calibration option ??? (highly non-linear curve)

Not sure whether this curve is the same for all units:
[image]
closed 2023-04-02 by TD-er