letscontrolit/ESPEasy #4510: feature request: new plugin to support TLC5940

tomatos-st, 2023-02-20:
Hi 
Is there a future plan for adding an plugin for TLC5940?
"The TLC5940 is a 16-channel, constant-current sink LED driver. Each channel has an individually adjustable 4096-step grayscale PWM brightness control and a 64-step, constant-current sink (dot correction). The dot correction adjusts the brightness variations between LED channels and other LED drivers. The dot correction data is stored in an integrated EEPROM. Both grayscale control and dot correction are accessible through a serial interface. A single external resistor sets the maximum current value of all 16 channels."

It seems interesting, maybe you should consider adding it to plugin library

TD-er, 2023-02-20:
One of the main reasons why a user may need a plugin for some chip is when it is being used in "cheap China modules".
Do you happen to know (or have) such a module where this chip is used?

tomatos-st, 2023-02-20:
I am planning to build something similar:
http://technika-laika.blogspot.com/2013/10/sterownik-lighting-schodow-na.html
WiFi control would be nice.
The system is also available in robotic shops as a module:
https://botland.com.pl/sterowniki-motor-dc/1856-tlc5940-sterownik-generator-pwm-sparkfun-bob-10616-5904422370503.html
https://www.technobotsonline.com/16-channel-pwm-tlc5940-breakout-board.html

tomatos-st, 2023-02-20:
For me even 4 pcs (64 LED) is enough  ;)

tonhuisman, 2023-02-20:
> For me even 4 pcs (64 LED) is enough ;)

But what about the other questions, for compensating the brightness-differences of the leds? Though you probably won't need that for the stair-lights 😉 

I'm looking for a somewhat more solid use-case, before starting a new plugin, as mentioned project is simple enough to be built stand-alone, using a regular Arduino board. Anyone needing a lot of on/off outputs can already use the [Output - Shift registers (74HC595)](https://espeasy.readthedocs.io/en/latest/Plugin/P126.html#p126-page) P126 plugin, that (theoretically) supports 255 8-output chips cascaded. Though that doesn't support PWM, obviously.

TD-er, 2023-02-20:
Not sure what range and PWM resolution they support, otherwise one might want to use it as a 16 channel servo controller? ;)

tomatos-st, 2023-02-20:
At first, maybe without PWM fine-tuning.
74HC595 is cool (I've used it before) but PWM is even cooler :)
Regarding the servo: 12bit should give enough resolution, only the frequency cannot be too high (50-200Hz) depending on the type of servo

tonhuisman, 2023-02-20:
> one might want to use it as a 16 channel servo controller?

That's actually one of the regular use-cases, and part of the demos I found. And supported via additional functions that f.e. limit the frequency to 50 Hz, and allow to control it by just providing a (servo) pin and angle.

Hm, maybe an ESPEasy-controlled house-robot then lies within the possibilities 🤔 😆

tomatos-st, 2023-02-20:
> Edit: Quite similar to the [PCA9685 IO extender](https://espeasy.readthedocs.io/en/latest/Plugin/P022.html#p022-page).

I'm stupid and blind how could I not see this plugin ????!!!

```
Device Settings:
mode2,<mode>
mode: 0 … 32
```

where can i find a detailed description of this?

tonhuisman, 2023-02-20:
> where can i find a detailed description of this?

That would probably be best directly from the manufacturer [NXP, PCA9685](https://www.nxp.com/products/power-management/lighting-driver-and-controller-ics/led-controllers/16-channel-12-bit-pwm-fm-plus-ic-bus-led-controller:PCA9685), where you can find the datasheet.

tomatos-st, 2023-02-24:
Ok, I guess we should close the topic. 
It will use the PCA9685 chip. Sorry for the confusion  :)
I don't know why I didn't see a plugin to support it. I guess I'm getting old... :)