letscontrolit/ESPEasy #4514: [Dummy] Naming with numericals yields unexpected behavior

TD-er, 2023-02-22:
Rules:
```
  On Rules#Timer=1 do  //When Timer1 expires, do
    Let,4,%unixtime% // Store the current UnixTime in variable #4
  
    TaskValueSet 7,1,[gps#long]
    TaskValueSet 7,2,[gps#lat]
    TaskValueSet 7,3,[gps#hdop]
    TaskValueSet 7,4,[gps#spd]
    
    TaskValueSet 8,1,[sunrise#co2]
    TaskValueSet 8,2,[bme#H]
    TaskValueSet 8,3,[bme#P]
    TaskValueSet 8,4,[bme#T]
    
    TaskRunAt,7,[int#4]
    TaskRunAt,8,[int#4]
  endon
```

Task 7 & 8 are Dummy tasks.
However they have as taskname: `7` and `8` and their task value names are left empty.
The result is that the dummy tasks will not behave as expected, since the connected controller(s) to those tasks never receive any data to process.

TD-er, 2023-02-22:
It looks like it is even worse, as the nrs are being interpreted as unixtime for `TaskRunAt`

tonhuisman, 2023-02-22:
Isn't that scenario (I know it's not yours 😃) destined to shoot yourself in both feet, while standing on a hot air balloon?

Why would anyone want to use `TaskRunAt`, at a time that has already passed?, when a simple `TaksRun,7` and `TaskRun,8` would suffice?

tonhuisman, 2023-02-22:
Yes, but looking at the code, I see that argument 2 is being parsed for that time-stamp, while it should be argument 3...

TD-er, 2023-02-22:
> Yes, but looking at the code, I see that argument 2 is being parsed for that time-stamp, while it should be argument 3...

Sounds like "the bug" ;)

PR #4515 (2023-02-22): [Scheduled Tasks] Parse correct argument for scheduling
Resolves #4514 

- Parse correct 3rd argument for:
  - `TaskRunAt,<task>,<unixtime>`
  - `ScheduleTaskRun,<task>,<offsetMsec>`
closed 2023-02-22 by TD-er