letscontrolit/ESPEasy #4523: letscontrolit.com down?

chromoxdor, 2023-02-28:
Is the sever down or is it maybe only me because i am abroad ?

TD-er, 2023-02-28:
Yep, the person responsible for the hosting is working on it.
Something went wrong with applying updates to the server.

chromoxdor, 2023-02-28:
Ah,ok.. thanks for the info!

chromoxdor, 2023-02-28:
> or is it maybe only me because i am abroad ?

Now that i read this again, it sounds really stupid. Of course i am probably always abroad relative to the server location… but right now i am a bit further away from home and i had the whole time issues with the internet connection here. So i thought it might be that. 

> Something went wrong with applying updates to the server.

I am running and maintaining a small server for a company and applying updates is always my greatest fear. I had fortunately only once a bigger issue and it was anything but fun to get the server back runnig.. especially when it is physically somwhere else. So i hope the problem with letscontrolit doesn’t cause to much headaches. Fingers crossed and good luck!

TD-er, 2023-02-28:
It's up again... running on a new server.

closed 2023-03-01 by chromoxdor