letscontrolit/ESPEasy #4524: [Dallas] wrong sensor value with parasitic powerd DS18B20

ZisMar, 2023-02-28:
Firmware: ESP_Easy_mega_20221224_normal_ESP8266_4M1M.bin
I have updated my ESP8266 from a firmware older than 2 years to the latest firmware and got a wrong sensor value at one of two connected DS18B20 sensors.
Both Sensors are connected to GPIO 12 (D6). Device one is connected by 3 wires (3.3V, GND and Data), device three is connected in parasitic power mode (GND and Data at ESP side, VSS and VDD at sensor connected together). See screenshot.

[image]

Device one is working fine, device three displays wrong temperature. real temperature is approx -16°C, displayed is approx 60°C.
I downgraded the firmware to older versions step by step. Right value is displayed with firmware "ESP_Easy_mega_20220616_normal_ESP8266_4M1M.bin". Beginning with firmware "ESP_Easy_mega_20220809_normal_ESP8266_4M1M.bin" the wrong value is displayed.

Correct values with firmware "ESP_Easy_mega_20220809_normal_ESP8266_4M1M.bin":

[image]

TD-er, 2023-02-28:
parasitic power is not supported.
Right now we do read the sensors 'interleaved', meaning while one sensor is performing the measurement, the next one is given a command to read.
This is not allowed for parasitic powered sensors, as you need to keep the data pin high during measurement for parasitic powered sensors.

uzi18, 2023-02-28:
@TD-er still in the same place.

ZisMar, 2023-03-01:
Thanks for the reply. Clear for me now. Will not use parasitic power any longer.

closed 2023-03-01 by ZisMar