letscontrolit/ESPEasy #4530: Tasks from ESPEasy P2P Networking in mega-20230304 can't be enabled

fly74, 2023-03-04:
Tasks from ESPEasy P2P Networking in mega-20230304 can't be enabled. This is necessary to use the P2P values in rules.

fly74, 2023-03-04:
Worked in the x-Mas release.

tonhuisman, 2023-03-04:
Can you show a task example?
A recent change was that each plugin should return `true` on `PLUGIN_INIT`, and I checked all plugins, so this is rather unexpected.

fly74, 2023-03-04:
Hi @tonhuisman,

[image]

enable and save does nothing.

Strange is that the P2P device shoes a I2C setting, I don't know if it was before in remote devoces.

[image]

TD-er, 2023-03-04:
Ahhhh.... remote devices....
That's for sure one thing we didn't test.

I think we will make a new build ASAP.

tonhuisman, 2023-03-04:
Ah, then it is disabled by the new I2C Device check, as the device isn't available locally. You can disable that check on Tools/Advanced page `Check I2C devices when enabled`

TD-er, 2023-03-04:
Even though it can be disabled, it is quite frustrating to realize that even though we do test extensively, there still are bugs without the hour after a release.... again.
So can you please test some more, so we can collect all those issues (if any) and make a new build tomorrow. (to have a new date)

fly74, 2023-03-04:
> h, then it

That fixed it.

fly74, 2023-03-04:
> Ah, then it is disabled by the new I2C Device check, as the device isn't available locally. You can disable that check on Tools/Advanced page `Check I2C devices when enabled`

That fixed it!

fly74, 2023-03-04:
> Even though it can be disabled, it is quite frustrating to realize that even though we do test extensively, there still are bugs without the hour after a release.... again. So can you please test [...]

It's not a big thing because you guys are quick in fixing!

tonhuisman, 2023-03-04:
@fly74 once [this Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/4331865257) is finished, can you re-test, please?

TD-er, 2023-03-04:
> It's not a big thing because you guys are quick in fixing!

Doesn't change the feeling of "Aw snap...." when a new bug is found almost immediately after a new build.

Jason2866, 2023-03-04:
@TD-er  You are not alone with this experience :-(  It is Frustrating!!

fly74, 2023-03-04:
> @fly74 once [this Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/4331865257) is finished, can you re-test, please?

Sure.

fly74, 2023-03-04:
> > It's not a big thing because you guys are quick in fixing!
> 
> 
> 
> Doesn't change the feeling of "Aw snap...." when a new bug is found almost immediately after a new build.

Next time I will wait with the post. 😁

fly74, 2023-03-04:
> @fly74 once [this Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/4331865257) is finished, can you re-test, please?

The positive thing is the task can now be activated with Check I2C devices when enabled = on, but but after the flash all devices are disabled and you have to activate they  manually  :/

fly74, 2023-03-04:
> The positive thing is the task can now be activated with Check I2C devices when enabled = on, but but after the flash all devices are disabled and you have to activate they manually :/

And after a reboot they are off again :(

tonhuisman, 2023-03-04:
You did save the settings, after enabling the tasks, I assume?

fly74, 2023-03-04:
> [and Light/Lux - BH1750 don't receive values.] - look's to an hardware issue, another device works.

I reverted to x-mas release and had to use the config backup to get the BH1750 work. It seems there is something with I2C not correct.

TD-er, 2023-03-04:
Does the sensor show up on the I2C scan on the ESP32 node?

fly74, 2023-03-04:
> You did save the settings, after enabling the tasks, I assume?

yes, 2 times, ESP32 device.

fly74, 2023-03-04:
> Does the sensor show up on the I2C scan on the ESP32 node?

Yes, but no values and no event rule fired. x-mas works again.

TD-er, 2023-03-04:
When the I2C device is not responding to an I2C scan, it is disabled. But not saved.
So if you don't save but just reboot, the tasks will be enabled again until they don't respond to an I2C scan...

fly74, 2023-03-04:
Maybe ESP32 related.

TD-er, 2023-03-04:
It could be the `BH1750` sensor is maybe responding not always to an I2C scan?
If you disable this check on an ESP32, does it then work?

fly74, 2023-03-04:
> It could be the `BH1750` sensor is maybe responding not always to an I2C scan? If you disable this check on an ESP32, does it then work?

I disabled the check, the enabled the task and made a reboot. Don't work. Task is on, no values no firing. x-mas without issues.

tonhuisman, 2023-03-04:
I've changed to default behavior of the check to return 'OK' (true) when it's called without a TaskIndex, as it was in the previous implementation, the [Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/4332319826) is running.

fly74, 2023-03-04:
Here the config of my device:

<html>
<body>

ESP Chip ID: | 10402644 (0x9EBB54)
-- | --
ESP Chip Frequency: | 240 MHz
ESP Crystal Frequency: | 40 MHz
ESP APB Frequency: | 80 MHz
ESP Chip Model: | ESP32-D0WDQ6-V3
ESP Chip Revision: | 3
ESP Chip Cores: | 2
ESP Board Name: | Espressif Generic ESP32 4M Flash ESPEasy 1810k Code/OTA 316k FS

</body>
</html>

[image]

Rule not firing:

´´´´
on DESKLX#Lux do
//if [SWON#State]=1
if %iswifi%=7
  NeoPixel,6,0,255,0
else
    NeoPixel,6,255,0,0
endif
 
if %eventvalue1% < 20
 [7SEG1].7db,0
 [7SEG2].7db,0
 [7SEG3].7db,0
 [7SEG4].7db,0
 NeoPixelBright,1
 
else
 [7SEG1].7db,1
 [7SEG2].7db,1
 [7SEG3].7db,1
 [7SEG4].7db,1
 NeoPixelBright,255
 endif

let,2,%eventvalue1%/30
let,2,round([var#2])
let,3,[var#2]+1

//  [7SEG1].7dtext,[var#2]

If [var#2] <3

NeoPixelLine,0,[var#2],255,255,0
NeoPixelLine,[var#3],3,255,0,0

Else
NeoPixelLine,1,3,255,255,0
Endif

endon
´´´

fly74, 2023-03-04:
> I've changed to default behavior of the check to return 'OK' (true) when it's called without a TaskIndex, as it was in the previous implementation, the [Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/4332319826) is running.

Same issue. "Check I2C devices when enabled" = on and tasks are off. :/

tonhuisman, 2023-03-04:
Was the last update successful? As changing the default return value is supposed to fix that.

fly74, 2023-03-04:
> Was the last update successful? As changing the default return value is supposed to fix that.

No, as I wrote. The task can be enabled. But after a reboot they are disabled again.

tonhuisman, 2023-03-04:
I think I've found the culprit, a fresh build is brewing, [over here](https://github.com/letscontrolit/ESPEasy/actions/runs/4333022970).

fly74, 2023-03-05:
> I think I've found the culprit, a fresh build is brewing, [over here](https://github.com/letscontrolit/ESPEasy/actions/runs/4333167957).
> 
> Edit: Updated Actions link to newer build

@tonhuisman @TD-er It looks that fixed it. I flashed it over x-mas release with x-mas config. "Check I2C devices when enabled" is on after flash, tasks are enabled even after reboot. works as expected.

fly74, 2023-03-05:
But another thing. One of the devices has a real bme280 and don't send the values to the p2p devices with this action build. Other devices with official 20230304 do it.

tonhuisman, 2023-03-05:
Can you check the log for the BME, to see if it is actually working as intended, with the 20230304 release? It might log errors if there's something irregular going on.

TD-er, 2023-03-05:
I've had one BME280 module once mounted outside, which apparently had some dirt on one or more pins, causing it to constantly flip between the 2 possible I2C addresses.
Something like that would for sure show up in the logs.

tonhuisman, 2023-03-05:
@fly74 You seem to have these BME280 issues on ESP32, correct? I've updated to a newer version of the ESP32 Arduino framework in [this Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/4338122290), hope you can test this successfully.

fly74, 2023-03-06:
The issue with BH1750 and BME280 ist a 8266 device the log show's:
```
10936 : Info   : EVENT: System#Boot
10942 : Info   : ACT  : timerSet,1,1
10946 : Info   : ACT  : NeoPixelAll,0,0,0,0
10950 : Info   : P038 : write - NeoPixelAll,0,0,0,0
10953 : Info   : ACT  : NeoPixel,3,255,0,0
10957 : Info   : P038 : write - NeoPixel,3,255,0,0
10982 : Info   : EVENT: WiFi#APmodeDisabled
11035 : Info   : BMx280: Unable to detect chip ID (0, failed)
11036 : Info   : BMx280: Unable to detect chip ID (0, failed)
11539 : Info   : BH1750 Address: 0x23 Mode: 0x63 : Light intensity: 0.00
```
But as I said, flash back to x-mas, all is fine.

fly74, 2023-03-06:
Log from x-mas:

```
Update: ESP_Easy_mega_20221224_normal_ESP8266_4M1M.bin
                                                      sleep disable
                                                                   .....................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................Update Success: 1024080
      Rebooting...

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
@cp:0
     ld
       ▒U14111 : Info   : WIFI : Set WiFi to AP+STA
18965 : Info   : WIFI : Set WiFi to OFF
19181 : Info   :

INIT : Booting version: ESP_Easy_mega_20221224_normal_ESP8266_4M1M, (GitHub Actions) mega-20221224_35f2ff8 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
19182 : Info   : INIT : Free RAM:27304
19184 : Info   : INIT : Cold Boot - Restart Reason: Software/System restart
19185 : Info   : FS   : Mounting...
19210 : Info   : FS   : Mount successful, used 76806 bytes of 957314
19292 : Info   : FS   : Success garbage collection
19345 : Info   : FS   : Success garbage collection
19395 : Info   : FS   : Success garbage collection
19402 : Info   : CRC  : Settings CRC           ...OK
19409 : Info   : CRC  : SecuritySettings CRC   ...OK
19422 : Info   : INIT : I2C
19423 : Info   : INIT : SPI not enabled
19538 : Info   : WIFI : Set WiFi to STA
19641 : Info   : WiFi : Start network scan all channels
21826 : Info   : WiFi : Scan finished, found: 1
21828 : Info   : WiFi : WifiDisconnect()
21928 : Info   : WIFI : Disconnected! Reason: '(1) Unspecified'
21929 : Info   : WIFI : Arduino wifi status: WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
22030 : Info   : Reset WiFi.
22031 : Info   : WiFi : Start network scan all channels
24214 : Info   : WiFi : Scan finished, found: 3
24217 : Info   : WiFi : Best AP candidate: #Hidden# xxxxxxxxxxxxxxxxxxxxx
24318 : Info   : WIFI : Arduino wifi status: WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
24318 : Info   : WIFI : Set WiFi to OFF
24432 : Info   : INIT : Free RAM:25816
24575 : Info   : INFO : Plugins: 47 [Normal] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
24576 : Info   : EVENT: System#Wake
24801 : Info   : WIFI : Set WiFi to STA
24906 : Info   : WiFi : Best AP candidate: xxxxxxxxxxxxxxx
24907 : Info   : WIFI : xxxxxxxxxxxxxxxxxxxx
24908 : Info   : IP   : Static IP : xxxxxxxxxxxxxxxxxxx
24916 : Info   : Webserver: start
24929 : Info   : EVENT: System#Boot
24934 : Info   : ACT  : timerSet,1,1
24938 : Info   : ACT  : NeoPixelAll,0,0,0,0
24942 : Info   : P038 : write - NeoPixelAll,0,0,0,0
24945 : Info   : ACT  : NeoPixel,3,255,0,0
24948 : Info   : P038 : write - NeoPixel,3,255,0,0
24973 : Info   : EVENT: WiFi#APmodeDisabled
25027 : Info   : BMx280 : Detected BME280
25536 : Info   : BH1750 Address: 0x23 Mode: 0x63 : Light intensity: 0.00
25563 : Info   : EVENT: WiFi#Disconnected
25593 : Info   : SW   : State 1
25670 : Info   : EVENT: TaskInit#SZ=4,1
25721 : Info   : SW  : GPIO=16 State=0 Output value=0
25786 : Info   : EVENT: TaskInit#LXSZT=7,1
25826 : Info   : BME280: Apply temp offset -2.70C humidity 43.67% => 47.44% temperature 24.07C => 21.37C dew point 9.69C
25854 : Info   : EVENT: TaskInit#SW=8,1
25910 : Info   : BME280 : Address: 0x76
25914 : Info   : BME280 : Temperature: 21.37
25916 : Info   : BME280 : Humidity: 47.44
25918 : Info   : BME280 : Barometric Pressure: 1004.24
```

TD-er, 2023-03-06:
So there actually is a difference in addressing the BME280 as the latest build you show does log quite a few `BMx280: Unable to detect chip ID (0, failed)` entries.

fly74, 2023-03-06:
> So there actually is a difference in addressing the BME280 as the latest build you show does log quite a few `BMx280: Unable to detect chip ID (0, failed)` entries.

That's my supposition. But I have some devices with bme280 working with the last official release 20230304. The only thing is that the working ones are bought long time ago and this bme is a newer one.

tonhuisman, 2023-03-06:
Does it help if you enable to use the Low I2C speed for that failing sensor?

fly74, 2023-03-06:
The "failing sensor" works great in x-mas. It looks the chip revision is not detected ?!?

tonhuisman, 2023-03-06:
Different Arduino versions do have different behavior in the I2C area, also on ESP8266, please try the Low speed option, to see if that works.

fly74, 2023-03-06:
Where is the low speed option?

tonhuisman, 2023-03-06:
A checkbox on the Device Configuration for the BMx280

fly74, 2023-03-06:
?

[image]

tonhuisman, 2023-03-06:
Hm, that's not as expected. There was a change in that area for the remote devices, I'll have a look asap, but you should be able to test that unit with the original 20230304 release build, there the I2C Address selection and Force Slow I2C speed options should be visible.

tonhuisman, 2023-03-06:
Strangely, that screenshot shows `[Detected: BME280]`...

fly74, 2023-03-06:
> Hm, that's not as expected. There was a change in that area for the remote devices, I'll have a look asap, but you should be able to test that unit with the original 20230304 release build, there the I2C Address selection and Force Slow I2C speed options should be visible.

Low speed seems to work, but this release has the not receiving P2P values issue. :/

tonhuisman, 2023-03-06:
The I2C options should be shown again with the fix available from [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4343104100), once it's finished.

Can't yet see why receiving via the P2P controller isn't working.

TD-er, 2023-03-06:
I also updated the release page for the 20230304 build to indicate these known issues

fly74, 2023-03-06:
> The I2C options should be shown again with the fix available from [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4343104100), once it's finished.
> 
> Can't yet see why receiving via the P2P controller isn't working.

@tonhuisman Seem to fix it. Only the remote tasks must be enabled again after the flash. Thumbs up!

fly74, 2023-03-06:
@TD-er Can this fix easy build to a new official?

TD-er, 2023-03-06:
I will discuss with Ton to see if something else needs to be done and if not, I will make a build later this evening.

closed 2023-03-06 by TD-er

TD-er, 2023-03-07:
> > > It's not a big thing because you guys are quick in fixing!
> > 
> > 
> > Doesn't change the feeling of "Aw snap...." when a new bug is found almost immediately after a new build.
> 
> Next time I will wait with the post. 😁

Thanks for not reporting any new issue with the 20230306 build for at least 11h :)

PR #4532 (2023-03-06): [I2C] Don't check device availability for remote I2C devices and other improvements
Several bug fixes and corrections:
- I2C devices should not be checked when not local (via P2P network)
- Remote devices should not show Pin/Port/Address configuration options
- Remove duplicate Release_notes.txt content
- Release.yml improved step name
- Add RTD Help button to Controller configuration page
- Update ESP32 Arduino framework lib (2023.01.02)
- P037: Always return success for `PLUGIN_INIT` (subscription will be retried on change of MQTT status)
- Devices: Show I2C options correctly for local devic [... cut]