letscontrolit/ESPEasy #4531: DFPlayer-Mini MP3 not working on ESP32 platform

Vladimir-by, 2023-03-04:
With any choice of GPIO → RX, there is no command to control DFPlayer-Mini MP3. Tested on ESP-WROOM-32, ESP32-WROVER-B, firmware mega_20230304_collection_A_ESP32, mega_20221224_collection_A and others. 
Found the following article, may be useful in solving this problem https://quadmeup.com/arduino-esp32-and-3-hardware-serial-ports/

tonhuisman, 2023-03-06:
> With any choice of GPIO → RX

What GPIO pin have you actually used?

> there is no command to control DFPlayer-Mini MP3

You did send the command `play,1` or similar to the ESP unit where the player is connected? And the SD card of the player does hold some MP3 files, I assume?
Easiest way to send such play command would be from the Web UI on the Tools tab. Be sure to enable the device. And possibly the volume needs to be set too, using the `vol,<volume>` command (`<volume>` range: 0..30)

ESPEasy uses ESPEasySerial, a tried and proven serial library, that works fine with other serial devices (CO2 sensors, P1 monitor, GPS receivers), so it should work equally fine with these MP3 players.

Vladimir-by, 2023-03-08:
Checked different free GPIO ports.
Tested on ESP-WROOM-32 (MH-ET LIVE), ESP32-WROVER-B (T18_3.0) boards. The firmware used is ESP_Easy_mega_20230306_collection_A_ESP32_4M316k.
A USB-UART adapter is connected to the selected GPIO (RX is used for receiving).
Control command sent -> Tools -> Command -> play,001 -> response received Command Output -> ok
Recommendations https://espeasy.readthedocs.io/en/latest/Plugin/P065.html used
Command sent http://http://192.168.0.105//control?cmd=play,001
In both cases, there is no control command, no USB-UART adapter activity, no LED flash.

Vladimir-by, 2023-03-08:
The ESP8266MOD (ESP-12F) module was tested under similar conditions.
Firmware ESP_Easy_mega_20221224_collection_A_ESP8266_4M1M.
Control command for DFPlayer-Mini MP3 is present.
The USB-UART adapter shows activity, the LED flashes. The original DFPlayer-Mini MP3 (DFR0299) plays the message play,001.

Vladimir-by, 2023-03-08:
Colleagues, please check the functionality of the DFPlayer-Mini MP3 plugin (P065_DFR0299_MP3) on your ESP32 debug module.

tonhuisman, 2023-03-11:
@Vladimir-by I have created a PR that should resolve this issue. Can you test using [this Github Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/4393982457), once that's finished?

Vladimir-by, 2023-03-11:
Thanks Ton Huisman for the corrections. Firmware ESP_Easy_mega_20230311_collection_A_ESP32_4M316k tested in different HW Serial modes (1, 2, 3) and different GPIO pins. The commands of the connected USB-UART adapter are generated. DFPlayer-Mini MP3 (DFR0299) works and plays a message on commands play,001 and others.

Vladimir-by, 2023-03-11:
[image]

Vladimir-by, 2023-03-11:
Setup menu Notify-DFPlayer-Mini MP3 contains redundant I2C Serial information.

TD-er, 2023-03-11:
I2C serial is for using those [SC16IS752](https://www.nxp.com/products/peripherals-and-logic/signal-chain/bridges/single-uart-with-ic-bus-spi-interface-64-bs-of-transmit-and-receive-fifos-irda-sir-built-in-support:SC16IS740_750_760) chips.
Maybe not the best choice ever to make those the first in the list as it is not the most likely option for most users.

Vladimir-by, 2023-03-11:
It is clear, I agree, it is necessary to leave this element for alternative connection for various projects.

tonhuisman, 2023-03-12:
I have re-ordered the list to show I2C Serial as the last option in the combo box, instead of the first.

Vladimir-by, 2023-03-15:
Firmware ESPEasy_mega_20230312_ESP32_PR#4542_4397836468.zip has been tested by ESP_Easy_mega_20230312_collection_A_ESP32_4M316k on the ESP32 platform and is currently in production. Please apply the changes in the official release.

PR #4542 (2023-03-24): [P065] Add Serial configuration to enable use on ESP32
Resolves #4531 

Features:
- Enable serial configuration so this plugin can be used on ESP32
  - Defaults to HW Serial1 on ESP32.
  - Converts GPIO pin for existing plugin installations, _not_ backward compatible!
- Updated documentation
- Re-order list of serial port options to show I2C Serial as last item.
- Hide I2C Serial completely when disabled

Additional fixes:
- P144: Disable additional logging

TODO:
- [x] Confirmation from issue reporter ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/4531#issuecomment-1465026277))
closed 2023-03-24 by TD-er