letscontrolit/ESPEasy #4535: mega-20230304 mqttimport disabled after reboot

Dickesplumpel, 2023-03-06:
Hi, I have updated two ESP32 and one ESP32-S2 from mega-20221224 to mega-20230304. When I reboot or reset the ESPs all MqttImport devices are disabled. When I enable the devices they work as exspected although the system variables page shows:
%ismqtt%   1   1
%ismqttimp%   0   0

A fresh flash with ESP_Easy_mega_20230304_normal_ESP32_4M316k.factory.bin on the ESP32 doesn`t change this behavior.
So long.

TD-er, 2023-03-06:
Sounds like it is slightly related to the reported issues with the I2C devices.
Like when the init fails of the plugin, it is disabled.
But when the unit starts, there isn't any network connection yet, so the init of the plugin fails (which is supposed to)

TD-er, 2023-03-06:
I also updated the release page for the 20230304 build to indicate these known issues

Dickesplumpel, 2023-03-06:
Okay, thank you!

closed 2023-03-06 by Dickesplumpel

Dickesplumpel, 2023-03-06:
ESP_Easy_mega_20230306_normal_ESP32_4M316k.bin works for me, mqttimport is enabled after reboot. Thank you very much!

tonhuisman, 2023-03-06:
> works for me, mqttimport is enabled after reboot

Because you already closed this issue (not needed as I linked it to the pull request to auto-close on merge), I assumed you had already tested it to work as intended. 😮 

Thanks for reporting your positive result 👍

PR #4532 (2023-03-06): [I2C] Don't check device availability for remote I2C devices and other improvements
Several bug fixes and corrections:
- I2C devices should not be checked when not local (via P2P network)
- Remote devices should not show Pin/Port/Address configuration options
- Remove duplicate Release_notes.txt content
- Release.yml improved step name
- Add RTD Help button to Controller configuration page
- Update ESP32 Arduino framework lib (2023.01.02)
- P037: Always return success for `PLUGIN_INIT` (subscription will be retried on change of MQTT status)
- Devices: Show I2C options correctly for local devic [... cut]