letscontrolit/ESPEasy #4539: New plugin HttpImport could be interesting

popaserge, 2023-03-08:
I am not able to code by myself a new plugin but I saw that MqttImport is able to parse a Json string.
I think that for specialists it could be very easy to use the parsing possibilities allready used in MgttImport and apply them in a new plugin using the SendToHttp command feature.
I have a device used before I happened to know EspEasy which is able to answer to a http request by a Json string after easy configuration in it of a cgi file and made a trial.
Answer in a browser : { PuissancePAC :620, IntensitePAC :3, Nappe_Arr :14.06, Exterieur :10.06, PAC_Retour :37.25, PACext_Ret :14.56, Interieur :19.31, PAC_Sortie :37.56, Nappe_Ret :13.25, PACext_Sor :14.75, Air_Pulse :33.75 }

In the tools/command  of Espeasy  I sent a SendToHttp command to this device and looked to the reaction in the log.
It seems that the data are well received; here is the log data and the way for such a new plugin is maybe not so far

32087776:  Webserver 1 Arguments: 0: 'cmd' length: 61
32087778: HTTP: SendToHTTP,http://admin:xxx@192.168.2.133/JSONTestNomHttp.CGI
32087780: HTTP after parseTemplate: SendToHTTP,http://admin:xxx@192.168.2.133/JSONTestNomHttp.CGI
32087783: Command: SendToHTTP
32087783: SendToHTTP,http://admin:xxx@192.168.2.133/JSONTestNomHttp.CGI
32087784: Par1: 0 Par2: 0 Par3: 0 Par4: 0 Par5: 0
32087785: SendToHTTP: Host: 192.168.2.133 port: 80 path: /JSONTestNomHttp.CGI
32087813: HTTP : SendToHTTP 192.168.2.133 GET... HTTP code: 200
32087815: HTTP : SendToHTTP 192.168.2.133 GET... HTTP code: 200 Received reply: {^^PuissancePAC^^:620,^^IntensitePAC^^ :3,^^ Nappe_Arr^^ 
32087887: EVENT: http#192.168.2.133=200
32087888: EVENT: http#192.168.2.133=200 Processing time:2 milliSeconds

agroszer, 2023-03-29:
+1 on this, fetch a target URL and parse the JSON response, that's it. No need to interfere with SendToHTTP

sjfaustino, 2023-04-10:
I'm currently using a ESP8266 with a relay shield connected to a 12v car horn as an alarm that reads the status of a webcam and sounds the horn if the camera is in an alarm state because of visual movement.
It would be awesome if this could be done with rules.

Current code (partial):
```
const String cam_PortariaEsq = "http://192.168.8.221:28021/get_status.cgi?user=myUser&pwd=myPwd";

    HTTPClient http;
    http.begin(wifiClient, cam_PortariaEsq);
    int httpCode = http.GET();
    // httpCode will be negative on error
    if (httpCode > 0) {
      if (httpCode == HTTP_CODE_OK) {
        //HTTP_CODE_OK means code == 200
        String payload = http.getString();  // gives us the message received by the GET Request        
        int tempindex = payload.indexOf("var alarm_status=1");
        if (tempindex > 0) {
          RelayAlarm(0.5, 0.5, 3);
          delay(30000); // Insures no more beeps until 30s passed
        }
      }
```

chromoxdor, 2026-01-31:
> how correctly writre functions reply?

There is a documentation for that under `SendToHttp` called `HTTPResponseParser`.
See here: https://espeasy.readthedocs.io/en/latest/Rules/HTTPResponseParser.html

First of all, you have to tell SendToHttp that there is JSON to parse.
You do that by adding either `#json` or `?json` to the end of the URL.
Since you do not group, you can catch the parsed reply with: `On JsonReply Do`.

> Most issues seem to be fixed in [#5337](https://github.com/letscontrolit/ESPEasy/pull/5337) , you can pull that branch local, then build, install and test on your device.

This PR is more robust, and you must only append `#json` since it will be cut off before the URL is sent to the server.

Note: all this only works with a custom build where `HTTPResponseParser` is enabled. (see first lines of the documentation)