letscontrolit/ESPEasy #4543: Regulator - Level Control - Feature Request

tomatos-st, 2023-03-13:
Recently I was building a reglator based on espeasy and again I encountered some inconvenience.
With a classic regulator using heating, everything is ok.
The problem begins when we use the regulator for cooling (turning on the cooling fan).
The problem is in the hysteresis operation.
Positive hysteresis loop system is currently used (a), for cooling: negative hysteresis loop system is needed (b)
I know that it can be done in rulers or with an additional task by negating the temperatures, but maybe an additional button defining the type of regulator (heating/cooling) could be added,
The matter would be simple, you just need to negate the regulator's output signal (Inversed output)

[image]

TD-er, 2023-03-13:
Just to make sure I do get it...
You want 2 different hysteresis values?
One for increasing trends and one for decreasing trends?

tomatos-st, 2023-03-13:
There can be one hysteresis but different directions of operation
e.g. at: Set Level: 20 , Hysteresis: 2
When heating: "**ON**" (start 0) - > "**OFF**" at 21 -> and at drop "**ON**" at 19
When cooling: "**OFF**" (start 0) -> "**ON**" at 21 -> and at drop "**OFF**" at 19

It can be done by negating the output state of the regulator

tomatos-st, 2023-03-13:
As I wrote, I know that it can be done in the rules, but this is "patching" the problem.
Besides, at the moment I do not use any GPIO, I send the state of the regulator immediately to Domoticz to a remote network socket that turns on the heater and fans. At this point, one ESP-01 4xDS18 supports me with 3 radiators and a fan.

tonhuisman, 2023-03-13:
@tomatos-st I've added an option to invert the Output value. Can you test [this Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/4409362583) once it's finished?

tomatos-st, 2023-03-14:
Works fine.
Now it's OK.
Thanks

PR #4544 (2023-03-23): [P021] Add Invert Output option
Resolves: #4543 

Features:
- Add Invert Output option to settings
- Update documentation

TODO:
- [x] Testing by requester ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/4543#issuecomment-1467953686))
closed 2023-03-23 by TD-er