letscontrolit/ESPEasy #4545: [P027] Ina2019 current readings fluctuate with dc motor

chromoxdor, 2023-03-14:
Just did some test with the ina2019 and a motor and i got heavy fluctuations in the current readings which made it unusable.

A litte bit of research brought up that:
https://electronics.stackexchange.com/questions/515455/ina219-current-sensor-not-working-properly-with-dc-motor

So i changed to `INA219_CONFIG_SADCRES_12BIT_128S_69MS` in the P027_data_struct.cpp for testing and got nice and steady results.

A possible solution could be to integrate this as a selectable option into the plugin...

TD-er, 2023-03-14:
So essentially enabling oversampling in the sensor?
Sounds like a good idea.
Have to check the datasheet to see which options are available and maybe add a more intuitive selector description to the options instead of these in-depth values like "128 samples, 69 msec".

For example, more like descriptive to their typical use case like "50/60Hz net freq".
Or if that's not really possible, just add all options and.... document it :)

chromoxdor, 2023-03-14:
> Have to check the datasheet to see which options are available and maybe add a more intuitive selector description to the options instead of these in-depth values like "128 samples, 69 msec".

The option are already listed in the cpp file :)

chromoxdor, 2023-03-14:
actually...since the ina219 reads dc and ESPeasy reads the values at a minimum of once per second, averaging with 128samples could be a default. Right now i don´t see another use case where anything else would be required. But maybe i am missing something...

TD-er, 2023-03-14:
Was just thinking the same myself.
Only thing I could think of, but that's something you can already check, is whether reading the sample is also taking 69 msec.
You can see this in the timing stats.

chromoxdor, 2023-03-14:
Looks good:

[image]

chromoxdor, 2023-03-14:
When i am back at home and there are no objections i will make a PR ...

TD-er, 2023-03-14:
No objections here... Simply setting it to use the suggested default setting sounds OK to me.

PR #4546 (2023-03-24): [P027] Change the default averaging to 128samples
Fixes: #4545
closed 2023-03-24 by TD-er