letscontrolit/ESPEasy #4548: Triple and quadruple button click event

Vladimir-by, 2023-03-14:
There is a need to track triple and quadruple button presses. In the Device Switch input - Switch settings, only the Doubleclick event is present. How to organize tracking of the event of triple and quadruple clicking on the button?

TD-er, 2023-03-15:
Just curious to what use case you want to have this feature for.
I can't think of any commercially available switch like device that does support > 3 taps.

Well... unless you want to implement an old-fashioned phone dialer wheel :)

Vladimir-by, 2023-03-15:
It is necessary to switch two pairs of blocks each of the four modules using fewer buttons. Each switch will be processed by a certain number of clicks.

Vladimir-by, 2023-03-15:
The Xiaomi WXKG13LM button, for example, has the ability to process the following values: single, double, triple, quintuple, hold, release, many.
https://www.zigbee2mqtt.io/devices/WXKG13LM.html  (zigbee2mqtt project)

Vladimir-by, 2023-03-19:
Please tell me a possible implementation of the rules code to apply the events of triple and quadruple clicks on the button.

UriUriK, 2023-04-01:
Dummy counter, limit timer with counter reset, event for working with the number of clicks. I use up to 15 clicks in 5 seconds.

tonhuisman, 2023-04-01:
> Dummy counter, limit timer with counter reset, event for working with the number of clicks. I use up to 15 clicks in 5 seconds.

Would be nice if you can share your code, assuming it's ESPEasy related?

Vladimir-by, 2023-04-01:
UriUriK, this topic is relevant for the implementation of my project. Please share your own solution to this issue.

UriUriK, 2023-04-02:
> @UriUriK Would be nice if you can share your code, assuming it's ESPEasy related?
> UriUriK, this topic is relevant for the implementation of my project. Please share your own solution to this issue.
[image]
[rules1.txt](https://github.com/letscontrolit/ESPEasy/files/11132841/rules1.txt)

UriUriK, 2023-04-03:
removed the dummy device, work with variables
[image]
[rules1 (1).txt](https://github.com/letscontrolit/ESPEasy/files/11139895/rules1.1.txt)

tonhuisman, 2023-04-03:
Thank you for sharing 👍