letscontrolit/ESPEasy #4552: OLED SSD1306/SH1106 - show content on webpage in "Values" column

MarkusH1975, 2023-03-16:
feature request for OLED SSD1306/SH1106

Hi,
please show current **display content** in the Devices overview page **in "Values" column.**
Like it is done already for other devices like DS18b20, ...

This can help to check while changing the content. Especially when filling with mqtt imported content.
In the case the display is not next to my computer and therefore i cannot see it.

Thanks,
Markus

TD-er, 2023-03-16:
Hmm sounds like a nice idea.
There is however one possible issue here which might cause quite a lot of unforeseen other issues.
A display may show strings and the JSON to format the updates may not be handling this very well. (or the JavaScript to parse this JSON)

Also this could mean the string generation of the Framed OLED plugin may be called a lot more often. It really isn't the most resource friendly function call to generate those strings.

tonhuisman, 2023-03-16:
@MarkusH1975 I've experimented a bit, and created a PR to show the on-display content on the Devices page. Can you test using [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4442212879) once that's finished?

tonhuisman, 2023-04-26:
@MarkusH1975 You haven't responded yet to requests to test this new feature, are you still available?
There is a [new Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4812212059), with the latest ESPEasy updates included, for you to test if it does what you requested.
Please reply here with your findings, positive or negative 😃

tonhuisman, 2023-05-14:
@MarkusH1975 You still haven't responded so should we assume you're no longer interested?
A new [Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4974463032) is available for testing. Please reply here with your findings, good or bad.
TIA.

tonhuisman, 2023-06-30:
@MarkusH1975 Another call-out to get a response. I've just started a re-build with the latest `mega` branch included [over here](https://github.com/letscontrolit/ESPEasy/actions/runs/5426129341). Please respond after testing this feature, your feedback is valuable. TIA.

PR #4554 (2023-08-25): [P036][P023] Show current on-display content on Devices page
Resolves #4552 

P036 - Display - OLED SSD1306/SH1106 Framed
P023 - Display - OLED SSD1306

Features:
- Show current on-display content on the Devices overview page (Values column)
- Font-size is reduced to 75% for the shown content
- Applied also to P023 - Display - OLED SSD1306
- P023: Manually set content via `oled,x,y,<content>` is also included on the Devices page
- P023: Trailing empty lines are omitted from the Devices page display to conserve space
- Make Interval optional
- P036: Center- and Right-alignment is applied in pre [... cut]
closed 2023-08-25 by TD-er