letscontrolit/ESPEasy #4556: I2C HTU21D broken after update 20230306

eigthyeight, 2023-03-18:
After updating a ESP32 to the latest 20230306 release a HTU21D sensor stopped working.
ESP has 98% load with P_14_Environment - SI70xx/HTU21D read but values stay at NaN.
Sensor is configured to read every 20 sec as slow I2C speed of 100000Hz and the ESP32 is also controlling two PCF8574 devices.
The sensor does show up in a I2C scan but wont read.
After downgrading to 20221224 everything works fine again.

tonhuisman, 2023-03-18:
Have you tried to set a I2C ClockStretchLimit value on the Tools/Advanced page? (usable values could be 1000..20000)
Some, mostly older, I2C sensors need that, and also there have been quite a few changes in I2C handling of the ESP32 framework libraries the last few months, so this might now be needed for this sensor.

Also, some of the log output, including data of this sensor, could be helpful in diagnosing this issue.

eigthyeight, 2023-03-18:
I've done some tests with values of 1000,10000 and 20000 at borh 100kHz and 400kHz, no success.

How do I get log outputs with sensor data?

tonhuisman, 2023-03-18:
> How do I get log outputs with sensor data?

- You can have a look at the Tools/Log page.
- You can connect an USB cable to the unit, and use a terminal program like Putty (115200,N,8,1) to look at the serial data returned from the unit.

Don't know exactly how much logging is produced by this plugin though.

eigthyeight, 2023-03-18:
Got an other (same) ESP32 Wroom with an additional HTU21D, nothing else connected; exact same issue older fw no problem, new fw; nan

log seems quite empty?
236244: sendcontent free: 203204 chunk size:118
236252: sendcontent free: 201404 chunk size:0
237348: sendcontent free: 203428 chunk size:4096
237427: sendcontent free: 201796 chunk size:4096
237446: DEBUG: String size:8192
237448: sendcontent free: 198420 chunk size:1357
237457: sendcontent free: 196884 chunk size:0
237467: UDP  : 84:F3:EB:1C:3C:C9,192.168.188.117,17
237484: sendcontent free: 202840 chunk size:778
237494: sendcontent free: 201352 chunk size:0
237511: UDP  : 84:F3:EB:1C:3C:C9,192.168.188.117,17
237520: static_file: /esp.css
237521: Serve 304: 1679155708 /esp.css
237521: HTML : Request file /esp.css
237552: static_file: /favicon.ico
237552: HTML : Request file /favicon.ico
237558: sendcontent free: 203108 chunk size:390
237567: sendcontent free: 201844 chunk size:0
238669: sendcontent free: 203792 chunk size:897
238683: sendcontent free: 201984 chunk size:0
238762: sendcontent free: 206184 chunk size:1351
238770: sendcontent free: 201084 chunk size:0
239931:  Webserver 2 Arguments: 0: 'index' length: 1 1: 'page' length: 1
239939: Read settings: TaskSettings index: 2
239960: sendcontent free: 203004 chunk size:4096
239986: DEBUG: String size:4096
239988: sendcontent free: 204636 chunk size:3282
240000: sendcontent free: 202840 chunk size:0
240026: sendcontent free: 204732 chunk size:963
240035: sendcontent free: 202944 chunk size:0
240113: sendcontent free: 207136 chunk size:1351
240122: sendcontent free: 202036 chunk size:0
240184: static_file: /favicon.ico
240184: HTML : Request file /favicon.ico
240189: sendcontent free: 205584 chunk size:390
240198: sendcontent free: 203788 chunk size:0
242390: sendcontent free: 205512 chunk size:762
242399: sendcontent free: 203716 chunk size:0
242411: UDP  : 80:7D:3A:37:6A:44,192.168.188.116,16
242416: UDP  : 80:7D:3A:37:6A:44,192.168.188.116,16
242421: UDP  : E8:DB:84:A8:EF:B3,192.168.188.110,10
243553: sendcontent free: 206496 chunk size:545
243562: sendcontent free: 204708 chunk size:0
244709: sendcontent free: 206228 chunk size:282
244720: sendcontent free: 204820 chunk size:0
245868: sendcontent free: 205924 chunk size:283
245876: sendcontent free: 204128 chunk size:0
245889: UDP  : 5C:CF:7F:B1:16:CD,192.168.188.118,18
245894: UDP  : 5C:CF:7F:B1:16:CD,192.168.188.118,18
245899: UDP  : 4C:EB:D6:7A:79:3C,192.168.188.106,15
245904: UDP  : 4C:EB:D6:7A:79:3C,192.168.188.106,15
247033: sendcontent free: 207224 chunk size:633
247042: sendcontent free: 205424 chunk size:0
248192: sendcontent free: 203564 chunk size:282
248203: sendcontent free: 205580 chunk size:0
249354: sendcontent free: 206920 chunk size:283
249362: sendcontent free: 205124 chunk size:0
249380: UDP  : 80:7D:3A:37:63:CA,192.168.188.119,19
249385: UDP  : 80:7D:3A:37:63:CA,192.168.188.119,19
250516: sendcontent free: 207364 chunk size:457
250523: sendcontent free: 202636 chunk size:0
251676: sendcontent free: 207840 chunk size:282
251685: sendcontent free: 205988 chunk size:0
252830: sendcontent free: 207848 chunk size:282
252838: sendcontent free: 205996 chunk size:0
253987: sendcontent free: 207636 chunk size:282
253995: sendcontent free: 205784 chunk size:0
[... 296 lines cut ...]
376813: UDP  : Send Sysinfo message
377945: sendcontent free: 211428 chunk size:760
377958: sendcontent free: 209096 chunk size:0
377979: LoopStats: shortestLoop: 60 longestLoop: 1829687 avgLoopDuration: 2297.53 loopCounterMax: 500000 loopCounterLast: 13113
377979: Scheduler stats: (called/tasks/max_length/idle%) 13087/514/15/4.28
379106: sendcontent free: 211428 chunk size:556
379115: sendcontent free: 209572 chunk size:0
380263: sendcontent free: 211032 chunk size:282
380272: sendcontent free: 209556 chunk size:0
381418: sendcontent free: 210616 chunk size:282
381427: sendcontent free: 209140 chunk size:0
381443: UDP  : EC:62:60:9D:BE:8C,192.168.188.112,12
381448: UDP  : EC:62:60:9D:BE:8C,192.168.188.112,12
382580: sendcontent free: 211428 chunk size:457
382588: sendcontent free: 209572 chunk size:0
383734: sendcontent free: 211052 chunk size:282
383743: sendcontent free: 209576 chunk size:0
384891: sendcontent free: 211032 chunk size:282
384900: sendcontent free: 209564 chunk size:0
386051: sendcontent free: 211032 chunk size:282
386061: sendcontent free: 209556 chunk size:0
387203: sendcontent free: 211004 chunk size:283
387211: sendcontent free: 209148 chunk size:0
387225: UDP  : 84:F3:EB:1C:3C:C9,192.168.188.117,17
387229: UDP  : 84:F3:EB:1C:3C:C9,192.168.188.117,17

tonhuisman, 2023-03-18:
No need to have a DEBUG level log, that's mostly confusing, most plugins log something on INFO level, but it seems this plugin doesn't log much, as I didn't find anything relevant (P014 or SI70xx).
Work has been done on this plugin some time ago, but AFAICS, that was before the 20221224 release.

To get the initialization log for the plugin (again) you can send the command `taskenable,htu21` (or whatever name you gave the task). That command can also be typed into the terminal (blind entry, no correction possibility, command will echo when pressing enter), or from the Tools page.
During initialization it should report the serial number of the device (`SI7013 : sn=xxxxxx-xxxxxx`) that would help in debugging this further, as part of that serial number is the chip-id.

tonhuisman, 2023-03-18:
What exact .bin file do you have installed on your ESP?

TD-er, 2023-03-18:
This P014 does have an issue with settings transitions from earlier builds.
Please remove the task and add it again.

eigthyeight, 2023-03-18:
Build: [ESP_Easy_mega_20230306_normal_ESP32_4M316k Mar 6 2023]

removing and adding again did not make a difference, nor did a factory reset.

log with task enable;

273276: UDP  : 48:3F:DA:5C:42:47,192.168.188.105,5
274412: sendcontent free: 209580 chunk size:373
274421: sendcontent free: 207792 chunk size:0
275580:  Webserver 1 Arguments: 0: 'cmd' length: 19
275581: HTTP: taskenable, i2ctemp
275582: HTTP after parseTemplate: taskenable, i2ctemp
275584: Command: taskenable
275584: taskenable, i2ctemp
275585: Par1: 0 Par2: 0 Par3: 0 Par4: 0 Par5: 0
275595: sendcontent free: 208136 chunk size:4096
275613: sendcontent free: 209720 chunk size:1427
275621: sendcontent free: 204620 chunk size:0
275633: UDP  : 48:3F:DA:59:32:07,192.168.188.47,2
275648: sendcontent free: 210036 chunk size:1084
275656: sendcontent free: 205284 chunk size:0
275671: UDP  : 48:3F:DA:59:32:07,192.168.188.47,2
275712: static_file: /favicon.ico
275713: HTML : Request file /favicon.ico
275718: sendcontent free: 209720 chunk size:390
275726: sendcontent free: 208460 chunk size:0
275741: SI7013 : sn=91cd6-32114854
275741: P014: chip_id=50
276897: sendcontent free: 210816 chunk size:813
276906: sendcontent free: 208964 chunk size:0
278119: sendcontent free: 209808 chunk size:1351
278132: sendcontent free: 206244 chunk size:0
278157: sendcontent free: 210828 chunk size:453
278165: sendcontent free: 208976 chunk size:0

tonhuisman, 2023-03-19:
As this may be related to updates in the Arduino framework libraries, can you test [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4462254537), where I have included a newer version of those libraries?

eigthyeight, 2023-03-19:
Sure!
Unfortunately this does not solve the issue, still no readout:(

TD-er, 2023-03-20:
Can you try with "Check I2C devices when enabled:" unchecked on the Tools->Advanced page?

eigthyeight, 2023-03-20:
that solved it! Thanks for helping :)
new feature which requires some more work?

TD-er, 2023-03-20:
> that solved it! Thanks for helping :) new feature which requires some more work?

Thank you for testing....

eigthyeight, 2023-03-20:
glad to help, if you need a fix tested please let me know

tonhuisman, 2023-03-20:
> glad to help, if you need a fix tested please let me know

Yes please, [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4472554613) should have a fix, where you should turn back on the `Check I2C devices when enabled` option in Tools/Advanced, and check if the HTU21D is responding as expected.

eigthyeight, 2023-03-21:
fix works, HTU21D is working as usual with "Check I2C devices when enabled" on
Thanks for all your great work on this project!

tonhuisman, 2023-03-21:
Thanks for testing 👍

PR #4557 (2023-03-21): [P014] Disable I2C check to correct HTU21D read
Resolves #4556 

Features:
- Fix for I2C check interfering with normal read on HTU21D

TODO:
- [x] Confirmation by issue reporter ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/4556#issuecomment-1478459934))
closed 2023-03-21 by TD-er