letscontrolit/ESPEasy #4558: RS232 command how it works ?

betux69, 2023-03-21:
Good morning,

First of all, I wanted to thank you for the incredible work done. I'm not a technician and I manage to create sensors in a simple way which seems amazing to me.

I want to turn my Epson video projector on and off with the rs232.
I followed this thread:
[serialsend is unknown command · Issue #3351 · letscontrolit/ESPEasy · GitHub](https://github.com/letscontrolit/ESPEasy/issues/3351)

The command I need to send is PWR ON.

There is very little documentation on the subject and not necessarily understandable by a beginner.

I use the console and not the rules to try to turn on the VP.

Question 1: Should a port be specified in the module Serial Server?

Question 2: I use the standard tx, rx. Some command notation there’s :
 serialsend,1, the command. What is 1?

Question 3: Do you need "" in the command?
Exemple :
Serialsend, “PWR ON”
Or 
Serialsend, PWR ON

Thanks a lot

Regards

tonhuisman, 2023-03-21:
For that to work, you need:
- Disable/uncheck the Tools/Advanced option `Enable Serial port`, to free the serial port for use by plugins (this implies that you won't be able to view logging via usb-serial, as that always uses HWSerial0).
- (optional: Hardware reset or power-cycle the ESP to free the serial port)
- A Serial Server task defined to use `HWSerial0` with the required serial settings like baud rate, data bits etc. and enable that task.
- Send the command using `serialsend,<any characters you need to send to the port>`, so if quotes are needed, then these should be included (this plugin uses a more 'plain' way of fetching that data from the command, _not_ the regular command parsing as described [here](https://github.com/letscontrolit/ESPEasy/issues/3839)).

NB: If you want to use different GPIO pins, so the regular serial logging is available (or if the above steps won't work), then on ESP8266 you can configure the task for Software Serial, or on ESP32 configure HWSerial1 or HWSerial2, using any available GPIO pin.

betux69, 2023-03-21:
thanks a lot for your answer.

point 1 done
point 2 done
point 3  :
[image]

1234 is it good for tcp port ?

Command send : serialsend,"PWR ON"

and ... no response of the video projector ;o( grr !!

tonhuisman, 2023-03-21:
You might want to enable `Process events without client` option, but for sending that shouldn't matter.

Maybe the projector expects a `CR` or `LF` character at the end of the command, you could try the key-combination `ctrl-M` or `ctrl-J` for that, but I'm not sure if that'll work 🤔. There is only the regular special character translation available, where `%R%` and `%N%` would be converted into a `CR` and `LF` character.

betux69, 2023-03-21:
in the documentation there is : Add a Carriage Return (CR) code (0Dh)
How i can do what  ?

betux69, 2023-03-21:
i tried 

serialsend, %R% "PWR ON" %N%
serialsend, %R% PWR ON %N%
serialsend, PWR ON %N%
serialsend, PWR ON %R%

With an other software this command work 
0xOD. »PWR ON »,OxOD

i think we're not far ... i hope ;o)

tonhuisman, 2023-03-21:
> serialsend, PWR ON %R%

I would expect that this should work, but maybe there _shouldn't_ be a space before PWR and between ON and %R%. (Computers can be quite picky about these extra characters 😉)
If the CR before the command is needed, it would then be `serialsend,%R%PWR ON%R%`

Is there a way you can monitor if the sent data is actually received at the projector?

betux69, 2023-03-21:
it doesn't work.
i think your right i need to find a way to check the signal.

tonhuisman, 2023-03-21:
You can also try to switch to Software Serial, that allows to select the GPIO pins, so it is not 'hindered' by the regular Serial pins (that have a tendency to quite stubbornly keep hanging on to the default RX/TX pins, locking them for use by a plugin like Serial Server). 9600 baud should be handled correctly via Software Serial.

TD-er, 2023-03-21:
Maybe also check to see if the logic levels are correct.
The ESP may not be able to send a logic "1" higher than 3.3V and perhaps your device needs higher logic levels.

A level converter may be required here (or simply using a transistor and 5V)

betux69, 2023-03-21:
i see what you both mean, thanks !
i'm going to work on what ways, keep in touch ;o)

betux69, 2023-03-21:
Is it possible to send this : 0x50,0x57,0x52,0x20,0x4F,0x4E,0x0D  ?

tonhuisman, 2023-03-21:
> Is it possible to send this : 0x50,0x57,0x52,0x20,0x4F,0x4E,0x0D ?

But that would be exactly like sending `serialsend,POW ON^M`  (where that ^M would be you, typing the <ctrl-M> keys).

That's not supported yet, but there is a similar PR open to support that kind of command in the P087 Serial Proxy plugin, though that's not yet ready to be merged.
It would be nice to have that also in this plugin, I'll have a look at that. Probably will be a different command, for backward compatibility, where you could combine (quoted) plain text and decimal or hex values transformed to ASCII.

Did you see the sent signal being received by the projector?

betux69, 2023-03-21:
the command serialsend is link to this project ? 
https://batchloaf.wordpress.com/serialsend/

Not yet i need some stuff to test the signal

betux69, 2023-03-21:
what is serial proxy plugin ? should u use it for my case ?

TD-er, 2023-03-21:
SerialProxy was originally intended to receive strings from the serial port and forward them to a MQTT controller.
You can send data to the serial port with that plugin, but I doubt it is better suited for this than the one you're already trying to use now.

I wonder if you need to wrap the command arguments in "" since you're using spaces in the strings you try to send.
Or if you need to use "" in the string itself, you could try to wrap the command argument in other quotes like a single quote or backtick.

tonhuisman, 2023-03-21:
> I wonder if you need to wrap the command arguments in "" since you're using spaces in the strings you try to send.

No, this plugin doesn't use `parseString()`, but a simple `substring(11)` for the remainder of the arguments.

tonhuisman, 2023-03-21:
> the command serialsend is link to this project ? 
> https://batchloaf.wordpress.com/serialsend/

No, `serialsend` in ESPEasy is a complete local interpretation of that concept, implemented explicitly in this plugin.

tonhuisman, 2023-03-22:
@betux69 I've made a few small improvements to the code, so a new build is getting ready, in [this GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4490158713).
Please test this latest build once it's completed, TIA!

tonhuisman, 2023-03-24:
Looking at the scale the scope is using (5.00V), and the height of the signal level, maybe you need to use a level converter (as suggested before) to get the signal on the projector input on +/- 12V, the regular RS232 signal level. Most likely the signal you show isn't enough to be accepted/recognized by the projector.

betux69, 2023-03-24:
With you're built i have no signal when i send nothing.
good point !

betux69, 2023-03-24:
do you have an exemple of serialsendmix command ?

betux69, 2023-03-24:
we're going to see to send 12v ;o)

betux69, 2023-03-24:
i try this :
 Command unknown: serialmixt,0xOD"PWR ON"OxOD

tonhuisman, 2023-03-24:
There is an explanation in the documentation, but that Sphinx source code isn't that readable, so here's a screenshot:

[image]

betux69, 2023-03-24:
i put max3232 to 5V it's better but it doesn't work yet.
Next try monday ;o)

[image]

closed 2023-04-01 by TD-er

betux69, 2023-04-03:
Sorry to be late a lot of work.

It's working !!!!!!!!!!!!
You're awesome thanks a lot !!!!!

This would be in the next firmeware ?

BIG THINK YOU

betux69, 2023-04-03:
Gosh! i've a little problem ..
Power On is working but power off not.
How is it possible ?

i send :
serialsendmix,0x0D,"PWR OFF",0x0D

betux69, 2023-04-03:
Found 👍 

serialsendmix,"PWR OFF",0x0D
serialsendmix,"PWR ON",0x0D

PR #4559 (2023-04-01): [P020] Add `serialsendmix` command
Resolves #4558 
Closes #3850 (alternative implementation)
Closes #3980 (alternative implementation) 

Features: (cherry picked from #4283)
- Introduce `parseHexTextString()` function in StringConverter
- Add command `serialsendmix,<"text content"|0xXX>[,...]` to concatenate text and hex character values
  - Hex values can be a single byte, or multiple bytes, starting with `0x` and separated by nothing, space, dash, colon, comma, period, semicolon.
- [P087] Add command `serialproxy_writemix` with the same syntax as `serialsendmix`.
- Added `parseHexTextData()` [... cut]
PR #4283 (2023-12-29): [P044] Led settings and events without client, merged into [P020]
Resolves #4269 
Resolves #4558

**P044 - Communication - P1 WiFi Gateway**
Features:
- Configure Led pin (Enabled, Pin, Inverted)
- Merged functionality into [P020] Communication - Serial Server to avoid code duplication
  - P044 is still available, is just uses P020 with a predefined, and somewhat limited, configuration options
  - Converts existing [P044] settings to [P020] on first settings-save after update.
- Option to include P1 data in `<taskname>#Data` event (heavy on resources!)

**P020 - Communication - Serial Server** [... cut]