letscontrolit/ESPEasy #4566: Arduino IDE compiling mega-20230306 fails -> fatal error: CircularBuffer.h: No such file or directory

LuRhe, 2023-03-25:
Arduino IDE Output:

`In file included from C:\Users\LutzR\OneDrive\Documents\Arduino\portable\sketchbook\ESPEasy\src\DataStructs\PluginTaskData_base.h:7,
                 from C:\Users\LutzR\OneDrive\Documents\Arduino\portable\sketchbook\ESPEasy\_Plugin_Helper.h:13,
                 from C:\Users\LutzR\OneDrive\Documents\Arduino\portable\sketchbook\ESPEasy\ESPEasy.ino:102:
C:\Users\LutzR\OneDrive\Documents\Arduino\portable\sketchbook\ESPEasy\src\DataStructs\PluginStats.h:12:11: fatal error: CircularBuffer.h: No such file or directory
   12 | # include <CircularBuffer.h>
      |           ^~~~~~~~~~~~~~~~~~
compilation terminated.
`

**At libraries still exist a folder CircularBuffer with the CircularBuffer.h**

My complete clean way I've tried: 
- clean portable Arduino IDE install (2.0.5), 
- configured for esp8266 and ESP32 boards,
- created directory /portable/sketchbook/ESPEasy and ...libraries
- downloaded sources ...mega-20230306, 
- copied anything from src to /portable/sketchbook/ESPEasy
- copied anything from lib to /portable/sketchbook/libraries
- copied Custom-sample.h to Custom.h and uncommented #define USE_CUSTOM_H at ESPEasy_common.h
- edit Custom.h with uncommenting some plugins
- select board Wemos D1 Mini ESP32
- compiling fails

What's wrong? My goal ist to get a simple way to make a custom build with a subset of normal plugins + one Energy only plugin for an ESP32 4M.
Is there any other way to get a special build from existing unmodified plugins? Sorry, Platformio is an overkill for me as newbie.
Thanks for your advices and help.

tonhuisman, 2023-03-25:
> Platformio is an overkill for me as newbie.

This project is based on PlatformIO, and not on Arduino IDE (anymore, for many years), so the suggestion for *any* newbie is to start with PIO to avoid such issues, and make life easy 😃. Been a software developer for 40+ years, and never even tried to compile this project using Arduino IDE because of all the hacks that have to be applied to even simply *compile* it using Arduino IDE 🤯 (because of the 'special requirements' of Arduino IDE...)
https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html

tonhuisman, 2023-09-09:
This issue can be closed, as there is no more response.

closed 2023-09-09 by TD-er