letscontrolit/ESPEasy #4567: Add Texas Instruments TMP117 I²C Temperature sensor

iz8mbw, 2023-03-26:
Hi!
Is it possible to add the TMP117 Temperature sensor in ESPEasy?

Adafruit Library: https://github.com/adafruit/Adafruit_TMP117

Many thanks!

iz8mbw, 2023-03-27:
Hi @tonhuisman @TD-er I have the TMP117 chip with me so if you need test let me know.
Thanks!

tonhuisman, 2023-04-07:
I have reserved PluginID 150 for this sensor.

tonhuisman, 2023-04-08:
Added a plugin that should basically work, You can test by downloading the build from [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4647146713), available in the `Collection F`, `Climate` and `MAX` builds.

iz8mbw, 2023-04-09:
Hi @tonhuisman.
Just tested TMP117 with "ESP_Easy_mega_20230408_climate_ESP32_4M316k.bin", it works thanks!

I have found only an issue, the TMP117 does not log values on serial output like others plugin do.
In this screenshot you can see only HDC10xx (ignore wrong temp and hum values since it's disconnected):
[image]

Tested also MQTT, it works:
[image]

iz8mbw, 2023-04-09:
and is it possible to add a check box in order to disable/enable the "Raw" value?
In my case, for example, I don't need the "Raw" value in MQTT, I need only Temperature values.
Thanks.

iz8mbw, 2023-04-09:
I noticed also a strange behavior about Polling interval.
It looks like the Polling interval cannot be less than 15 seconds (or the sensor does not answer in a time less than 15 seconds).

See here the results about Polling interval set to 1 second, 5 seconds and 30 seconds:

**With 1 seconds of polling:**
```
[2023-04-09 09:06:04] espeasy/tmp/Temperature 17.61
[2023-04-09 09:06:04] espeasy/tmp/Raw 2254

[2023-04-09 09:06:20] espeasy/tmp/Temperature 17.61
[2023-04-09 09:06:20] espeasy/tmp/Raw 2254

[2023-04-09 09:06:36] espeasy/tmp/Temperature 17.62
[2023-04-09 09:06:36] espeasy/tmp/Raw 2255
```

**With 5 seconds of polling:**
```
[2023-04-09 09:04:38] espeasy/tmp/Temperature 17.57
[2023-04-09 09:04:38] espeasy/tmp/Raw 2249

[2023-04-09 09:04:53] espeasy/tmp/Temperature 17.56
[2023-04-09 09:04:53] espeasy/tmp/Raw 2248

[2023-04-09 09:05:08] espeasy/tmp/Temperature 17.56
[2023-04-09 09:05:08] espeasy/tmp/Raw 2248
```

**With 30 seconds of polling:**
```
[2023-04-09 09:07:25] espeasy/tmp/Temperature 17.63
[2023-04-09 09:07:25] espeasy/tmp/Raw 2257

[2023-04-09 09:07:55] espeasy/tmp/Temperature 17.64
[2023-04-09 09:07:55] espeasy/tmp/Raw 2258

[2023-04-09 09:08:25] espeasy/tmp/Temperature 17.64
[2023-04-09 09:08:25] espeasy/tmp/Raw 2258
```

As you can see when polling is 1 second or 15 seconds anyway the result is 15 seconds.
With 30 seconds of polling it respect the 30 seconds of interval.

tonhuisman, 2023-04-09:
Thanks for your test reports, really valuable 👍 
Logging will be added, as well as enabling/disabling the `Raw` value.
The polling is limited to the sensor returning a measurement, so it depends on the configured settings. My sensor hasn't arrived yet, so I haven't been able to actually test myself... 😊 
And also, documentation hasn't been written, but I already planned a section about this (expected) behavior.

iz8mbw, 2023-04-09:
Thanks!

Ciao.

tonhuisman, 2023-04-09:
A new build is brewing in [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4650699907), can you re-test? Best is to delete the current task, then re-add, to get the plugin-defaults set correctly.

Only thing I haven't been able to tackle is the Interval behavior, as that requires the sensor to be connected. What are your settings for `Averaging`, `Conversion mode` and `Conversion cycle` time?

TD-er, 2023-04-09:
As I also mentioned in the code review.
I don't think there should be a checkbox in this plugin for conversion to Fahrenheit.
The formula field can be used to convert to any other unit of measure.
If it is a more "professional" kind of sensor, I would think it would make more sense to have it output in Kelvin for example.

So better to have such conversions made for all instead of seemingly random per plugin.
But to have it global, we first need to add some concept of unit of measure in ESPEasy, so that's for later features.

With the concept of unit of measure, we can also have proper value checks and error values for example and set global default conversions.

iz8mbw, 2023-04-09:
In my honest opinion and based on my experience on the sensor chips, all the Temperature sensor chips provide temperature in Celsius (°C). Per datasheets the temperature ranges, errors and resolutions are all in Celsius (°C).
So, for me, in ESPEasy all Temperature should be in Celsius (°C) and if it's needed the values in Fahrenheit users can use the formula field to covert.

TD-er, 2023-04-09:
Yep.
When I worked at some company making geometric metrology CNC machines (electronics/software/firmware), we used for all internal values the SI-units. Only exception was distance, which was in mm.
Thus temperature was internal in Kelvin and for display/input purposes it could be converted to/from another unit of measure.
As a joke, I also added the conversion to light-year as unit of length to this conversion class :)

iz8mbw, 2023-04-09:
Hi again @tonhuisman, just tested last [Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4650699907).
Here my report:

Log to Serial now works, via "Log measured values (INFO)" option, thanks.
"Raw" value option added, thanks, it works enabling/disabling.

About Polling interval, leaving defaults value:
[image]

ans set an Interval to 5 seconds:
[image]

It works every 5 seconds:
```
[2023-04-09 18:18:06] espeasy/tmp/Temperature 18.62
[2023-04-09 18:18:11] espeasy/tmp/Temperature 18.62
[2023-04-09 18:18:16] espeasy/tmp/Temperature 18.62
[2023-04-09 18:18:21] espeasy/tmp/Temperature 18.62
[2023-04-09 18:18:26] espeasy/tmp/Temperature 18.62
```

**but** if I change "Conversion cycle time" to another value, such as "500 msec" like this:
[image]

the sensor starts to provide Temperature values every 16 seconds and not more every 5 seconds:
```
[2023-04-09 18:21:09] espeasy/tmp/Temperature 18.73
[2023-04-09 18:21:24] espeasy/tmp/Temperature 18.73
[2023-04-09 18:21:42] espeasy_1/sht/Temperature 17.96
[2023-04-09 18:21:59] espeasy/tmp/Temperature 18.69
```
so also if I change the "Conversion cycle time" to a smaller or bigger value anyway the sensor provide Temperature values every 16 seconds.

To "recover it" (so to have again the "Polling interval" working to 5 seconds for example) I re-set "Conversion cycle time" to 1 second (the Default value), then poweroff - poweron ESP32.
In this way the Temperature values are provided every 5 seconds:
```
[2023-04-09 18:26:59] espeasy/tmp/Temperature 18.87
[2023-04-09 18:27:04] espeasy/tmp/Temperature 18.88
[2023-04-09 18:27:09] espeasy/tmp/Temperature 18.88
[2023-04-09 18:27:14] espeasy/tmp/Temperature 18.89
```

I hope I was clear.
Thanks.

tonhuisman, 2023-04-13:
I have received my ordered sensor, so I can now also test myself.
I've fixed the issue with the interval, and also have the one-shot mode working as intended. You can test once [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4693830286) is completed

iz8mbw, 2023-04-13:
Thanks, going to test tomorrow.
Appreciated your effort.
Thanks.

iz8mbw, 2023-04-14:
Hi @tonhuisman 
just tested the new code for TMP117 and it looks like better then before but for now I found an issue: 
I got always the same previous temperature if you do thes 2 steps one after the other:
1. Switch the Conversion mode from Continuous (default) to One-Shot
2. Switch back the Conversion mode from One-Shot to Continuous

if you do these 2 steps one after the other you will se that the returned Temperature value is always the same (try to touch the sensor in order to see if temperature values changes).

To recover reading the correct temperature values: or reset the board or Switch again the Conversion mode from Continuous to One-Shot

tonhuisman, 2023-04-15:
Found and fixed the issue of not applying the changed configuration, and did some more (UI/UX) improvements. A new [Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4707153918) is available, once finished you can test again.

iz8mbw, 2023-04-15:
Ciao @tonhuisman 
Ton, now YES! Code works very well!!! Many thanks.
For me you can merge the TMP117 code into Master.
Still thanks!!

tonhuisman, 2023-04-15:
I have now correctly placed the plugin in the `Collection F` set (instead of Energy 😊), and added documentation. A new [Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4707781129) is working on that.

iz8mbw, 2023-04-15:
> I have now correctly placed the plugin in the `Collection F` set (instead of Energy 😊), and added documentation. A new [Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4707781129) is working on that.

ok but I suppose it will remain into Climate :-)

tonhuisman, 2023-04-15:
> I suppose it will remain into Climate

Yes, that hasn't changed, I just added the plugin in a different section instead of Collection F.

iz8mbw, 2023-04-15:
Great!

tonhuisman, 2023-04-15:
Here's a preview of the documentation: 
[Environment - TMP117 Temperature — ESP Easy 2.1-beta1 documentation.pdf](https://github.com/letscontrolit/ESPEasy/files/11239496/Environment.-.TMP117.Temperature.ESP.Easy.2.1-beta1.documentation.pdf)

iz8mbw, 2023-04-15:
Great!
As Hardware sensor image you can also add this: https://m.media-amazon.com/images/I/71BaVsDlL9L._SL1500_.jpg

iz8mbw, 2023-04-16:
Hi @tonhuisman decide you if/when close this Issue since you code for TMP117 in ESPEasy works well.
Still thanks for your effort!

tonhuisman, 2023-04-16:
Thanks for the feedback 👍 

This issue will be closed automatically when the linked PR is merged, so just leave it as is.

PR #4596 (2023-04-21): [P150] Add plugin Environment - TMP117 Temperature
Resolves #4567 

Features:
- Add support for TMP117 Temperature sensor
- Available in the `Collection F`, `Climate` and `MAX` builds
- Make `Raw` value optional
- Add optional logging of values (INFO level)

TODO:
- [x] Testing, testing, testing ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/4567#issuecomment-1509755039))
- [x] Log values during operation
- [x] Make `Raw` value optional
- [x] Investigate/explain interval behavior
- [x] Documentation
closed 2023-04-21 by TD-er