letscontrolit/ESPEasy #4569: Servo command failed

jeanbatfr, 2023-03-26:
Hello, I'm trying to control 1 servo with the last espeasy release on a wemos d1 mini : ESP_Easy_mega_20230306_normal_ESP8266_1M.bin

Flashing worked well. Commands like PWM are working, but the "Servo" commands return "Failed" every time.
My command : Servo,1,2,90

I tried to flash an older version but my d1 mini was delivered without reset button so it didn't reflash.
Because I don't find any similar issue on google, I guess it's a new issue from the release maybe?

What can I do? Thank you

TD-er, 2023-03-26:
The "Servo" command may not be included in builds with LIMIT_BUILD_SIZE defined, like on builds for 1M.

If you connect GPIO-0 to GND during power on, the ESP will enter flash mode, so you then can flash another build.

Do you really need the 1M build?
I don't know of any D1 mini with only 1M flash. Most have at least 4M and some even 16M of flash.

jeanbatfr, 2023-03-26:
Oh ok, I just followed an older tutorial using the 1M version without explanation.
Work great now 
thank you

TD-er, 2023-03-26:
Great!

closed 2023-03-26 by TD-er