letscontrolit/ESPEasy #4570: ADS1115 channel results mixed

UserRS232, 2023-03-26:
Hi, 
I have a Wemos D1 with ESPEasy_mega_20230306_ESP82xx_binaries
connected to an ADS1115 via I2C.  ADS1115 VCC: 5V

Task1: Device ADS1115
Name: V1
Address 0x48
Gain 1x
Input : AIN0-GND
Interval: 1s
Pin AIN0 connected to 5V

Task2: Device ADS1115
Name: V2
Address 0x48
Gain 1x
Input : AIN1-GND
Interval: 1s
Pin AIN1 connected to GND

Now the Problem:
Task1 Enabled and Task2 Disabled  = Task1 Value 32767   --> as expected
Task1 Disabled and Task2 Enabled  = Task2 Value 0           --> as expected
Task1 Enabled and Task2 Enabled  = Task1 Value 0  and Task2 Value 32767 --> ???? Why ????

If i set the Interval from Task2 to 5sek the Log show
9146598: EVENT: V1#Analog=32767.00     --> correct
9147538: EVENT: V1#Analog=32767.00     --> correct
9148540: EVENT: V1#Analog=32767.00     --> correct
9149540: EVENT: V1#Analog=32767.00     --> correct
9150212: EVENT: V2#Analog=32767.00     --> wrong
9150545: EVENT: V1#Analog=0.00             --> wrong
9151540: EVENT: V1#Analog=32767.00     --> correct

it seems that if more than 1 value is read at the same time, the value is output to the wrong task.

I have tested another Wemos D1, and another ADS1115 --> Same Problem
If I read out the values of AIN2 and AIN3 with Task 3 and 4, the error continues --> the measured value is in the wrong task

Now I'm going to test earlier versions... does anyone have an idea of what could be the reason?

UserRS232, 2023-03-26:
All versions back to 20210223 have the same issue...

tonhuisman, 2023-03-26:
> does anyone have an idea of what could be the reason?

The multiplexer in the ADS1115 is reprogrammed when the read is started, and as there is some time-delay involved during the reading process, the next task does possibly get some time to execute, starting with the re-programming of the multiplexer, so the actual read is then returning the unexpected result.

This can probably only be solved by introducing some kind of lock mechanism when the read process is started, and released when that's completed.

TD-er, 2023-03-26:
This has been fixed in a PR which has been merged quite recently (less than a week ago??)
So please test with one of the latest GH Actions builds.

For example this one: https://github.com/letscontrolit/ESPEasy/actions/runs/4515724002

TD-er, 2023-03-28:
Can you check this PR: https://github.com/letscontrolit/ESPEasy/pull/4576
Test build (still running): https://github.com/letscontrolit/ESPEasy/actions/runs/4544252976

TD-er, 2023-03-28:
> Pin AIN0 connected to 5V

According to the datasheet, you should not go over Vdd + 0.3V
What voltage is the ADS1115 running at?

tonhuisman, 2023-03-31:
> With this build I don't get any values displayed at all... always 0.00

Have tested with the latest code changes, and the plugin is (still) showing the expected results here. (Working with 3.3V though, to protect my I2C GPIO-pins.)

PR #4576 (2023-03-31): [ADS1115] Fix channel results mixed #4570
Fixes: #4570
commit a3dc1c6d (2023-03-31): [ADS1115] Fix channel results mixed #4570

Fixes: #4570
closed 2023-03-31 by TD-er