letscontrolit/ESPEasy #4571: Occasionally values error of HDC1080 sensor (device)

iz8mbw, 2023-03-26:
Hello.
Using ESP_Easy_mega_20230306_climate_ESP32_4M316k on my ESP32 board that publish temperature and humidity data via MQTT.
I use the device (sensor) "Environment - HDC1080 (I2C)" and sometimes I have temperature values error: 125 °C and -40 °C.
No issues on Humidity values.

See here:
[image]

The high peak you see is 125 °C, the low peak you see is -40 °C.

As polling interval I set 60 seconds (so not too much, I don't like to hammer sensors):
[image]

Tell me if you need more details.

Many thanks.

tonhuisman, 2023-03-26:
Can you turn _off_ the option `Check I2C devices when enabled` on the Tools/Advanced page, and see if this still happens? If that fixes it, I'll disable this sensor from that check too, as the check might interfere with the regular measuring sequence (it seems to do that on some sensors).

iz8mbw, 2023-03-26:
OK, `Check I2C devices when enabled` option disabled, let's see what happens in the future.
Thanks for now!

iz8mbw, 2023-03-28:
Hi.
With `Check I2C devices when enabled` option disabled I don't have more reading issue.
Thanks for your support and thanks for ESPEasy!

PR #4575 (2023-03-29): Disable I2C dev check HDC1080  #4571
Fixes: #4571
Fixes: #4578
commit 20d35393 (2023-03-29): Disable I2C dev check HDC1080  #4571

Fixes: #4571