letscontrolit/ESPEasy #4578: Rename HDC1080 with HDC10x0

iz8mbw, 2023-03-29:
Hello.
According to Texas Instruments datasheets, the Humidity and Temperature sensor HDC1080 is software compatible with these other Texas Instruments sensors:

HDC1000, HDC1008, HDC1010, HDC1050

In my case, in fact, I use a HDC1010 sensor using the HDC1080 "device" in ESPEasy.

So, to avoid confusion and to extend the possibility to use different sensor, I suggest to rename HDC1080 with HDC10x0.

Thanks.

TD-er, 2023-03-29:
Do you have a version of a datasheet where TI states compatibility?

TD-er, 2023-03-29:
I meant some document stating these can be controlled in the same way.
Otherwise I have to go through all datasheets.

I did find this though: https://lore.kernel.org/lkml/1497278950-12973-2-git-send-email-m-stecklein@ti.com/

commit 32de7881 (2023-03-29): Rename HDC1080 to HDC10xx #4578

Apparently it supports HDC1000/HDC1008/HDC1010/HDC1050/HDC1080

Fixes: #4578
PR #4575 (2023-03-29): Disable I2C dev check HDC1080  #4571
Fixes: #4571
Fixes: #4578
closed 2023-03-29 by TD-er