letscontrolit/ESPEasy #4584: Wifi setup on page main does not use the default css settings

uwekaditz, 2023-04-02:
Using the latest mega (authored 2.4.2023 @ 16:53) does not use the css settings if the button `WiFi Setup` ist clicked on the page `Main`. 
The `WiFi scan` on the page `Tools `looks like expected!
The `Web light/dark mode` is set to auto.
Main page:
[image]
WiFi Setup page:
[image]

TD-er, 2023-04-02:
Yep, had this myself a few times.
For the setup page I guess we really need to have the CSS inline.

tonhuisman, 2023-04-02:
> For the setup page I guess we really need to have the CSS inline.

That will depend on the `FEATURE` flags defined... and seeing `Uwes Build` in the Git Build line, it might be using a flag that's disabling the inlining of the CSS.

TD-er, 2023-04-02:
If the CSS is served from CDN servers, then it isn't a problem.
It is very likely already cached in the browser or else it will then be downloaded.

However fetching a 2nd file from an ESP host that's no longer active on that channel will extremely likely not reply to this extra call :)

tonhuisman, 2023-04-02:
Looking a bit closer, it is more likely that the CSS is not loaded at all for that page. Browsing to the `/setup` page on a connected unit (4M1M Collection A build) also doesn't use the styling, though other pages do use that.

uwekaditz, 2023-04-02:
I used the standard build provided by `custom_IR_ESP8266_4M1M`. The css style works on all pages except the WiFi setup.

PR #4639 (2023-05-05): [UI/UX] Setup page doesn't serve css
Resolves #4584 

Bugfix:
- On the `/setup` page no css was served, giving a rather poor user experience. Now css is served if possible/available (may require access to the internet from the browser or the css already in browser-cache).
closed 2023-05-05 by TD-er