letscontrolit/ESPEasy #4586: C011 configuration page changed not accurately

popaserge, 2023-04-02:
It is no more possible to expand the header and body windows to the right. This is not comfortable if the string to write is long.

[image]

popaserge, 2023-08-19:
The bug was found in a self build version.
With the help of Firefox Web development tools and https://www.w3schools.com/cssref/index.php I happened to understand the power of css files.

Using a static esp.css downloaded to a device I could test several CSS statements and succeded to correct the bug.

Hereafter the statements I changed or added if that helps to improve the online espeasy_default.css.

Line 162
.button.link.wide {
	display: inline-block;
	text-align: center;
	width: 100%;
}

Line 195
.wide {
	margin-bottom: 5px;
	max-width: 1000px;
	min-width: 85px;
	
}
I added also following statements
select.wide {
	margin-bottom: 5px;
	max-width: 300px;
	min-width: 85px;
	width: 80%
}

input.wide {
	margin-bottom: 5px;
	max-width: 500px;
	min-width: 85px;
	width: 80%
}
[image]

tonhuisman, 2023-08-19:
I didn't yet get to this, but now that you've provided fixes, I'll add these to a PR that's still open

tonhuisman, 2023-08-20:
@popaserge I've only applied the `max-width` change to the `textarea` fields, by overriding _only_ that attribute (as CSS is intended 😃).
I have not applied the other suggested `input` width change, as that would make other settings pages (Devices) look somewhat messy, though I do understand why you'd want that changed, long input values a mostly an exception.

PR #4767 (2023-08-20): [Generic] Improvements and fixes part 2, Aug 2023
Features:
- Some small improvements and fixes:
  - [AdafruitGFX_Helper] Log an error if the `bmp` file to display doesn't match the required signature
  - [P113] Correct typo in debug log statement
  - [UI] Widen `textarea` fields for easier input of larger/wider text

Resolves: #4586
closed 2023-08-20 by TD-er