letscontrolit/ESPEasy #4588: I2C & Nextion

betux69, 2023-04-03:
Hello,

I would like to use a touch screen with i2C.
I work on nextion but it's tx/rx and it doesn't to exist with I2C.
Do you have an other touch screen support by espeasy ?

Thanks

tonhuisman, 2023-04-03:
The only currently supported touchscreen is the XPT2048 in plugin [P099](https://espeasy.readthedocs.io/en/latest/Plugin/P099.html#p099-page) included in the Display and MAX builds, but that uses the SPI interface. It is often mounted on ILI934x/ILI948x TFT displays using plugin [P095](https://espeasy.readthedocs.io/en/latest/Plugin/P095.html#p095-page) that also use the SPI interface.
For I2C touchscreen support there is an open PR #4016 to support the FT6xxx touch modules (plugin P123), that can be found on the [WT32-SC01](https://github.com/letscontrolit/ESPEasy/issues/3831) units (3.5" screen) and M5Stack M5Core2 units (2" screen) (and probably some other M5Stack units).

I've also planned to support the GT911 touchscreen, connected via I2C, as available in the [TTGO Lily Pi ESP32](http://www.lilygo.cn/prod_view.aspx?TypeId=50032&Id=1353) (3.5" screen), that's going to get the same configuration options as P123, which is also to be made available for P099, but this is all still in the planning phase 😅.

betux69, 2023-04-03:
thanks for your answer.
Question : How many tx/tr component can i use on esp8266 ?

tonhuisman, 2023-04-03:
> Question : How many tx/tr component can i use on esp8266 ?

Well, that depends, when disabling the serial log (Uncheck the `Enable Serial Port` on the Tools/Advanced page, so it can be used for plugins) and using another task configured for Software Serial, you can have 2 devices using serial, but combining multiple Nextion tasks, with Software Serial can possibly exhaust the resources (both memory and CPU) of an ESP8266. [see the docs](https://espeasy.readthedocs.io/en/latest/Plugin/P075.html#serial-communication)
There is also the option of using an I2C serial board (usually has 2 channels). But still the limitation of resources can be an issue.

TD-er, 2023-04-03:
ESP8266 does have 2 hardware serial ports, however "Serial1" is "TX-only" so hardly ever usable.
If you need upto 3 HW serial ports, I suggest looking for ESP32.
ESPEasy also supports the NXP SC16IS752 chip, which does have 2 HW serial ports and is connected to ESPEasy via I2C.
Everywhere in ESPEasy where you can select a serial port, you can also select such an I2C port.