letscontrolit/ESPEasy #4589: Waveshare 7.5inch E-Paper HAT Display Module, 800×480

fly74, 2023-04-05:
Hi Devs,

is this  [Cool](https://www.amazon.de/Waveshare-Resolution-Supports-Compatible-Raspberry/dp/B0BFQKDRLR) supported by a plugin?

TD-er, 2023-04-05:
Hmm those are indeed really cool, but also rather expensive.
Not sure what has to be done to make it work (if any), but I guess Ton will probably look into this later today :)

fly74, 2023-04-05:
@tonhuisman Maybe you have a look.

tonhuisman, 2023-04-05:
I already looked at these kind of displays, and there are even models that support additional (1, 2, 3, 4, 7 or 8, and 15) (mostly primary) colors, but the trouble with the libraries is that a) the controller is often unique for each resolution and use quite complex (much data) initialization sequences, and b) there is a relatively large in-memory pixel-buffer _for each color_, that is to be filled with the data to display, that's then sent to the controller/display in a batch operation. Especially the second 'feature' makes it pretty hard to use these displays with an ESP8266 that's running ESPEasy.
I already have encountered similar issues while extending P096 with some extra displays in PR #4223, causing that PR to be delayed until I have found/designed a solution for that.
Because of these resource-related issues, I assume to be the reason that most of these displays come with 'native' RPi HATs, though technically they _can_ run on any Arduino-like controller, like ESPs.
And I surely want to _avoid_ having a separate plugin for each of the many available display resolutions and color options.

TD-er, 2023-04-05:
For B/W (1 bit) displays, the amount of memory for the frame buffer is already 100x480 bytes = 48k
That's for sure out of ESP8266's reach.

Maybe, if the display supports partial updates, you could do it without a full frame buffer, but that does limit the use of the display.

fly74, 2023-04-05:
Which resolution and display size is already supported?

fly74, 2023-04-05:
[this one](Waveshare Tri-Color 1.54 Inch E-Ink Display Module Three-Color Resolution 200x200 E-Paper Electronic Screen Panel with Embedded Controller SPI Interface Support for Raspberry Pi/Jetson Nano,etc https://amzn.eu/d/eJCyYgO) ?

tonhuisman, 2023-04-05:
Some of these 200x200 Waveshare displays are intended to be included in the PR mentioned above...

> Which resolution and display size is already supported?

``` C++
/****************************************************************************
 * toString: Display-value for the EPD/eInk device selected
 ***************************************************************************/
const __FlashStringHelper* toString(EPD_type_e device) {
  switch (device) {
    case EPD_type_e::EPD_IL3897: return F("IL3897 (Lolin 250 x 122px)");
    case EPD_type_e::EPD_UC8151D: return F("UC8151D (white/black/red 212 x 104px)");
    case EPD_type_e::EPD_SSD1680: return F("SSD1680 (250 x 212px)");
    # if P096_USE_WAVESHARE_2IN7
    case EPD_type_e::EPD_WS2IN7: return F("Waveshare 2.7\" (264 x 176px)");
    # endif // if P096_USE_WAVESHARE_2IN7
    # if P096_USE_WAVESHARE_1IN54B
    case EPD_type_e::EPD_WS1IN54B: return F("Waveshare 1.54\" (200 x 200px)");
    # endif // if P096_USE_WAVESHARE_1IN54B
    # if P096_USE_MH_ET_LIVE_1IN54
    case EPD_type_e::EPD_MHET1IN54: return F("MH-ET Live 1.54\" (white/black/red 200 x 200px)");
    # endif // if P096_USE_MH_ET_LIVE_1IN54
  }
  return F("Unsupported type!");
}
```

fly74, 2023-04-06:
I will order one for testing.

closed 2023-04-06 by fly74