letscontrolit/ESPEasy #4592: Additional I²C bus (Master)

iz8mbw, 2023-04-07:
Hello.
For what I know currently ESPEasy supports only one I²C master bus.
In fact in the Hardware tab we can specify the GPIO pins to assign to the I²C master bus:
[image]
but always referred to only one bus.

For what I know the ESP32 board supports also a second I²C bus (Master).

Is it possible to add or enable the second I²C bus (Master) in ESPEasy using so other two GPIO pins?

Thanks!

TD-er, 2023-04-07:
Nope, that's not (yet) possible as it requires all (!!!) I2C related code to be changed and reviewed to make sure the right I2C bus is being used.

We do support I2C extenders if you need to use multiple I2C devices which would otherwise cause an address conflict.
Other use cases for I2C extenders:
- Mix 5V and 3V3 I2C devices, since each output of such extender needs its own pull-up resistors and is essentially a separate bus.
- Mix (old) I2C devices which really don't like high frequency clocks, or need a much lower clock due to use of long wires.

tonhuisman, 2023-04-07:
> Is it possible to add or enable the second I2C bus (Master) in ESPEasy

What's your use-case for having a second I2C bus? As mentioned by TD-er, the technical limitations of some I2C devices can be resolved by using an I2C multiplexer. (I know 1 use-case, but I'll not pollute your opinion with that 😺).

I've been thinking about this also, but the expected amount of work involved has stopped me from working on it, so far.

iz8mbw, 2023-04-12:
yes, the user case is always the same: same chip with same I2C address.

TD-er, 2023-04-12:
So that's already supported via an I2C extender.

Now that I think of it, a 2nd port is something like a virtual I2C extender.
But still that doesn't change the fact that all I2C related code must be inspected to make sure it will work using the set port.

tonhuisman, 2023-04-12:
> yes, the user case is always the same: same chip with same I2C address.

Then you can use an I2C Multiplexer, as ESPEasy has support for 4 (closely related) models. This support however is not included in all builds (most notably, not in the Collection builds...).

iz8mbw, 2023-04-12:
yes guys, to have an integrated additional I2C bus in ESPEasy it's more easy to have another chip that act as I2C Multiplexer...
More I think about to I2C clock speed, with two different bus you can have two different (and so independent) clock speed.

Anyway if is a mess to include an additional I2C bus in ESPEasy dont' care.

tonhuisman, 2023-04-12:
> with two different bus you can have two different (and so independent) clock speed.

ESPEasy already switches the clock speed per (I2C) task, when needed, and that seems to work fine for most sensors, though there will always be sensors that just fail to work correctly when the higher clock speed is used, even when they aren't addressed. That will also be 'taken care of' by the multiplexer, AFAIK.

TD-er, 2023-04-12:
The only I2C feature which will not work with an I2C multiplexer is switching between clock speeds > 1 MHz and slower.
This for the simple reason that those are actually different and have different requirements for cable capacitance.

iz8mbw, 2023-04-12:
> needed, and that seems to work fine for most sensors, though there will always be sensors that jus

Sorry but if the bus is only one (the same one), how does is it possible to specify different bus clock speed (using Force Slow I2C speed) per device?
For example, if I have three I2C devices (sensors) on the same bus that is (by default) at 400 kHz and for one of these three devices I set "Force Slow I2C speed" (so to 100 kHz) I suppose the bus will go to down from 400 kHz to 100 kHz so the "Force Slow I2C speed" setting will affect all the sensors connected to the I2C bus, since the bus is only one and the same one.

tonhuisman, 2023-04-12:
The clock speed is switched just before accessing the intended sensor, and switched back after the sensor was accessed (only if it was changed before accessing). And the same goes for using a multiplexer, the correct channel(s) is/are enabled just before accessing the sensor, and switched back after that's done. That's working fine in like 99.99% of all cases.

tonhuisman, 2023-04-12:
> This support however is not included in all builds (most notably, not in the Collection builds...).

My mistake, I forgot that I enabled I2C Multiplexer support in nearly all builds for ESP's with 4M Flash (that didn't have this included yet) a couple of months ago, so you should be able to use it.

PR #5235 (2025-03-26): [I2C] Add support for multiple I2C buses
Resolves: #4592 

Features:
- Add support for multiple I2C buses (interfaces) (ESP32 only!), default 2, max. 3.
  - Uses software-multiplexing, to avoid having to change (and test!) many libraries and plugins.
  - Explicitly disabled for ESP8266.
- Also enables to have an optional I2C multiplexer per I2C bus.
- Per directly addressed I2C hardware (External Time Source, WatchDog hardware, PCF/MCP addressing without task like `monitor` command) an I2C Bus selector is available.

Extra fixes/features:
- Some selectors appended undesired `(default)` for `-  [... cut]
closed 2025-03-26 by TD-er