letscontrolit/ESPEasy #4593: WiFi#APmodeEnabled event depends on Serial Log Level setting

chemmex, 2023-04-07:
I am trying to visualize WiFi states with single WS2812 LED, basically I need to indicate when fresh unit with unconfigured WiFi enters AP mode.  I found that with 20230306 release (and also last github sources) WiFi#APmodeEnabled event generates only in case if Serial logging is enabled. When I disable the logging, the last event I can see (based on LED indication) is WiFi#APmodeDisabled. The AP, in fact, is accessible.

chemmex, 2023-04-07:
Below is the startup log with Serial logging enabled. Last Neopixel command corresponds to AP mode enabled state.

tonhuisman, 2023-04-07:
Found and fixed 😃 you can test using [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4640006316) once that's finished.

closed 2023-04-07 by TD-er

chemmex, 2023-04-08:
It works, thanks!

PR #4594 (2023-04-07): [WiFi] APmodeEnabled event not generated when logging is disabled #4593
Resolves #4593 

Event `WiFi#APmodeEnabled` was not generated when logging was below INFO level (ERROR or None).