letscontrolit/ESPEasy #4595: [ESP32-S2] Only two hardware UARTS available

flashmark, 2023-04-08:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

When using the generic serial library for ESP32-S2 it allows to configure 3 UARTs. This is correct for the ESP32, but for ESP32-S2 only two hardware UARTs are available. Note that the first two UARTs do work when configured. Configuring the 3th UART is possible, but does not work nor gives an error. This is inconvenient for the user who does not know these hardware details.

### Expected behavior

Drop-down menu should be limited to two physical UARTS (HW Serial0, HW Serial1)

### Actual behavior

Dorp-down menu shows three UARTs ((HW Serial0, HW Serial1), HW Serial2)

### Steps to reproduce
Select a task using serial communication and select the serial port using the drop-down menu.

It is always reproducing

### System configuration

Hardware: Wemos ESP32-S2 mini module

ESP Easy version: Custom build ESP32-S2 Mega branch 18 march 2023

ESP Easy settings/screenshots:

TD-er, 2023-04-08:
Huh?
Do you have an example of a plugin which allows for the ESP32-S2 to set a 3rd serial port?

TD-er, 2023-04-08:
Just took out an ESP32-S2 and indeed you can select a "Serial2"

That's indeed a bug and I thought I had limited this...

TD-er, 2023-04-08:
Can you test this PR: https://github.com/letscontrolit/ESPEasy/pull/4597
GitHub Actions test build: https://github.com/letscontrolit/ESPEasy/actions/runs/4647204358

flashmark, 2023-04-10:
Thanks for the quick fix. It works with the P144 Vindrikning which uses the standard libraries.

TD-er, 2023-04-10:
Thanks for testing, I will merge it.

PR #4597 (2023-04-10): [ESP32-S2] Fix selection of only 2 HW serial ports ( #4595)
Fixes: #4595
commit 69400196 (2023-04-10): [ESP32-S2] Fix selection of only 2 HW serial ports ( #4595)

Fixes: #4595
closed 2023-04-10 by TD-er