letscontrolit/ESPEasy #4599: ESP_Easy_mega_20230409_normal_ESP32_4M316k Dummy Device not firing an event

fly74, 2023-04-09:
Hi Devs,

in ESP_Easy_mega_20230409_normal_ESP32_4M316k the dummy devices not firing an event:

Rule is:

```
On [WZD#Dummy] Do
  
  If [WZD#Dummy] <36
    NeoPixelLine,5,5,0,255,0
    Else
      If [WZD#Dummy] <86
        NeoPixelLine,5,5,255,136,0
      Else
        NeoPixelLine,5,5,255,0,0
      Endif
  Endif
  
Endon
```

[image]
[image]

The Neo ist not the problem, manual set of the Neo works.

Sorry for reporting an issue soon after release.

fly74, 2023-04-09:
workes in ESP_Easy_mega_20230306_normal_ESP32_4M316k

TD-er, 2023-04-09:
```
On WZD#Dummy Do
  If [WZD#Dummy] <36
    NeoPixelLine,5,5,0,255,0
    Else
      If [WZD#Dummy] <86
        NeoPixelLine,5,5,255,136,0
      Else
        NeoPixelLine,5,5,255,0,0
      Endif
  Endif
Endon
```

The on....do line should not have []
There was some "fix" in the code to remove those, but this should not be there as it is incorrect syntax.
It also broke any option to use variables in event blocks where you want to act on a specific time defined by variables.

fly74, 2023-04-09:
Okay, you're right.

closed 2023-04-09 by fly74

TD-er, 2023-04-09:
Thanks for testing the new builds so quickly (as always :) )

fly74, 2023-04-09:
The batch updater flashes easily my grid. So it shows early if there is something.

TD-er, 2023-04-09:
I didn't have time yet to update the web flasher page... 
So you were quick in testing :)

fly74, 2023-04-09:
You guys are quick in fixing.