letscontrolit/ESPEasy #4601: About 1-Wire (also know as DS18b20)

iz8mbw, 2023-04-10:
Hi my friends.
Doing some test on 1-Wire.
First of all, in my option, to give name "DS18b20" to the 1-Wire implementation in ESPEasy is "limitative".
The Plugin "DS18b20" should be renamed to "1-Wire" because as you well know, the Plugin "DS18b20" is an implementation of 1-Wire protocol and not only an implementation of a sigle chip called "DS18B20".

In my case, for example, I'm testing on ESPEasy a MAX31826 chip in Parasite Power and it works.
[image]

Well, according to what I see, my Temperature sensor for ESPeasy is a DS1825 but it's not true, I'm using a MAX31826.
Giving a look on Internet about the 1-Wire "Family Code" I can see that can be different sensors for the same "Family Code".

So, I suggest to specify (rename) in ESPEasy the "chip combinations" for each Family Code.
I found these links about Family Codes:

https://interfaceinnovations.org/onewirefamilycodes.html
https://owfs.org/index_php_page_family-code-list.html

So for example, "DS1825" should be renamed to "DS1825, MAX31826, MAX31850" since more chip can have the same Family Code.

About "Resolution" I found some of strange, see here:
[image]

it looks like for ESPEasy my chip has 9 bit of Resolution, but it's not true since MAX31826 (and also DS1825) has (also) 12 bit of resolution.
If I change Resolution from 12 bit to 9 bit, I always have `Temperature: Error!`, see here:
```
2837514 : Info   : DS   : Temperature: Error! (3b-14-fe-14-00-00-00-17 [DS1825])
2842515 : Info   : DS   : Temperature: Error! (3b-14-fe-14-00-00-00-17 [DS1825])
2847521 : Info   : DS   : Temperature: Error! (3b-14-fe-14-00-00-00-17 [DS1825])
```

if I change back to 12 bit of Resolution I see in console the error `Error  : DS   : Cannot set resolution`, see here:
```
autoselect' length: 2 8: 'dallas_addr0' length: 2 9: 'res' length: 2 10: 'err' length: 1 11: 'pconf_2' length: 1 12: 'TDT' length: 1 13: 'TDVN1' length: 11 14: 'TDF1' length: 0 15: 'TDVD1' length: 1 16: 'edit' length: 1 17: 'page' length: 1
2911227 : Error  : DS   : Cannot set resolution
2911230 : Info   : SaveToFile: TaskSettings index: 1
2911231 : Info   : SaveToFile: free stack: 7040
2911332 : Info   : FILE : Saved config.dat offset: 34816 size: 536
2911333 : Info   : SaveToFile: free stack after: 7040
2911356 : Info   : SaveToFile: free stack: 7168
2911426 : Info   : FILE : Saved config.dat offset: 0 size: 3024
2911427 : Info   : SaveToFile: free stack after: 7168
2911428 : Info   : Skip saving SecuritySettings, not changed
2911440 : Info   : mDNS : Starting mDNS...
2911445 : Info   : mDNS : Started, with name: espeasy.local
2911887 : Info   : static_file: /esp.css
2912459 : Info   : DS   : Temperature: 18.19 (3b-14-fe-14-00-00-00-17 [DS1825])
2917462 : Info   : DS   : Temperature: 18.25 (3b-14-fe-14-00-00-00-17 [DS1825])
2919099 : Info   : WD   : Uptime 49 ConnectFailures 0 FreeMem 203880 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
2922467 : Info   : DS   : Temperature: 18.19 (3b-14-fe-14-00-00-00-17 [DS1825])
2927469 : Info   : DS   : Temperature: 18.19 (3b-14-fe-14-00-00-00-17 [DS1825])
```

but I can read Temperature:
```
2952488 : Info   : DS   : Temperature: 18.31 (3b-14-fe-14-00-00-00-17 [DS1825])
2957491 : Info   : DS   : Temperature: 18.44 (3b-14-fe-14-00-00-00-17 [DS1825])
2962495 : Info   : DS   : Temperature: 18.50 (3b-14-fe-14-00-00-00-17 [DS1825])
2967497 : Info   : DS   : Temperature: 18.56 (3b-14-fe-14-00-00-00-17 [DS1825])
2972500 : Info   : DS   : Temperature: 18.56 (3b-14-fe-14-00-00-00-17 [DS1825])
2977505 : Info   : DS   : Temperature: 18.56 (3b-14-fe-14-00-00-00-17 [DS1825])
2979095 : Info   : WD   : Uptime 50 ConnectFailures 0 FreeMem 205228 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
2982507 : Info   : DS   : Temperature: 18.56 (3b-14-fe-14-00-00-00-17 [DS1825])
2987510 : Info   : DS   : Temperature: 18.62 (3b-14-fe-14-00-00-00-17 [DS1825])
2992515 : Info   : DS   : Temperature: 18.62 (3b-14-fe-14-00-00-00-17 [DS1825])
2997515 : Info   : DS   : Temperature: 18.62 (3b-14-fe-14-00-00-00-17 [DS1825])
3002517 : Info   : DS   : Temperature: 18.69 (3b-14-fe-14-00-00-00-17 [DS1825])
3009363 : Info   : DS   : Temperature: 18.69 (3b-14-fe-14-00-00-00-17 [DS1825])
3009374 : Info   : WD   : Uptime 50 ConnectFailures 0 FreeMem 205228 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
3012529 : Info   : DS   : Temperature: 18.69 (3b-14-fe-14-00-00-00-17 [DS1825])
3017529 : Info   : DS   : Temperature: 18.69 (3b-14-fe-14-00-00-00-17 [DS1825])
```

So, as told before, I suggest to:
1. Rename the Plugin "DS18b20" to "1-Wire"
2. Update the "Family Code" list (also adding more chip for the same Family Code)

and please give a look at the "Resolution" error may be there is a more recent 1-Wire library to update in ESPEasy,

I'm available to test if needed.

Thanks!

TD-er, 2023-04-10:
There is already some 1-Wire counter plugin and 1-wire iButton, so if there's going to be a rename, it will be something like "1-Wire Temperature"

The resolution is stored on the sensor and read back from the sensor. At least on the DS18b20's
It is very well possible those newer ones behave differently, but I don't have those, so I can't really test what those are doing.

Maybe if you don't set to automatically use the 1st one, but explicitly define one, it does behave differently?
If so then this should be looked into in ESPEasy as this might also not work as expected on the DS18b20's.  (at least not on the fake/clone ones)
Can you test this by unchecking this checkbox and save with an actual sensor selected?

iz8mbw, 2023-04-10:
I have disabled "Auto Select Sensor" and the behavior is the same if I specify 9 bit of resolution: `[Temperature: Error! (3b-14-fe-14-00-00-00-17 [DS1825])`

When I come back to 12 bit of resolution it starts to work again but always having the message `Error  : DS   : Cannot set resolution`

According to the datasheet it's 12 bit of resolution so can make sense that is impossible to "force" it to 9 bit.
But it's strange why I see always 9 bit of Resolution in ESPEasy:
[image]

TD-er, 2023-04-10:
As far as I can remember, the reported resolution is read from the sensor.
But it is also possible that failing to set the resultion results in a revert to the default, which is the default for the DS18b20.

Problem is that there are lots and lots of fake sensors out there, so there are quite some extra checks to see how the sensor operates to make even the fake ones work.

iz8mbw, 2023-04-10:
changed to 4 decimal values and I can confirm the resolution is 12 bit:
```
4793395 : Info   : DS   : Temperature: 18.4375 (3b-14-fe-14-00-00-00-17 [DS1825])
4798254 : Info   : DS   : Temperature: 18.4375 (3b-14-fe-14-00-00-00-17 [DS1825])
4810187 : Info   : DS   : Temperature: 18.3750 (3b-14-fe-14-00-00-00-17 [DS1825])
4815189 : Info   : DS   : Temperature: 18.3125 (3b-14-fe-14-00-00-00-17 [DS1825])
4820194 : Info   : DS   : Temperature: 18.3125 (3b-14-fe-14-00-00-00-17 [DS1825])
4825196 : Info   : DS   : Temperature: 18.3125 (3b-14-fe-14-00-00-00-17 [DS1825])
4830200 : Info   : DS   : Temperature: 18.3750 (3b-14-fe-14-00-00-00-17 [DS1825])
4835203 : Info   : DS   : Temperature: 18.3750 (3b-14-fe-14-00-00-00-17 [DS1825])
4840205 : Info   : DS   : Temperature: 18.4375 (3b-14-fe-14-00-00-00-17 [DS1825])
4845209 : Info   : DS   : Temperature: 18.4375 (3b-14-fe-14-00-00-00-17 [DS1825])
4850213 : Info   : DS   : Temperature: 18.4375 (3b-14-fe-14-00-00-00-17 [DS1825])
4855214 : Info   : DS   : Temperature: 18.4375 (3b-14-fe-14-00-00-00-17 [DS1825])
4860216 : Info   : DS   : Temperature: 18.3750 (3b-14-fe-14-00-00-00-17 [DS1825])
4865223 : Info   : DS   : Temperature: 18.3750 (3b-14-fe-14-00-00-00-17 [DS1825])
4870228 : Info   : DS   : Temperature: 18.3750 (3b-14-fe-14-00-00-00-17 [DS1825])
4875232 : Info   : DS   : Temperature: 18.4375 (3b-14-fe-14-00-00-00-17 [DS1825])
4880238 : Info   : DS   : Temperature: 18.4375 (3b-14-fe-14-00-00-00-17 [DS1825])
4885236 : Info   : DS   : Temperature: 18.5625 (3b-14-fe-14-00-00-00-17 [DS1825])
4890239 : Info   : DS   : Temperature: 18.5625 (3b-14-fe-14-00-00-00-17 [DS1825])
4895241 : Info   : DS   : Temperature: 18.5625 (3b-14-fe-14-00-00-00-17 [DS1825])
```

uzi18, 2023-04-10:
```
For parasite-powered DS1825s, the master must enable a strong pullup on the 1-Wire bus during temperature
conversions and copies from the scratchpad to EEPROM. No other bus activity may take place during this
time.
```
https://www.analog.com/media/en/technical-documentation/data-sheets/DS1825.pdf
Page 12

TD-er, 2023-04-10:
We don't support parasite-powering of 1-Wire sensors.

iz8mbw, 2023-04-10:
> We don't support parasite-powering of 1-Wire sensors.

But it seems to work in ESPEasy

iz8mbw, 2023-04-10:
> ```
> For parasite-powered DS1825s, the master must enable a strong pullup on the 1-Wire bus during temperature
> conversions and copies from the scratchpad to EEPROM. No other bus activity may take place during this
> time.
> ```
> 
> https://www.analog.com/media/en/technical-documentation/data-sheets/DS1825.pdf Page 12
> 
> So it could be possible to fail with set resolution.

I use MAX31826 not DS1825

uzi18, 2023-04-10:
MAX31826 does not support write scratchpad command. It's fixed to 12bit.
It can be eventually detected by checking bytes in scratchpad. Bytes 2,3,5,6,7 should be 0xff and bits from 7 to 4 of byte 4 are always ones. 

https://www.analog.com/media/en/technical-documentation/data-sheets/MAX31826.pdf

TD-er, 2023-04-10:
> > We don't support parasite-powering of 1-Wire sensors.
> 
> But it seems to work in ESPEasy

Only for 1 sensor per ESPEasy node and only when using strong pull-ups.
Just browsed through the datasheet linked by @uzi18 and there are lots and lots of notes about parasite-powered use that even though it seems to work, my official standpoint is that it isn't supported.

When using multiple 1-Wire sensors in the same task, the reading is performed interleaved.
Meaning all sensors in the same task will be sent a command to start taking a message and the task will re-schedule itself to collect those measurements.
Any next task running this plugin may also start the measurement when it sees fit, which may be during the measurement period of another task.
So this is the reason why it will absolutely not work when using >1 of such sensors.
It might work when using only 1, but like I said, not supported.

TD-er, 2023-04-10:
> MAX31826 does not support write scratchpad command. It's fixed to 12bit. It can be eventually detected by checking bytes in scratchpad. Bytes 2,3,5,6,7 should be 0xff and bits from 7 to 4 of byte 4 are always ones.
> 
> https://www.analog.com/media/en/technical-documentation/data-sheets/MAX31826.pdf

This is something we can for sure add to the plugin code.

uzi18, 2023-04-10:
> This is something we can for sure add to the plugin code.

And family code 0x3B

tonhuisman, 2023-04-12:
> So, as told before, I suggest to:
> 
>     1. Rename the Plugin "DS18b20" to "1-Wire"

I do not like that idea, as many people are looking for that name when selecting the plugin or looking at their devices overview. I expect it will cause a lot of confusion.

uzi18, 2023-04-12:
OneWire temperature sensor

iz8mbw, 2023-04-12:
> > So, as told before, I suggest to:
> > 
> > 1. Rename the Plugin "DS18b20" to "1-Wire"
> 
> I do not like that idea, as many people are looking for that name when selecting the plugin or looking at their devices overview. I expect it will cause a lot of confusion.

I get your point, there is the "fear" that users don't understand that ESPEasy supports the "world wide famous" DS18B20 if the Plugin will be renamed.
Well, but, in the opposite side, can be the "not knowledge" that users know that ESPEasy supports (almost?) all the 1-Wire temperature sensors.
So "the truth" can be in middle and rename the Plugin to: "1-Wire Temperature (DS18B20)".
And update the doc specifying that ESPEasy supports the 1-Wire Temperature Sensors like the DS18B20.

TD-er, 2023-04-12:
I totally understand the reasoning for renaming.
I like `1-Wire Temperature (DS18b20)` as it could be a good name (with lower case 'b' :) ).
But on the other hand, it does again not mention the other sensors, making it again very similar to what the naming is now...

Also I don't know what's better known "1-Wire" or "Dallas".
As far as I know, there is vendor other than TI, using this protocol in their devices.

iz8mbw, 2023-04-12:
Yessss... 'b' in lower case according to the Datasheet:
[image]

in fact I cannot see any 'b' in lower case :-)
https://www.analog.com/media/en/technical-documentation/data-sheets/ds18b20.pdf

Still call it Dallas? Dallas is the **old** Company name developed it, then Maxim Integrated, now Analog Devices.
So the name is simply "1-Wire" like for "I²C", we don't call Philips I²C but simply I²C.

TD-er, 2023-04-12:
Hmm, I was sure it was with a lower-case 'b', mainly because it looked so strange...
No idea where I got this idea from.

iz8mbw, 2023-04-12:
Hmm, so rename "DS18**b**20" to "DS18**B**20"   :-)

iz8mbw, 2023-04-16:
Any decision made on this?
Rename or not?

tonhuisman, 2023-04-16:
@iz8mbw I have renamed the plugin and added support for the MAX31826 sensor. Can be downloaded from [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4715037414), so if you would be able to test, that would be helpful, as I don't own that type of sensor...

iz8mbw, 2023-04-16:
Ok, see you tomorrow with results.

tonhuisman, 2023-04-16:
I've made a suggested change to first check the CRC of the received data before checking for fixed resolution. This should be tested separately. Available in [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4715344193), once that's finished.

iz8mbw, 2023-04-16:
Ok so I'll test only the latest one?

tonhuisman, 2023-04-16:
Test both please.

iz8mbw, 2023-04-16:
Ok

iz8mbw, 2023-04-17:
Hi @tonhuisman 
Results of test using [this Action run](https://github.com/letscontrolit/ESPEasy/actions/runs/4715037414):
It recognize sensor as MAX31826:
[image]

It works by default in 12 bit Resolution (in fact it's written `Resolution:	12 (fixed)`).
If I try to set 9 bit of Resolution web page changes to 9 bit:
[image]

but I cannot read sensor:

```
672186 : Info   : DS   : Temperature: Error! (3b-14-fe-14-00-00-00-17 [MAX31826])
677188 : Info   : DS   : Temperature: Error! (3b-14-fe-14-00-00-00-17 [MAX31826])
```
Ok, it's fixed to 12 bit so make sense to have reading errors.

More I have done another test for @TD-er, I have added another 1-Wire sensor, our best friend DS18B20!!
So the DS18B20 with MAX31826 are on the same 1-Wire bus and both in Parasite Power and both works:
[image]

So now we can say that ESPEasy supports 1-Wire in Parasite Power configuration but/also if in the web interface for the second sensor (DS18B2 in my case) I read `Parasite Powered:	false`:
[image]

_________________________________________________________

Now the Results of test using [this Action run](https://github.com/letscontrolit/ESPEasy/actions/runs/4715344193):

All the same respect the other Action run.
I did the same test and results are the same.

Let me know if you need more.

Ciao!

iz8mbw, 2023-04-17:
ah, another note, my ESP32 does NOT connect to WiFi if there is voltage on the GPIO14 (the one used for the 1-Wire) (@TD-er also if I disable Band Steering).
So in order to let ESP32 connect to the WiFI, I removed the pull up, let's it connect to the Wifi and after gives power to the GPIO14.

tonhuisman, 2023-04-17:
> If I try to set 9 bit of Resolution web page changes to 9 bit:

Does it _only_ fail if the resolution is set different from 12? So it _does_ work as expected when set to 12?

iz8mbw, 2023-04-17:
I don't know what is expected (since I don't know the code), but the web interfacce gives me the possibility to change resolution from 12 to 9 bit.
So what fails is the sensor reading when I manually change from 12 to 9 bit.

tonhuisman, 2023-04-17:
> So what fails is the sensor reading when I manually change from 12 to 9 bit.

OK, so I'll have to protect setting the resolution when it is fixed 👍

iz8mbw, 2023-04-17:
if I set to 10 or 11 or 12 bit I still can read temperature.
If I set to 9 bit I have: Temperature: Error!

iz8mbw, 2023-04-17:
basically if from web interface you protect to change resolution when resolution is fixed it's the best way :-)

iz8mbw, 2023-04-17:
I agree with you, no sense to reduce resolution.
You can put a text note in the web interface when the sensors have fixed resolution like "do to change resolution since it's fixed on the chip"

iz8mbw, 2023-04-17:
anyway to have more than one 1-Wire sensor in Parasite Power configuration it's good!
More I think, if the Library supports it, it's quite normal that it works.
More than anything else it's strange that one sensor is recognized as `Parasite Powered: Yes` and the other one `No`.

tonhuisman, 2023-04-17:
> it's strange that one sensor is recognized as `Parasite Powered: Yes` and the other one `No` if both are in Parasite Power.

That is detected by the chip itself, ESPEasy just reads back the status from the chip. Maybe the detection algorithm is a bit 'touchy'.

iz8mbw, 2023-04-17:
The test "lab":
[image]

:-)

tonhuisman, 2023-04-17:
I've adjusted the reading, as there is a timing-dependency based on the resolution, so it now uses the resolution configured (or fixed) for that sensor. Available from [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4724939562).

Can you please test by configuring both sensors on a single task, and switch the resolution to 9 bit, so the MAX31826 will be read at 12 bit, and the DS18B20 at 9 bit? It is the intention to get that to work correctly.

tonhuisman, 2023-04-17:
Hm, last Actions run failed for network reasons.
Did an improvement when reading multiple sensors, now using the highest resolution time-out, as reading a 12 bit (fixed) sensor won't succeed when expecting only a 9 bit result.
New [Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4725149010) started.

iz8mbw, 2023-04-17:
Ok so in this i test only latest?
True?

tonhuisman, 2023-04-17:
Yup 👍

tonhuisman, 2023-04-17:
Next run will remove the extra log line I forgot to take out.

iz8mbw, 2023-04-17:
testing but how can I add 2 sensors on the same task?

iz8mbw, 2023-04-17:
ok got it

iz8mbw, 2023-04-17:
ok, but when I select "Dual" I can read only DS18B20 (the second one) because MAX31826 (the first one) is in read error

iz8mbw, 2023-04-17:
if I select only MAX31826 I can read it:
[image]

[image]

iz8mbw, 2023-04-17:
and now if I select 9bit I can still read it since in console I read:
[image]

iz8mbw, 2023-04-17:
but if I invert the sensor order, so DS18B20 as first sensor and MAX31826 as second sensor, I can read both:
[image]

[image]

iz8mbw, 2023-04-17:
when I have Dual sensors, always DS18B20 as first sensor, I can set 9bit of resolution and so DS18B20 will be in 9 bit and MAX31826 remains in 12 bit. So resolution set works.
[image]

iz8mbw, 2023-04-17:
Tested also to have them separatley and works:
[image]

tonhuisman, 2023-04-17:
I don't like the number of failed reads, can you configure them on 1 task again, as before, set the interval to 10 seconds, and retrieve some log from the usb serial output or tools/log page? (Tomorrow is fine, no hurry)

iz8mbw, 2023-04-17:
Yes will do it tomorrow morning.
But as I wrote, if I set DS18B20 as first sensor and MAX31826 as second sensor I don't have any temperature readings issue.

TD-er, 2023-04-17:
You can't (really this does not work!!!!) use parasite powered if using >1 OneWire sensor.

It really doesn't work as you need to have the data pin strongly pulled up during read and with >1 OneWire sensor (even among several tasks) on the same pin you will have interleaved reads.

iz8mbw, 2023-04-18:
@TD-er yesterday I wrong because MAX31826 was in Parasite Power (using one wire) and DS18B20 was NOT in Parasite Power (using two wires), sorry for that.
Today I did others test and I can confirm that more than one sensor in Parasite Power doesn't work, but this is true for the same Task.
If I use two Tasks I can read both MAX31826 and DS18B20 both in Parasite Power (using one wire).
[image]

DS18B20:
[image]

MAX31826:
[image]

As you cans e both works Parasite Power `Parasite Powered:	true`

TD-er, 2023-04-18:
Yes, it might work, but only under specific conditions.
Problem is that the tasks are not synced, so it may work for a while, but after some time you may end up with an entire sequence of where the parasite powered one is only having error.

If you really need this, you can either perform taskrun calls from the rules with at least 1 sec between the next call, or set both tasks to like 60 sec for the first one and 61 for the second.
This way only one in 60 calls may fail, which is probably acceptable.

TD-er, 2023-04-18:
Just to give an idea of how the sensors are being addressed:
S = start
R = read

```
S1      R1
  S2      R2
    S3      R3
      S4      R4
```
So it is possible that the "R1" is short enough to not interupt the reading of sensor 2.
However when the parasite sensor is sensor 1, the "S2" may interrupt the reading of this parasite powered chip.

iz8mbw, 2023-04-18:
OK, @tonhuisman do you need other test?
I mean, do you will do some code changing on 1-Wire that need future test?

iz8mbw, 2023-04-18:
Just for test I did another test: connect only sensor, the DS18B20 in Parasite Power, 10 seconds of polling and it does **not** work also if the address is recognized:

[image]

[image]

[image]

iz8mbw, 2023-04-18:
Now 3 sensors:
**Two DS18B20 in normal power configuration using default resolution (12 bit) + MAX31826 (12 bit) in Parasite Power all in the same task:**
[image]

tonhuisman, 2023-04-18:
Where I wrote 'both sensors' I intended to say 'both a DS18B20 and a MAX31826'; to see if having a resolution of 9 bits combined with the 12 bit fixed resolution will interfere with reading either sensor. And to revert the order of the sensors to see if that makes a difference.

[10 more replies]

PR #4617 (2023-05-07): [P004] Rename to 1-Wire Temperature, add support for MAX31826 sensor
Resolves #4601 

Features:
- Rename from `DS18b20` to `1-Wire Temperature`.
- Add support for MAX31826 sensor, that has a fixed 12 bit resolution.

TODO:
- [x] Testing ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/4601#issuecomment-1512786347), with some remarks)
- [x] Update documentation
closed 2023-05-07 by TD-er